lbr-stack/lbr_fri_ros2_stack #1: Missing dependencies

Giggus, 2021-05-14:
Controller_spawner is failing because some dependencies are not satisfied when running

```
"rosdep install --rosdistro melodic --ignore-src --from-paths src"
```

Fixed by installing **ros-melodic-moveit** and **ros-melodic-joint-trajectory-controller**

It might be worth investigating which are the missing packages in the package dependencies rather then installing these two packages which include many others that might not be useful.

anubhav-dogra, 2022-05-04:
Can you please help me with the following errors. 
The simulation is working fine with the melodic, but I am not able to understand the issue with Foxy. Please help to understand what I am missing here 
Thanks. 

[spawner.py-6] [INFO] [1651662207.865197859] [spawner_position_trajectory_controller]: Waiting for /controller_manager services
[rviz2-8] Warning: Invalid frame ID "lbr_link_1" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-8] Warning: Invalid frame ID "lbr_link_2" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-8] Warning: Invalid frame ID "lbr_link_3" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-8] Warning: Invalid frame ID "lbr_link_4" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-8] Warning: Invalid frame ID "lbr_link_5" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-8] Warning: Invalid frame ID "lbr_link_6" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp

mhubii, 2022-05-19:
Hi and sorry for seeing this so late. This usually happens when the communication to the robot is not setup for some reason

mhubii, 2022-05-19:
what realtime priority do you have on your system? The launch file launches the controller with a realtime priority of 99 

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0d2cc6e2f6bcfbff860fa87ecb87a64520f55181/lbr_bringup/launch/lbr_control.launch.py#L69

If the realtime priority is not set, the communicating thread will not be launched 

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/issues/14

anubhav-dogra, 2022-05-31:
Hi, there
thank you for the information. 
I did change the real time priority as mentioned. But I'm still facing the same error. 
as of now, my arg for simulation is true as, sim:= true

I am using ubuntu on a Virtual machine.. and that should not be a problem I guess (?)

mhubii, 2022-05-31:
if you wish to connect to the real robot, you'll have to set `sim:=false`. Also you'll have to launch the [LBRServer](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/main/server/fri_ros2/LBRServer.java) application on the SmartPad

anubhav-dogra, 2022-05-31:
Yes I'm **able to work** with the real robot with the current "lbr_fri_ros_stack" in melodic... 

I'm trying to workout the same for **ros2 stack** but I'm stuck with the error mentioned above in my earlier comment. 
I edited the **rtprio** in the system, but the **error still persists**. 

"[spawner.py-6] [INFO] [1651662207.865197859] [spawner_position_trajectory_controller]: Waiting for /controller_manager services......."

anubhav-dogra, 2022-05-31:
- Right now Im trying with the sim:=true, but the same occurs even with sim:=false. 
- Yes I have replaced user with my username !
- please find below the entire terminal output with the command as mentioned by you with | grep ERROR and sim:=true. 

anubhav@ubuntu:~/lbr_fri_ros2_stack_ws$ ros2 launch lbr_bringup lbr_bringup.launch.py model:=med14 sim:=true | grep ERROR
[move_group-7] [ERROR] [1654036072.629839378] [moveit.ros.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[ERROR] [gzserver-1]: process has died [pid 2632, exit code 255, cmd 'gzserver                                                                       -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[spawner.py-6] [ERROR] [1654036082.251245291] [spawner_position_trajectory_controller]: Controller manager not available
[spawner.py-5] [ERROR] [1654036082.253981082] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner.py-6]: process has died [pid 2644, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py position_trajectory_controller --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner.py-5]: process has died [pid 2642, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[move_group-7] [ERROR] [1654036087.953478906] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected: position_trajectory_controller/follow_joint_trajectory

mhubii, 2022-05-31:
oh okay sorry I misunderstood you. I though it was working well in simulation for foxy, but it doesn't. Please apologize. What is the error about the Gazebo server? Does just launching Gazebo work? In a terminal write `gazebo`

mhubii, 2022-05-31:
I'll try this later today, as I currently don't have access to a PC. Hopefully I can replicate the issue and find a fix. Thank you for sharing

mhubii, 2022-11-07:
hey @anubhav-dogra , did you get to look into this again? I recently tried the system with the newest foxy release and everything worked well.

mhubii, 2024-01-11:
hi @Nicolai-98 . Yes this is a very well known issue. The very first launch is somehow extremely slow and doesn't spawn the model.

Run a launch until Gazebo opens, then exit (ctrl+c). Re-run and everything should work fine. I think this is some Gazebo related bug. Please let me know if this works

mhubii, 2024-01-18:
hi @Nicolai-98 , great to hear that worked! The error `no transform` suggests no model was loaded into Gazebo, hence the `robot_state_publisher` can't provide them.

Would you be happy to share the docker file, maybe create a pull request, I can create a `dev-humble-docker` branch. That would create a lot of value for many people. If not, no problem!

mhubii, 2024-01-18:
Closing this issue for now, as dependencies seem to work well. Opening #149

closed 2024-01-18 by mhubii