lbr-stack/lbr_fri_ros2_stack #3: KONI IP address

qsh-zh, 2021-11-03:
Thanks for sharing the amazing project.

I am wonder can we set the KONI FRI address similar to x66. The default value for those two are 192.170.10.2 and 172.31.1.147. The jave file, [this line](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0942eeabd06e376670e19671240d775eb4650cc8/server/com.kuka.connectivity.fri.fri_ros/LBRServer.java#L36) builds connection on 172.31.1.148. Do we need some changes to the KONI setting?

mhubii, 2021-11-03:
hi @qsh-zh sorry this project is very raw atm. I can have a look at it tomorrow and add a flexible IP

mhubii, 2021-11-10:
hi @qsh-zh sorry currently have some other things to do and did not get a look at it yet. The ip adress "172.31.1.148" specifies the ip of the client, the client in this case is your computer that you connect to the KUKA controller, which is the FRI server. Currently I connect through port x66, not through the KONI, but keep in mind that the FRI is a UDP connection, so it is not necessary to go through KONI.

I will investigate this further and let you know once I came around doing it. Sorry again

qsh-zh, 2021-11-11:
Thanks for your feedback and I do not realize we can communicate FRI via x66. Thanks.

mhubii, 2022-02-03:
Hi @qsh-zh. X66 supports control rates >2ms. If you want to use KONI, you can set the IP at https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/256668422fb083dd73a2051ac29492dba0a31736/server/com.kuka.connectivity.fri.fri_ros/LBRServer.java#L36 to 192.170.10.2, for example. However, I recently experience connection drops. They are caused by the ros2_control_node. I'll work on a fix now, and add something so one can choose between KONI/X66

mhubii, 2022-02-08:
Hi @qsh-zh, the connection issues should be gone now. On running the [LBRServer](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/foxy/server/fri_ros2/LBRServer.java), you'll also be asked to set the URL now. Therefore, if you want to connect via KONI, you need to set your computer's IP to 192.170.10.1, and select it on the SmartPAD

closed 2022-02-14 by mhubii