lbr-stack/lbr_fri_ros2_stack #8: Cross Platform Support

mhubii, 2022-03-01:
Add cross platform support

mathadakshay, 2024-09-01:
Hello,

I will work on this issue. can you please describe a bit in detail about the issue ? I am a beginner on github

mhubii, 2024-09-07:
hi @mathadakshay. Problem here is to support Windows. This would entail:

- Visibility control: https://ms-iot.github.io/ROSOnWindows/Porting/SymbolVisibility.html
- Build recipies for Windows: https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/4c98e26975364d00b46d132e3860462ebb050ad0/.github/workflows/build.yml#L18
- This survey might be relevant: https://discourse.ros.org/t/windows-survey-results/39502

mhubii, 2025-10-16:
Windows support should now be investigated with the upcoming switch to `pixi`/`conda`

https://discourse.openrobotics.org/t/upcoming-switch-of-windows-installation-to-pixi-conda/41916

mhubii, 2026-10-09:
The stack can now be installed via:

```shell
pixi add \ 
    ros-jazzy-lbr-fri-ros2-stack \
    lbr-fri-client-sdk==1.15
```

Currently, only FRI `1.x` are supported. `2.x` support is blocked by a `lbr_fri_idl` supporting `1.x` and `2.x` on a single branch.