lbr-stack/lbr_fri_ros2_stack #15: Compatability with ubuntu 18. 04

Yitian-Xian, 2022-05-25:
Hi, may I know if I can use this package in ubuntu 18.04? I have melodic installed but I am really new at ros2. Thanks!

anubhav-dogra, 2022-06-01:
you can melodic version of this stack with lbr_fri_ros_stack. 
You can select melodic branch from all the branches provided.

Yitian-Xian, 2022-06-02:
@anubhav-dogra Thanks for your reply! I am so stupid I did not check! Sorry!

Yitian-Xian, 2022-06-02:
@anubhav-dogra Hi, when we are using this package to run Med 7, the robot would have some 'beep' noise. Have you encountered this problem before? Everything looks good except this issue.

anubhav-dogra, 2022-06-03:
You might need to check errors or warnings on the KUKA SmartPad.

Yitian-Xian, 2022-06-15:
@mhubii Typically I use 5 ms, I also tried 10 ms, and I do not think communication rates is an issue. I found this case always happens if I make the robot move fast. For example, after the robot moves, when I command it to return its original pose, I only give a desired position without controlling its velocity. Then the robot would make some noise. After I give a small velocity, this case rarely happens. Hope this can help!

mhubii, 2022-06-15:
Thanks for the hints @Yitian-Xian, then let's keep this issue open for visibility. Maybe someone can solve it. I currently don't have access to a robot so can't verify

mhubii, 2023-01-13:
We have now double checked this matter.

It appears that a beeping noise occurs when noisy commands are sent to the motors. Make sure smooth controls are sent when using the system.

closed 2023-01-13 by mhubii