lbr-stack/lbr_fri_ros2_stack #16: Group 'arm' was not found.

yiwangyw, 2022-08-03:
Hi,

Thank you for your impressive work first.

When I try to run example to test simulation, error occurred. Some details is shown below.

I run `source devel/setup.bash` and `roslaunch lbr_moveit moveit_planning_execution.launch model:=iiwa14 sim:=true` in command and  gazebo and rviz work. However, when I open a new terminal and run `cd src/lbr_fri_ros_stack/examples/moveit_python_examples/src` and `rosrun moveit_python_examples src/moveit_motion_examples.py`, the error occurs:

`[ERROR] [1659491678.046215306]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1659491678.047591526]: Loading robot model 'iiwa14'...
[ INFO] [1659491678.047669979]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1659491678.156816603, 3945.746000000]: Group 'arm' was not found.
Traceback (most recent call last):
  File "/home/killentest/lbr_fri_ros_ws/src/lbr_fri_ros_stack/examples/moveit_python_examples/src/moveit_motion_examples.py", line 18, in <module>
    group = moveit_commander.MoveGroupCommander(group_name)
  File "/opt/ros/noetic/lib/python3/dist-packages/moveit_commander/move_group.py", line 66, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(
RuntimeError: Group 'arm' was not found.`

I try to run `rosparam list | grep robot_description_semantic`. It shows `lbr/robot_description_semantic'. And I check the iiwa14.srdf, the group name is "arm".

Could you help me deal with it? Thank you for your time!

mhubii, 2022-08-03:
sorry I'll have to look into this

yiwangyw, 2022-08-04:
> 
It may be caused by the  warning below when run`roslaunch lbr_moveit moveit_planning_execution.launch model:=iiwa14 sim:=true`.

`[ WARN] [1659577650.735201349]: Falling back to using the move_group node's namespace (deprecated Melodic behavior).
[ INFO] [1659577650.748149938]: Loading robot model 'iiwa14'...
[ INFO] [1659577650.748251986]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1659577650.882758389]: Could not identify parent group for end-effector 'ee'`

yiwangyw, 2022-08-04:
I check the codes of moveit. It will show warning when namespace is empty. If there is no namespace, the `<group>` may be not able to work

yiwangyw, 2022-08-04:
Thanks for your time! The environment variable `ROS_NAMESPACE` was missing. Run `export ROS_NAMESPACE=lbr` and ' rosrun moveit_python_examples moveit_motion_examples.py` and it works.

closed 2022-08-04 by yiwangyw

mhubii, 2022-08-08:
thanks so much!