lbr-stack/lbr_fri_ros2_stack #17: Problem about synchronize project

yiwangyw, 2022-08-05:
When we download the LBRServer.java in noetic branch and add it into `src` file inside our new project, we meet the error `The project 'SunriseProject' contains Java compilation errors. Please fix the errors to allow project synchronization.`

I am not sure if there is the problem because of the different edition of sunrise OS. We use Sunrise OS, workbench and FRI 1.10. Do them have different editions of JAVA? Could you help us with this problem?

yiwangyw, 2022-08-06:
I am not familiar with Java but we follow the automatic instruction of IDE and it works.

abaicoming, 2025-01-03:
hi @yiwangyw ，I had the same situation as you, as shown below.
[image]
You said you followed the ''automatic instruction of IDE'' and it works. How do I get the ''automatic instruction of IDE''? Thank you very much.