lbr-stack/lbr_fri_ros2_stack #18: Unbelievable mass in xacro/URDF

cmower, 2022-10-31:
The mass in the med7 xacro file on the `melodic` branch seems to me to be incorrect - 46kg for the first link is nearly double the weight of the robot itself! The other links have equally unbelievable masses.

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/81dfea30dc89903d27df454f431269a52c6610ea/lbr_description/urdf/med7.xacro#L36

On the `main` branch, things are much more sensible:

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0d2cc6e2f6bcfbff860fa87ecb87a64520f55181/lbr_description/urdf/med7/med7_description.urdf.xacro#L29

mhubii, 2022-11-01:
thanks for raising this issue. These are obtained from autodesk. Let's look into fixing it

cmower, 2022-11-01:
I would also note that the inertia tags are different, and I guess incorrect, too.

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/81dfea30dc89903d27df454f431269a52c6610ea/lbr_description/urdf/med7.xacro#L37

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0d2cc6e2f6bcfbff860fa87ecb87a64520f55181/lbr_description/urdf/med7/med7_description.urdf.xacro#L30

My guess is the `main` branch is the correct.

mhubii, 2022-11-01:
I wouldn't trust any of these values

mhubii, 2022-11-28:
fixed in the above