lbr-stack/lbr_fri_ros2_stack #19: Cartesian space controller

IanYangChina, 2022-11-02:
Hi, thank you for your valuable contribution to the community!
I'm wondering if this package supports controlling the robot's end-effector directly in the Cartesian space without using moveit?
Thank you very much?

mhubii, 2022-11-07:
hi @IanYangChina can you elaborate a bit further on what you are trying to do? This package just exposes the robot to `ros2_control`, which then for example is used by moveit. Alternatively, you can decide not to launch moveit and implement a controller yourself

IanYangChina, 2022-11-07:
@mhubii Hi, thanks for the reply! I want to directly send PostStamp message to move a Lbr14 arm in the Cartesian space. I used to do it with a topic in the iiwastack ros topics. But I couldn't find a similar ros topic while using this Fri ros driver. Btw, I'm using the melodic branch. 
Thanks a lot!

mhubii, 2022-11-08:
well there are different ways to go about this, are you planning to send incremental updates? Or say just the final pose?

matthias-mayr, 2022-11-08:
We had the same issue and implemented a [Cartesian impedance controller](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) that we use with `iiwa_ros` and that FRI ROS1 driver. I guess if you're using ROS1 it could serve your needs and would be compatible with this repo (if I am assuming correctly that it exposes effort interfaces via ros_control).

Typically we use it in combination with a Cartesian trajectory generator that I was planning to open source as well, but you could also send poses directly to the robot and use the filtering.

IanYangChina, 2022-11-08:
@mhubii final pose will suffice.

mhubii, 2022-11-08:
you can launch the robot
```bash
source devel/setup.bash
roslaunch lbr_moveit moveit_planning_execution.launch model:=med7 sim:=false
```
then launch a pose node (this is taken from the `iiwa_stack` (as highlighted in readme)
```bash
source devel/setup.bash
roslaunch lbr_control cartesian_pose.launch
```
then publish a pose
```
rostopic pub /lbr/command/CartesianPose geometry_msgs/PoseStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
pose:
  position:
    x: 0.0
    y: 0.0
    z: 1.2
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0"
```
you should see the robot move. This still uses moveit under the hood. You can also do this programmatically by publishing to `/lbr/command/CartesianPose`. You can get the pose from `/lbr/state/CartesianPose`.

For bypassing movetit you'd have to implement this yourself. Sorry but our focus lies on ros2 at the moment, hope it still helps

IanYangChina, 2022-11-08:
@mhubii Thank you. Your answer does help!

IanYangChina, 2022-11-08:
@matthias-mayr Thanks for the suggestion. I will have a look at your package :)

closed 2022-11-08 by IanYangChina