lbr-stack/lbr_fri_ros2_stack #21: Support multiple robots

mhubii, 2022-11-08:
Investigate support for multiple robots

TheMipmap, 2023-10-27:
Hi @mhubii, 

Really looking forward to this enhancement!

My group and I, at Aalborg University, are currently working on implementing two Kuka IIWA 7s on a 'torso'. As such, the support for multiple arms is a necessity for us. :)

So far, we have made a custom URDF, with the torso and both of the robots and reconfigured moveit with the setup assistant. Sadly, however, even though the simulation shows the correct URDF with both robots, it cannot plan trajectories for any of them. 

It would be highly appreciated if you could give us a few pointers as to what need to be changed in order for two robots to work concurrently.

mhubii, 2023-10-30:
Hi @TheMipmap and sorry for the late response. This is actually functioning now. 

I can respond in more detail tomorrow. But in essence you can 

- connect the two robots to a switch
- in sunrise workbench, you configure a unique IP address for each robot
- in your Java application, you assign one robot port ID 30200 and the other robot 30201

when you launch this stack, you can amend the launch with port_id:=30201.

With regard to moveit, it somehow  has issues with namespacing multiple robots. But we can chat about that

mhubii, 2023-10-30:
Supported since #121. Will open new issues indicating necessary documentation and the issues regarding moveit in multi-robot tomorrow

mhubii, 2023-10-30:
May I kindly ask you to open a new issue regarding missing documentation? That would be great. Thank you for the help @TheMipmap

TheMipmap, 2023-10-31:
Of course, I'll do that! :)

mhubii, 2023-11-02:
Closing this issue for now as introduced in #121. Added issue #131 for documentation etc

closed 2023-11-02 by mhubii