lbr-stack/lbr_fri_ros2_stack #24: FRI Session Could Not Be Established

MrstupidJ, 2022-11-15:
Hi we have followed all the steps to conenct the robot controller to the windows PC to setup the LBR Server application in the controller. Now we are trying to establish a connection between the robot and the linux computer running ROS2 application. However, when we start the LBR server from the Kuka Smart Pad we are getting the following error which points to that the error is coming due to configuration issues. We have gone over the network configuration multiple times and they all seem correct. Are we meant to run a certain version of the sunrise software? We are currently running Sunrise Workbench - Med 1.15.

We do not have access to your school's sharepoint to look at any of the additional resources. We are able to successfully ping the robot controller from the linux computer. 

[image]

Please look at this image of the error we get from the smart pad. It is pointing at this [line](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0d2cc6e2f6bcfbff860fa87ecb87a64520f55181/server/fri_ros2/LBRServer.java#L111) 

[image]
[image]

Any help here would be greatly appreciated.

mhubii, 2022-11-17:
hey @MrstupidJ , are you sure you configured the IP correctly? From the image above, it seems you can't ping the robot. The robot has 2 ethernet ports, KONI under `192.170.10.2` and the non real-time one under `172.31.1.147`. Where did you plug in the ethernet cable?

mhubii, 2022-11-17:
also you set the the Client command mode to `NO_COMMAND_MODE`. This defines what the commands you send to the robot. To begin with, select `POSITION`

MrstupidJ, 2022-11-18:
> Thanks for your reply @mhubii ! 

I commented out the [line](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0d2cc6e2f6bcfbff860fa87ecb87a64520f55181/lbr_bringup/launch/lbr_control.launch.py#L69)  and it worked! 

Just to clarify, the DUP! thing is normal when you try to Ping the KUKA FRI. It seems like it means to do so to achieve high frequency real-time data exchange. And yes it is a good and safe practice to run in T1 mode first,

closed 2022-11-18 by MrstupidJ

mhubii, 2022-11-18:
great, thank you for the feedback!