lbr-stack/lbr_fri_ros2_stack #25: Build URDF at compile time

cmower, 2022-11-17:
It's useful to have access to URDF file generated from the xacro description (e.g. [PyBullet doesn't support loading from xacro, only URDF](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.sbnykoneq1me)), in [ROS1 this is possible](http://wiki.ros.org/xacro#Building_from_CMakeLists.txt). I assume it should be possible in ROS2 - this would be a worthwhile feature to add to the `lbr_description` package.

mhubii, 2022-11-28:
this is still possible, see https://github.com/ros2-gbp/xacro-release/blob/debian/foxy/xacro/cmake/xacro-extras.cmake.

When doing so, variables are evaluated at compile time, meaning e.g. `robot_name` and `sim` in https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/5ccc5263ed7883c08f6890d0339a21dd6114e2a9/lbr_description/urdf/med7/med7.urdf.xacro#L9 will be frozen

shaoyifei96, 2025-01-17:
Thanks for making this! 
I wonder how accurate the xacro file is, and how did you obtain it? Would it be possible to do inverse dynamics control using this model?