lbr-stack/lbr_fri_ros2_stack #26: Gazebo not loading and robot state not being published

cmower, 2022-11-23:
After updating my system via `apt update; apt upgrade`, Gazebo fails to load and the robot model is not published (see below). 

[image]

The terminal output is as follows.

```
$ ros2 launch lbr_bringup lbr_bringup.launch.py model:=med7 sim:=false
[INFO] [launch]: All log files can be found below /home/chrismo/.ros/log/2022-11-23-11-25-27-384196-sie-063-8934
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [fri_ros2_control_node-1]: process started with pid [8938]
[INFO] [robot_state_publisher-2]: process started with pid [8940]
[INFO] [spawner.py-3]: process started with pid [8942]
[INFO] [spawner.py-4]: process started with pid [8944]
[INFO] [move_group-5]: process started with pid [8946]
[INFO] [rviz2-6]: process started with pid [8948]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link lbr_link_0 had 1 children
[robot_state_publisher-2] Link lbr_link_1 had 1 children
[robot_state_publisher-2] Link lbr_link_2 had 1 children
[robot_state_publisher-2] Link lbr_link_3 had 1 children
[robot_state_publisher-2] Link lbr_link_4 had 1 children
[robot_state_publisher-2] Link lbr_link_5 had 1 children
[robot_state_publisher-2] Link lbr_link_6 had 1 children
[robot_state_publisher-2] Link lbr_link_7 had 1 children
[robot_state_publisher-2] Link lbr_link_ee had 0 children
[robot_state_publisher-2] [INFO] [1669202727.856092293] [robot_state_publisher]: got segment lbr_link_0
[robot_state_publisher-2] [INFO] [1669202727.856181646] [robot_state_publisher]: got segment lbr_link_1
[robot_state_publisher-2] [INFO] [1669202727.856193074] [robot_state_publisher]: got segment lbr_link_2
[robot_state_publisher-2] [INFO] [1669202727.856198636] [robot_state_publisher]: got segment lbr_link_3
[robot_state_publisher-2] [INFO] [1669202727.856203758] [robot_state_publisher]: got segment lbr_link_4
[robot_state_publisher-2] [INFO] [1669202727.856208634] [robot_state_publisher]: got segment lbr_link_5
[robot_state_publisher-2] [INFO] [1669202727.856213467] [robot_state_publisher]: got segment lbr_link_6
[robot_state_publisher-2] [INFO] [1669202727.856218401] [robot_state_publisher]: got segment lbr_link_7
[robot_state_publisher-2] [INFO] [1669202727.856223301] [robot_state_publisher]: got segment lbr_link_ee
[robot_state_publisher-2] [INFO] [1669202727.856228205] [robot_state_publisher]: got segment world
[move_group-5] [WARN] [1669202727.893521881] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-5] [WARN] [1669202727.893575986] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-5] Parsing robot urdf xml string.
[move_group-5] [INFO] [1669202727.896132640] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.000901863 seconds
[move_group-5] [INFO] [1669202727.896163263] [moveit_robot_model.robot_model]: Loading robot model 'med7'...
[move_group-5] [INFO] [1669202727.896174923] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-5] Link lbr_link_0 had 1 children
[move_group-5] Link lbr_link_1 had 1 children
[move_group-5] Link lbr_link_2 had 1 children
[move_group-5] Link lbr_link_3 had 1 children
[move_group-5] Link lbr_link_4 had 1 children
[move_group-5] Link lbr_link_5 had 1 children
[move_group-5] Link lbr_link_6 had 1 children
[move_group-5] Link lbr_link_7 had 1 children
[move_group-5] Link lbr_link_ee had 0 children
[fri_ros2_control_node-1] [INFO] [1669202728.030027653] [controller_manager]: Loading controller 'position_trajectory_controller'
[spawner.py-4] [INFO] [1669202728.047236486] [spawner_position_trajectory_controller]: Loaded position_trajectory_controller
[fri_ros2_control_node-1] [INFO] [1669202728.047996932] [controller_manager]: Configuring controller 'position_trajectory_controller'
[fri_ros2_control_node-1] [INFO] [1669202728.048120234] [position_trajectory_controller]: Command interfaces are [position] and and state interfaces are [position].
[fri_ros2_control_node-1] [INFO] [1669202728.049041002] [position_trajectory_controller]: Controller state will be published at 50.00 Hz.
[fri_ros2_control_node-1] [INFO] [1669202728.049986720] [position_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[fri_ros2_control_node-1] [INFO] [1669202728.053447614] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner.py-3] [INFO] [1669202728.066228102] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[move_group-5] [INFO] [1669202728.210257125] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-5] [INFO] [1669202728.210422377] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-5] [INFO] [1669202728.210931595] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-5] [INFO] [1669202728.211717265] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-5] [INFO] [1669202728.211736157] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-5] [INFO] [1669202728.211910639] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-5] [INFO] [1669202728.211921962] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-5] [INFO] [1669202728.212149595] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-5] [INFO] [1669202728.212400382] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-5] [WARN] [1669202728.212437399] [moveit.ros.occupancy_map_monitor]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-5] [ERROR] [1669202728.212477943] [moveit.ros.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[... 38 lines cut ...]
[move_group-5] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-5] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-5] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-5] Loading 'move_group/MoveGroupMoveAction'...
[move_group-5] Loading 'move_group/MoveGroupPlanService'...
[move_group-5] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-5] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-5] 
[move_group-5] You can start planning now!
[move_group-5] 
[rviz2-6] [INFO] [1669202728.600498425] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] [INFO] [1669202728.600616733] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-6] [INFO] [1669202728.640191407] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] Parsing robot urdf xml string.
[rviz2-6] Warning: Invalid frame ID "lbr_link_1" passed to canTransform argument source_frame - frame does not exist
[rviz2-6]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-6] Warning: Invalid frame ID "lbr_link_2" passed to canTransform argument source_frame - frame does not exist
[rviz2-6]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-6] Warning: Invalid frame ID "lbr_link_3" passed to canTransform argument source_frame - frame does not exist
[rviz2-6]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-6] Warning: Invalid frame ID "lbr_link_4" passed to canTransform argument source_frame - frame does not exist
[rviz2-6]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
```

This was not an issue prior to updating/upgrading and I have seen the same problem on @Royo666's system - perhaps a bug in one of the dependencies?

mhubii, 2022-11-23:
thanks for the feedback. What branch are you one? Can you please set `sim:=true` when you plan to launch into simulation

mhubii, 2022-11-23:
setting `sim:=true` solved the issue indeed

closed 2022-11-23 by mhubii

Royo666, 2022-11-25:
Hello,

I usually set ```sim:=false``` to avoid launching the Gazebo when I just want to test the trajectory on Moveit. However, even I try to set ```sim:=true```, it does launch the Gazebo simulation GUI but no model is inside. Please see follows.
[image]
[image]

The part of the coding after the launching code is follows
```
[INFO] [launch]: All log files can be found below /home/zhanchong/.ros/log/2022-11-25-10-09-50-337519-zhanchong-virtual-machine-32166
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [32170]
[INFO] [gzclient   -2]: process started with pid [32173]
[INFO] [spawn_entity.py-3]: process started with pid [32176]
[INFO] [robot_state_publisher-4]: process started with pid [32178]
[INFO] [spawner.py-5]: process started with pid [32182]
[INFO] [spawner.py-6]: process started with pid [32187]
[INFO] [move_group-7]: process started with pid [32194]
[INFO] [rviz2-8]: process started with pid [32196]
[gzserver-1] sh: 0: getcwd() failed: No such file or directory
[gzclient   -2] sh: 0: getcwd() failed: No such file or directory
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link lbr_link_0 had 1 children
[robot_state_publisher-4] Link lbr_link_1 had 1 children
[robot_state_publisher-4] Link lbr_link_2 had 1 children
[robot_state_publisher-4] Link lbr_link_3 had 1 children
[robot_state_publisher-4] Link lbr_link_4 had 1 children
[robot_state_publisher-4] Link lbr_link_5 had 1 children
[robot_state_publisher-4] Link lbr_link_6 had 1 children
[robot_state_publisher-4] Link lbr_link_7 had 1 children
[robot_state_publisher-4] Link lbr_link_ee had 0 children
[robot_state_publisher-4] [INFO] [1669370990.794841155] [robot_state_publisher]: got segment lbr_link_0
[robot_state_publisher-4] [INFO] [1669370990.794892118] [robot_state_publisher]: got segment lbr_link_1
[robot_state_publisher-4] [INFO] [1669370990.794895899] [robot_state_publisher]: got segment lbr_link_2
[robot_state_publisher-4] [INFO] [1669370990.794898578] [robot_state_publisher]: got segment lbr_link_3
[robot_state_publisher-4] [INFO] [1669370990.794900980] [robot_state_publisher]: got segment lbr_link_4
[robot_state_publisher-4] [INFO] [1669370990.794903179] [robot_state_publisher]: got segment lbr_link_5
[robot_state_publisher-4] [INFO] [1669370990.794905623] [robot_state_publisher]: got segment lbr_link_6
[robot_state_publisher-4] [INFO] [1669370990.794907828] [robot_state_publisher]: got segment lbr_link_7
[robot_state_publisher-4] [INFO] [1669370990.794910152] [robot_state_publisher]: got segment lbr_link_ee
[robot_state_publisher-4] [INFO] [1669370990.794912427] [robot_state_publisher]: got segment world
[move_group-7] [WARN] [1669370990.831985292] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-7] [WARN] [1669370990.832017759] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-7] Parsing robot urdf xml string.
[move_group-7] [INFO] [1669370990.833593596] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-7] [INFO] [1669370990.833614830] [moveit_robot_model.robot_model]: Loading robot model 'med7'...
[move_group-7] [INFO] [1669370990.833619456] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-7] Link lbr_link_0 had 1 children
[move_group-7] Link lbr_link_1 had 1 children
[move_group-7] Link lbr_link_2 had 1 children
[move_group-7] Link lbr_link_3 had 1 children
[move_group-7] Link lbr_link_4 had 1 children
[move_group-7] Link lbr_link_5 had 1 children
[move_group-7] Link lbr_link_6 had 1 children
[move_group-7] Link lbr_link_7 had 1 children
[move_group-7] Link lbr_link_ee had 0 children
[spawner.py-6] [INFO] [1669370991.070138152] [spawner_position_trajectory_controller]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1669370991.092248679] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[ERROR] [gzserver-1]: process has died [pid 32170, exit code 255, cmd 'gzserver                                                                       -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[spawn_entity.py-3] [INFO] [1669370991.163013553] [spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1669370991.163251470] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] [INFO] [1669370991.164744645] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-3] [INFO] [1669370991.177514296] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1669370991.177713325] [spawn_entity]: Waiting for service /spawn_entity
[move_group-7] [INFO] [1669370991.185849349] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-7] [INFO] [1669370991.185984911] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-7] [INFO] [1669370991.186496601] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[... cut ...]
[spawn_entity.py-3]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 710, in spin_once
[spawn_entity.py-3]     handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
[spawn_entity.py-3]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 696, in wait_for_ready_callbacks
[spawn_entity.py-3]     return next(self._cb_iter)
[spawn_entity.py-3]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 593, in _wait_for_ready_callbacks
[spawn_entity.py-3]     _rclpy.rclpy_wait(wait_set, timeout_nsec)
```

Thanks

cmower, 2022-11-25:
> I usually set ```sim:=false``` to avoid launching the Gazebo when I just want to test the trajectory on Moveit. However, even I try to set ```sim:=true```, it does launch the Gazebo simulation GUI but no model is inside. Please see follows.
> 

@Royo666, this is expected. When ```sim:=false``` this means you should be connected to the real robot - rviz is expecting to hear messages from the real robot. Hence, rviz doesn't know where to put the meshes, and the default position for the frames is 0,0,0.

I think your problem is unrelated to this issue.