lbr-stack/lbr_fri_ros2_stack #33: FRI session drops during joint torque control

MrstupidJ, 2022-12-06:
Hi, 

Thanks for the work! We are doing joint torque control via FRI with this repo and it works well most of time. However we find the FRI session gets disposed when we have some physical interaction with the robot (like doing some small handguiding motions). From the log it seems like the connection quality drops to "poor" and hence get disconnected (please see the screenshot below). 

[image]

Any idea on what could be the cause? 

Btw I think here should be "FRI command mode" instead of control
https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/0d2cc6e2f6bcfbff860fa87ecb87a64520f55181/server/fri_ros2/LBRServer.java#L94

Thanks for your help!

mhubii, 2022-12-06:
hi @MrstupidJ and thanks for the feedback! Not exactly sure why the connection quality drop, sorry.

We are currently working on a major re-work, which should be live end of this week. I hope this fixes all connection issues.

May I ask what branch you are working on?

MrstupidJ, 2022-12-06:
Thanks for your reply. We are using the foxy branch. 

For the torque control we just choose Joint impedance control mode and toque command mode when running the Java app from KUKA smartPad.

mhubii, 2022-12-13:
that's the correct way to do it. When using the torque command mode a communication rate of at least 5ms is to be maintained, see below.
[image]

when launching the java app, you'll have to select <=5ms, maybe that solves the issue

mhubii, 2022-12-13:
sorry for the late response! Been busy moving this to humble. It will become a whole lot easier to use this stack. It will also become possible to bypass ros2_control. There will be a `/lbr_state` topic and a `/lbr_command`. Very straight forward

mhubii, 2022-12-13:
i wasn't able to work on this code for half a year but am now fully commited to making this repository a good experience. Thanks for the understanding

mhubii, 2022-12-13:
re foxy: we will try to move the changes made for humble to all other branches as well ASAP

mhubii, 2022-12-16:
you can find the changes in the `dev-foxy-lbr-fri-ros2` branch https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/tree/dev-foxy-lbr-fri-ros2 

this should work already but isn't merged into foxy yet. Could you please try that and report if your issue persists? Thanks!

MrstupidJ, 2022-12-16:
Thank you so much for your hard work! I will have a try and let you know how it goes.

MrstupidJ, 2022-12-22:
> you can find the changes in the `foxy` branch.
> 
> Could you please try that and report if your issue persists? Thanks!

Hi, We've tried the latest commit in `foxy` branch, and the basic function looks fine for now. 

For the connection dropping issue, we are suspecting it is caused by the real-time ubuntu kernel. Would you mind checking which version of ubuntu you are using? You can get it by typing `uname -r` in terminal.

mhubii, 2023-01-06:
we are using ubuntu 20 / 22 right now

MrstupidJ, 2023-01-12:
Thanks for your work and reply, and happy new year! 

I kind of think the real-time kernel matters when we use FRI. We tried two ubuntu 20.04 machines, and the one without real-time kernel gets disconnection error much more often (You can see "Quality change signalled POOR" from KUKA Smartpad). The robot cannot take any command with poor connection quality. 
[image]

closed 2023-01-12 by MrstupidJ