lbr-stack/lbr_fri_ros2_stack #35: Velocity limit seems too high

cmower, 2022-12-06:
The velocity limit defined in the URDF seems too high.

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/d73115b9f088395b62cfc0073cf1d0ca6aa21c1e/lbr_description/urdf/med7/med7_description.urdf.xacro#L10

Perhaps this is in degrees/sec? The [unit for the velocity limit should be in rad/sec](http://wiki.ros.org/urdf/XML/joint#Elements). For ease in setting joint limit constraints in optas, we should tune the velocity limit.

mhubii, 2023-02-10:
great catch indeed. The velocity limits should be extracted and tested in the same fashion as position limits, refer to https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/foxy/lbr_description/test/test_urdf.py

mhubii, 2023-03-22:
addressed in #69 #70 , to be moved to `foxy` `galactic` `noetic` `melodic`

mhubii, 2023-03-22:
addressed in #72 #73 , `noetic` and `melodic` missing

closed 2023-03-22 by mhubii