lbr-stack/lbr_fri_ros2_stack #48: Data exchange fails every second sequences, receiving a nan for joint position

MrstupidJ, 2023-01-17:
Hi, 

We are running Foxy branch in Ubuntu 20.04 real-time kernel. The robot is KUKA lbr Med14, with KUKA sunrise software version 1.15 installed on the controller.  

After building, we launched the bringup file:
`ros2 launch lbr_bringup lbr_bringup.launch.py model:=med14 sim:=false`
Then we got some message from node "[lbr_spinner-1]" and "[ros2_control_node-2]", saying that joint position is nan for joint 0. 
To further examine, we added some lines to print joint positions in each iteration. As you can see below, every second iteration the joint position bacame a nan for joint 0. 
[image]

This problem happened when we tried to include a moveit2 package (from moveit2 tutorial). However, after we removed the moveit2 package from workspace (i.e. only the lbr_fri_ros2_stack and its dependencies in workspace/src) and did a clean build, the problem still presents. 
 
Any help will be highly appreciated! Thanks.

mhubii, 2023-01-17:
hi @MrstupidJ and thanks for the feedback. This is expected behavior indeed.

It is not saying that the joint position is nan, is is saying that the joint command is nan. Until you send a first command to the `/lbr_command` topic, everything is initialized as nan.

Btw, moveit is already integrated. When you launch the system, you can add motion planning in rviz. Therefore, click on Add->MotionPlanning->Ok

Let me know if this helps

mhubii, 2023-01-17:
I am also happy to discuss this in a meeting if you'd like. So I can better understand your application and add the necessary features. Feel free to drop me a mail at martin.huber@kcl.ac.uk.

(Closing this for now because this is intended behavior, feel free to re-open)

closed 2023-01-17 by mhubii

CarlDegio, 2023-02-07:
@mhubii Hi, his problem occured may because the FRI frequency is twice as big as moveit command frequency (moveit use /position_trajectory_controller/joint_trajectory topic usually. and if FRI don't receive command at the setting time, it will transfer a zero command). 
Besides, I found another problem about this. Even the command frequency ( I use /lbr_command, so that means the timer period setting ) is equal to the FRI frequency, the FRI can't receive command sometimes. I think the possible reasons are two: ros timer frequency fluctuation, or sometimes the calculation time is too long.
Do you have some good idea about it?