lbr-stack/lbr_fri_ros2_stack #49: FRI session state changes and disposes in SmartPad

MrstupidJ, 2023-01-17:
Hi, 

Sorry to bother you guys multiple times, but we do have another problem regarding to running the [LBRServer.java](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/foxy/server/fri_ros2/LBRServer.java).  

When we run the LBRServer application on KUKA SmartPad, the FRI session state will change to "MONITORING_READY" after some time (around 60s). And soon after this the FRI connection disposes as you can see below in the screenshot. 
[image]

We are checking the FRI manual from KUKA but have not figured out why it is happening. Please let us know if you have any idea on how to increase the timeout or prevent the FRI session from getting disconnected. 

Thanks for your help!

MrstupidJ, 2023-01-25:
Thanks for your reply! We are still trying to figure out why it is happenning. The cable and connection seem fine. 

What we found so far is that, the LBRServer application can keep running if we reboot the KUKA robot (and its controller of course), which is similar to your experience. 

Btw may I ask why do you use `com.kuka.roboticsAPI.motionModel.BasicMotions.positionHold` instead of `com.kuka.roboticsAPI.motionModel.positionHold` at the line below? Are they the same?  
https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/d6d5b4eaa547eba4756add527a37a77c6208fd7c/server/fri_ros2/LBRServer.java#L3

mhubii, 2023-02-02:
Sorry for replying so late! No there is not specific reason. Will double check tomorrow. Have you used the other one before ? The Java side of the code could definitely use an improvement !

mhubii, 2023-02-03:
re lost connection. The error here is possibly this line

https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/d6d5b4eaa547eba4756add527a37a77c6208fd7c/server/fri_ros2/LBRServer.java#L157

When an error is thrown here, the error is caught and printed, but not handled. Therefore, dispose is never called. This causes subsequent connections to drop. This needs to be fixed

MrstupidJ, 2023-02-05:
> Sorry for replying so late! No there is not specific reason. Will double check tomorrow. Have you used the other one before ? The Java side of the code could definitely use an improvement !

No worries. Thanks for your hard work! 

We have tried both methods a bit but did not notice much difference. I feel like `com.kuka.roboticsAPI.motionModel.positionHold` is an older version of `com.kuka.roboticsAPI.motionModel.BasicMotions.positionHold`. Also, it (positionHold) may not be cancelled as well when the FRI connection get disposed as we specify timeout as "-1".

mhubii, 2023-02-10:
this issue should be resolved now, feel free to re-open in case in occurs again. Please make sure to update [LBRServer.java](https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/blob/foxy/server/fri_ros2/LBRServer.java) on your controller

closed 2023-02-10 by mhubii

MrstupidJ, 2023-02-13:
Many thanks! Now the FRI session can be properly disposed.