lbr-stack/lbr_fri_ros2_stack #52: Humble Support Progress

jclinton830, 2023-02-06:
Hi, 

Is humble branch building and working properly in ubuntu 22.04 or is it still under active development and testing?

mhubii, 2023-02-06:
hi @jclinton830, the `humble` branch works well, only moveit is not added therein, since moveit made some changes. There is a `dev-humble-moveit` branch where you can track the progress. This should be added this week.

mhubii, 2023-02-15:
sorry might take a little longer to finish this as there are some deadlines. Do you need this urgenlty @jclinton830 ?

jclinton830, 2023-02-16:
> sorry might take a little longer to finish this as there are some deadlines. Do you need this urgenlty @jclinton830 ?

All good. It is not urgently needed.

FrGe2016, 2023-02-21:
Very interesting. I am having a lot of problems to have humble moveit2 and gazebo working together with my home made robot. From the many exemples that i have found in the net, no one worked with humble. Please let me know when you will have a working solution.

dcconner, 2023-03-09:
I'm coming to your stack after posting a question at https://answers.ros.org/question/413181/recommendations-for-ros-2-humble-and-gazebo-demo-for-robot-arm/ 

It seems your current quick start launches only simulation and base, and I couldn't get your `ros2 run lbr_examples lbr_command_node.py` command to move anything.  

I did a rebase of humble into your dev-humble-moveit branch and it built.  Do you have any specific directions for launch with a moveit demo?  Do you have the standard MoveIt RViz setup working?

mhubii, 2023-03-20:
> Very interesting. I am having a lot of problems to have humble moveit2 and gazebo working together with my home made robot. From the many exemples that i have found in the net, no one worked with humble. Please let me know when you will have a working solution.

Hi @FrGe2016 , please check out the dev-humble-moveit branch, there is a working solution. Please also check out the universal robots, which seem to have a solution

mhubii, 2023-03-20:
> I'm coming to your stack after posting a question at https://answers.ros.org/question/413181/recommendations-for-ros-2-humble-and-gazebo-demo-for-robot-arm/ 
> 
> 
> 
> It seems your current quick start launches only simulation and base, and I couldn't get your `ros2 run lbr_examples lbr_command_node.py` command to move anything.  
> 
> 
> 
> I did a rebase of humble into your dev-humble-moveit branch and it built.  Do you have any specific directions for launch with a moveit demo?  Do you have the standard MoveIt RViz setup working?  
> 
> 
> 
> I am very interested in finding a demo that will work with MoveIt and Gazebo in Humble 

Hi @dcconner,sorry this is hardware specific and won't run in simulation. If you want to control the robot in simulation, please send joint states to the position trajectory topic. Feel free to open a new issue for missing examples.

jclinton830, 2023-03-20:
> sorry might take a little longer to finish this as there are some deadlines. Do you need this urgenlty @jclinton830 ?

@mhubii How are you going with integrating Moveit2 with this stack? Happy to help with the development if you have a clear idea of what needs to be done.

mhubii, 2023-03-21:
Hi @jclinton830 , moveit2 needs to be added to the humble and rolling branches. It is mostly about getting launch files sorted

FrGe2016, 2023-03-21:
I have found a complete set  of robots working properly with ros2 humble, moveit2 and gazebo

https://github.com/IFRA-Cranfield/ros2_RobotSimulation/tree/humble

At time of writing Panda, irb1200, irb120 are working perfectly, the transition to humble is not completed for the other ones.

mhubii, 2023-08-03:
MoveIt 2 is finally back! 🥳

Sorry for the delay, any feedback is welcome!

Feel free to re-open / add a new issue in case you find problems!

closed 2023-08-03 by mhubii