lbr-stack/lbr_fri_ros2_stack #63: Two questions about vcs command and base_interface.hpp missing

Tianhuanyu, 2023-03-06:
I find problems when i install lbr_fri_ros2_stack:

    The first problem I meet is that when i follow the steps . The command ''vcs'' is not defined but i cannot find an avaliable vcs-installing command for Ubuntu 20.04.

    Most of installing methods shows

        ``Collecting vcs
          Using cached vcs-0.4.0.tar.gz (482 kB)
            ERROR: Command errored out with exit status 1:
             command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-2_kpq018/vcs/setup.py'"'"'; __file__='"'"'/tmp/pip-install-2_kpq018/vcs/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-2_kpq018/vcs/pip-egg-info
                 cwd: /tmp/pip-install-2_kpq018/vcs/
            Complete output (6 lines):
            Traceback (most recent call last):
              File "<string>", line 1, in <module>
              File "/tmp/pip-install-2_kpq018/vcs/setup.py", line 11
                except IOError, err:
                              ^
            SyntaxError: invalid syntax
            ----------------------------------------
        ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.``

      2.When i skip the vcs installation and repos installation, I git clone the fri lib according to the repos.yml.  After colcon, the bash showed '' No such file or directory
   13 | #include "hardware_interface/base_interface.hpp"''

cmower, 2023-03-06:
Thanks @Tianhuanyu, try installing `vcstool` here: http://wiki.ros.org/vcstool

Tianhuanyu, 2023-03-06:
I find now when i try ''rosdep install --from-paths src --ignore-src -r -y''
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
lbr_description: No definition of [joint_state_publisher_gui] for OS version [focal]
lbr_bringup: No definition of [gazebo_ros2_control] for OS version [focal]
lbr_hardware_interface: No definition of [ros2_control] for OS version [focal]
lbr_moveit: No definition of [moveit] for OS version [focal]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Tianhuanyu, 2023-03-06:
should i install the corresponding libs respectively?

cmower, 2023-03-06:
Which directory are you running this command from?

cmower, 2023-03-06:
I am assuming that installing `vcstool` worked?

Tianhuanyu, 2023-03-06:
yes, vcs worked

Tianhuanyu, 2023-03-06:
I run the command in workspace of ROS2.  it looks like this in terminal ~/ros2_ws$.

cmower, 2023-03-06:
Try running `$ sudo apt update`

Tianhuanyu, 2023-03-06:
Thanks. It works!

cmower, 2023-03-06:
Great, closing this issue now. It will be fully resolved when #64 is merged.

closed 2023-03-06 by cmower