lbr-stack/lbr_fri_ros2_stack #65: file not exist tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp

Tianhuanyu, 2023-03-06:
A warning problem occured after run 

``ros2 launch lbr_bringup lbr_bringup.launch.py model:=med7 sim:=true # model:=[iiwa7/iiwa14/med7/med14]'' 

the result shows:
[rviz2-8] Warning: Invalid frame ID "lbr_link_6" passed to canTransform argument source_frame - frame does not exist
[rviz2-8]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp

Besides, robotic model and links seems not been shown in Gazebo and Rviz

Tianhuanyu, 2023-03-06:
I tried once again and then it works. I don't know why it is not available for the first time. Maybe Gazebo causes some time-delay and the program bring-up sequences is incorrect. Anyway, it works now.

mhubii, 2023-03-06:
hi @Tianhuanyu and thanks for reporting this issue. This is a common issue on first start-up. Duplicate #51, #26

Feel free to re-open

closed 2023-03-06 by mhubii