lbr-stack/lbr_fri_ros2_stack #67: FRI connection fails for ros noetic

RaquelMotzfeldtTirach, 2023-03-17:
Hello, 
First of all thank you for this repo, it looks really interesting and exciting! 

We have gone through the setup, using the noetic branch, but unfortunately encountered a few issues. We have a LBR iiwa 14, to which we have added the FRI package and the LBRserver java file in the src folder. The install and sync went well. 
We then connected our client PC which runs on ubuntu 20.04 and with ros Noetic to the controller through the two possibilities: 
- the X66 port and using 172.31.1.148 as IP, rate of 10ms
- the KONI port and using 192.170.10.1 as IP, rate of 10ms

In both cases, we are able to ping the controller but unable to create a connection. 
We first start the LBRserver application on the SmartPad (rate 10ms, Position Control) and then we run on the client PC the moveit_planning_execution.launch, but the LBRserver stops pretty quickly after not getting any response from the client PC with a "timout before FRI connection was good or excellent" error. 
We used wireshark to have a look at the UDP communication, and we can clearly see that the controller is sending packages but our client PC is not responding. 
[image]

The outcome is the same for both ports/IPs.

mhubii, 2023-03-17:
happy to chat in person if that is the case. We stopped development of the noetic branch, however, it should still work.

When you launch `roslaunch lbr_moveit moveit_planning_execution.launch model:=med7 sim:=false`, did you set `sim:=false`?

RaquelMotzfeldtTirach, 2023-03-17:
I'm sorry to hear about that, we hope this is not too much trouble then. 
Yes we did indeed change the sim argument to false.

mhubii, 2023-03-17:
can you find any errors in the terminal, ie `... | grep ERROR` for a quick check

mhubii, 2023-03-17:
the most common issue are missing dependencies

RaquelMotzfeldtTirach, 2023-03-20:
The only error I get the launch file terminal is the following: 
**[ERROR] [1679319207.644945605]: Could not find the planner configuration 'None' on the param server**

RaquelMotzfeldtTirach, 2023-03-20:
[image]
[image]

mhubii, 2023-03-20:
thanks for the feedback. Is this related to #68 ?

mhubii, 2023-03-20:
this repository is built using FRI 1.15, maybe there might be an issue

RaquelMotzfeldtTirach, 2023-03-20:
That might explain it indeed. Any idea if KUKA will let us update it? It's quite old and not supported anymore I believe

RaquelMotzfeldtTirach, 2023-03-20:
But even if the FRI version is too old, shouldn't I see in Wireshark some udp msg from my PC towards the robot controller? 
I looks like the problem is on the client PC end no?

mhubii, 2023-03-20:
it could be that from fri `1.14` -> `1.15` there is a new message field, I am trying to look this up right now

mhubii, 2023-03-20:
May I ask for your help? I can't access the cpp source code for the FRI 1.14. You can extract it through 

[image]

You should see the source code here

[image]

In the `.zip` file, check there is a license stating

```
Redistribution and use of the software in source and binary forms, with or
without modification, are permitted
```

If you could send me that zip file to `martin.huber@kcl.ac.uk`, I am happy to investigate. If this too much of a burden for you, no problem

RaquelMotzfeldtTirach, 2023-03-20:
Absolutely no problem, I'll gladly help out! Give me a minute, have to change OS

mhubii, 2023-03-20:
okay so on the client source side there are absolutely no differences

RaquelMotzfeldtTirach, 2023-03-20:
Shoot ... so something with my setup on the PC or something with the connection is not right. Might also be the iiwa itself ... this repo only uses the FRI package?

mhubii, 2023-03-20:
you can check whether the fri runs stand alone. E.g. build and install it

https://github.com/KCL-BMEIS/fri/tree/fri-1.14#cross-platform-build-and-installation

run one of the example applications

https://github.com/KCL-BMEIS/fri/tree/fri-1.14#client-side---laptop

mhubii, 2023-03-20:
the LBRJointSineOverlay sample application, by default (you'd have to check in Sunrise), sets the ip to 127.0.0.1

RaquelMotzfeldtTirach, 2023-03-20:
I'm about to test it.

RaquelMotzfeldtTirach, 2023-03-20:
Unfortunately, even now with the right IP, it does not work

RaquelMotzfeldtTirach, 2023-03-20:
I'll redo the sunrise project setup from scratch following https://github.com/KCL-BMEIS/fri/tree/fri-1.14#server-side---kuka-controller to be sure

mhubii, 2023-03-20:
sorry to hear that. The instructions might differ a little bit for your robot

RaquelMotzfeldtTirach, 2023-03-20:
Yes, we have a different media flange

RaquelMotzfeldtTirach, 2023-03-20:
Still not working unfortunately

mhubii, 2023-03-20:
Really sorry to hear that and no idea what the issue could be.

Maybe it makes sense to try and get the fri running standalone first.

There is also the iiwa_stack, which you could try.

mhubii, 2023-03-20:
I'll take this feedback seriously and will try to improve the unit testing. This particular issue has probably to do with the fri or some network settings

mhubii, 2023-03-21:
thank you! I'll forward you some FRI documentation

RaquelMotzfeldtTirach, 2023-03-22:
I think I figured it out: 
Just tried only KUKA's FRI and the connection worked as soon as I disabled my firewall! 
Still have to setup my firewall correctly and try this repo again, but looks promising!

RaquelMotzfeldtTirach, 2023-03-22:
To setup the firewall on ubuntu 20.04: 
`sudo ufw allow proto udp from 172.31.1.147 to 172.31.1.148 port 30200`

RaquelMotzfeldtTirach, 2023-03-22:
It all works fine now, thank you for your help !