lbr-stack/lbr_fri_ros2_stack #68: LBRServer Error

m1541, 2023-03-19:
Hi,

I'm tying to set up the server on the real robot but I'm unable to create a connection. Has anyone encountered this error before?

[image]

mhubii, 2023-03-20:
hello @m1541 , and thanks for the feedback.

In the image, you set the client command mode to `NO_COMMAND_MODE`. Could you for example please try to set it to position command instead?

mhubii, 2023-03-20:
hi @m1541 could you please try and restart the robot?

mhubii, 2023-03-20:
On the other side, it doesn't seem a connection is established. Can you double check that your pc's IP is set to `172.31.1.3`?

mhubii, 2023-03-22:
Could you please check if this is a firewall related issue, refer to https://github.com/KCL-BMEIS/lbr_fri_ros2_stack/issues/67#issuecomment-1479794623

mhubii, 2023-05-08:
Any updates @m1541 ?

mhubii, 2023-08-06:
closing due to inactivity

closed 2023-08-06 by mhubii