lbr-stack/lbr_fri_ros2_stack #92: Release mode build

mhubii, 2023-05-29:
Might need to add cmake flag for release build or set a default build unless specified, ie `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`

tarunsamanta2k20, 2023-05-29:
@mhubii where do I have to add?

mhubii, 2023-06-07:
sorry for the late reply! Just saw it. I came across it somewhere else and realized this might be an issue here, too.

You might have to add this during the build, ie when calling `colcon build`

tarunsamanta2k20, 2023-06-07:
> sorry for the late reply! Just saw it. I came across it somewhere else and realized this might be an issue here, too.
> 
> You might have to add this during the build, ie when calling `colcon build`

May I know that file name or location where I have to make changes.

mhubii, 2023-06-09:
when you build this repository, you might have to add this as a flag to the build. No file needs changing, but one might add a default build type to the cmake files for the hardware interface and the fri

closed 2023-06-19 by mhubii