lbr-stack/lbr_fri_ros2_stack #104: offline build option?

CarlDegio, 2023-08-08:
I found new version of SDK has introduced fri_vendor to auto download fri package when build. However, this leads to the PC must link to internet when build. But it's hard when PC link to the kuka.

So is there a method to check fri_vendor already download fri so cancel network connection.

mhubii, 2023-08-08:
oh thank you for raising this issue. We did introduce this to have a single source for the FRI, see https://github.com/lbr-stack/lbr_fri_ros2_stack/pull/85#issuecomment-1540553910

we will try to add this asap, sorry for the inconvenience

CarlDegio, 2023-08-08:
I use humble branch, version 1.2.3. Now use topics mainly.

mhubii, 2023-08-08:
do you run through python or cpp?

CarlDegio, 2023-08-08:
Yes, I use both of python and cpp. About this I have some thinking. 

The FRI interface use the fixed frequency. If the interval is comed but ROS driver doesn't receive the further instructions, the FRI will make robot stop to prevent uncommanded movement just like emergency stop. 
However, if the controller thread slower than commanding thread, the commanding thread will receive command with interval. Just like emergency stop and motion switch regularly. At the high frequency, this lead to noise. The most notable example is ROS driver(lbr_bringup) run with 100Hz and command topic publish topic with 50Hz. The motion state will change every 10ms (move->stop->move->stop...).
At the same time, the frequency fluctuation also lead to a tick noise. Because they are not synchronized precisely. Maybe some small motion buffer would be useful.

Besides, I'm looking for end-effort control method using this package. Do you have some good demo? I tried using kinpy but it's a bit slow to solve (about 10Hz).

CarlDegio, 2023-08-08:
> oh thank you for raising this issue. We did introduce this to have a single source for the FRI for python bindings https://github.com/lbr-stack/pyFRI, see [#85 (comment)](https://github.com/lbr-stack/lbr_fri_ros2_stack/pull/85#issuecomment-1540553910)
> 
> we will try to add this ASAP, sorry for the inconvenience.
> 
> Maybe in the meantime build via `colcon build --packages-select name_of_the_package_you_want_to_re_compile`, so only your package gets compiled.
> 
> May I ask:
> 
> * what branch you are on @CarlDegio ?
> * are you using this stack through `MoveIt`, `ros2_control`, or `topics`?

Actually I also used moveit2 (servo) and want to accomplish an end-effort realtime control. But I found the slowly solution and some bug on foxy moveit2 servo so I give up it. It also has some frequency problem.

But the trajectory planning motion on moveit2 works well. Only hard to move in real time.

mhubii, 2023-08-08:
> ).

yes exactly the problem, this will be fixed tomorrow!

mhubii, 2023-08-08:
> The motion state will change every 10ms (move->stop->move->stop...).

This will be fixed tomorrow

mhubii, 2023-08-09:
okay if you pull the latest changes you will notice smooth robot behavior even for an asynchronous control loop.

The joint names are now `A1`,... `A7`, there used to be a robot name prefix

There are some other changes which shouldn't affect you. I'll try to fix the build issue but might take some time.

CarlDegio, 2023-08-09:
> okay if you pull the latest changes you will notice smooth robot behavior even for an asynchronous control loop.
>
> The joint names are now `A1`,... `A7`, there used to be a robot name prefix
>
> There [...]

Thanks for your contribution! I believe lbr_fri_ros2_stack is expected to be the best choice for kuka robots. Now many students in our laboratory are using this package!

CarlDegio, 2023-08-19:
> 
> Besides, I'm looking for end-effort control method using this package. Do you have some good demo? I tried using kinpy but it's a bit slow to solve (about 10Hz).

I write a end control node with reference to demo advance c++, link here:https://github.com/CarlDegio/lbr_end_control.git

mhubii, 2023-09-07:
Very cool, what does it do? Happy to add this to the demos if you think this could be of general interest

mhubii, 2024-05-13:
this problem is fixed with #147. The FRI source is now downloaded before building from https://github.com/lbr-stack/fri.