lbr-stack/lbr_fri_ros2_stack #106: lbr_bring_up doesn't publish /lbr/state?

CarlDegio, 2023-08-18:
In the past, the launch file bringup.launch.py will publish /lbr/command and /lbr/state topics. But now I found it doesn't. Is this normal?

And now the cpp demo doesn't need to launch bringup.launch.py cause the app_node is used(Inconsistent with README).

mhubii, 2023-08-19:
sorry for the confusion.

The core is always the App https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/lbr_fri_ros2/include/lbr_fri_ros2/app.hpp. It reads commands from `/robot-name/command` and sends states to `/robot-name/state`.

The bringup launches the hardware interface and sets the `robot-name` to `_robot-name`, hence topics still exist, but are hidden.

CarlDegio, 2023-08-22:
Thank you! I got that. In addition, I would like to ask  ros2 control interface is expected to use with lbr_bringup currently? Since the users interactive kuka with jointstate.

The current readme does not mention the difference between app and lbr_bringup. Only topic difference in demo: https://lbr-fri-ros2-stack-doc.readthedocs.io/en/humble/lbr_fri_ros2_stack/lbr_demos/doc/lbr_demos.html and https://lbr-fri-ros2-stack-doc.readthedocs.io/en/humble/lbr_fri_ros2_stack/lbr_demos/lbr_demos_ros2_control_cpp/doc/lbr_demos_ros2_control_cpp.html

mhubii, 2023-08-22:
sorry not sure I got the question, correctly. 

Most ROS users are familiar with the hardware interface, which is launched in the `lbr_bringup` package. For research, one can also launch `lbr_fri_ros2` standalone.

There is some documentation on the bringup here https://lbr-fri-ros2-stack-doc.readthedocs.io/en/humble/lbr_fri_ros2_stack/lbr_bringup/doc/lbr_bringup.html

It should probably clarified in the links you sent.

CarlDegio, 2023-08-22:
Oh, i just noticed it. In fact, I think this is caused by the experience of using this library before XD. Maybe it will be easier for a new user to understand.

mhubii, 2023-08-22:
You raised a couple of valid points and we will try to address them, so thank you very much @CarlDegio !

mhubii, 2023-10-26:
this will be added in #130

mhubii, 2023-12-09:
this issue is now fully resolved. 

System can for example be launched via

```shell
ros2 launch lbr_bringup bringup.launch.py model:=iiwa7 sim:=false ctrl:=forward_lbr_position_command_controller
```

This will spawn:
- joint state broadcaster (`/lbr/joint_states`)
- lbr state broadcaster (`/lbr/state`)
- forward lbr position command controller (`/lbr/command/join_position`)

This will also spawn RViZ. More documentation is on the way.

closed 2023-12-09 by mhubii