lbr-stack/lbr_fri_ros2_stack #115: Torque and wrench overlay demo do not work

WZHwerk5, 2023-09-06:
Hi, I have tried demos in lbr_demos_fri_ros2_cpp, the JointSineOverlay works smoothly but wrench and torque not. After launch there is no error/warning message from both ROS2 side and FRI side, and I can even see the command topic with values in rqt, but on our iiwa7, nothing happens. Need your help!

Thanks in advance!

WZHwerk5, 2023-09-07:
I've also tried the FRI-Client-SDK_Cpp, and wrench/torque overlay both worked.

mhubii, 2023-09-07:
hi @WZHwerk5 and thank you for the feedback. May I ask what procedure you followed?

WZHwerk5, 2023-09-07:
I followed the guidance here[https://lbr-fri-ros2-stack-doc.readthedocs.io/en/humble/lbr_fri_ros2_stack/lbr_demos/lbr_demos_fri_ros2_cpp/doc/lbr_demos_fri_ros2_cpp.html](url)

WZHwerk5, 2023-09-07:
Yes, the FRI connection can be established. Both smart pad and my Ubuntu terminal have no error/warning message.

mhubii, 2023-09-07:
okay, the issue is probably the QoS policy

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/c3766c61047382a1b39dd9dacd7e163f4b5ff66b/lbr_demos/lbr_demos_fri_ros2_cpp/src/torque_sine_overlay_node.cpp#L25

May have gone out of date. I'll test and be back in half an hour

WZHwerk5, 2023-09-07:
Actually I have manually set it as 1ms, but still does not work...

mhubii, 2023-09-07:
so in `LBRWrenchSineOverlay.java` and `LBRTorqueSineOverlay.java` you'll find 

```java
friConfiguration.setSendPeriodMilliSec(5);
```
The `lbr_fri_ros2` package therefore creates pub/sub with 5 ms QoS deadline.

To fix:
- change deadlines to 5 ms:

pub: https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/c3766c61047382a1b39dd9dacd7e163f4b5ff66b/lbr_demos/lbr_demos_fri_ros2_cpp/src/torque_sine_overlay_node.cpp#L25

sub: https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/c3766c61047382a1b39dd9dacd7e163f4b5ff66b/lbr_demos/lbr_demos_fri_ros2_cpp/src/torque_sine_overlay_node.cpp#L32

- re-compile: 
```bash
colcon build --packages-select lbr_demos_fri_ros2_cpp --cmake-args -DCMAKE_BUILD_TYPE=Release
```

mhubii, 2023-09-07:
To be fixed in #116

WZHwerk5, 2023-09-07:
Thank you so much!!

mhubii, 2023-09-07:
Np. Note, when you push too hard against the robot in you might see the connection drop. This has to do with the command guard and how limits are checked. This needs fixing

WZHwerk5, 2023-09-07:
Okay I see. By the way, is it possible to dynamically change the impedance values (stiffness and damping) of Cartesian impedance mode?

mhubii, 2023-09-07:
no sorry, the FRI doesn't support that. You'd have to restart the FRI each time. That's on KUKA

WZHwerk5, 2023-09-13:
I see, sorry but another question is: I want to control the robot joint position together with wrench, but when I launched the bringup in wrench command mode, the robot started to move suddenly with high velocity and was quickly stopped because it is not on the planned path, according to FRI message. Do you know why? Many thanks.

mhubii, 2023-09-15:
In wrench command mode the FRI takes target wrench and target position as input. Maybe you set the target position to zero?

WZHwerk5, 2023-09-15:
Thanks, it's very clear. But actually I was using the default bringup launch file `bringup.launch.py` with `moveit:=true`, so the target position should not be zero or not set, am I right?

WZHwerk5, 2023-09-27:
Or do you think that in this case I must send eg. ` ipo_joint_position ` additionally as a target command?

mhubii, 2023-09-28:
Ah okay I understand. Sorry for the late reply.

So on the smartPad you chose: 

Control mode: Cartesian Impedance
Command mode: Position?

WZHwerk5, 2023-10-10:
On the smartPad I chose:

Control mode: Cartesian Impedance
Command mode: Wrench

What I want is to use the bringup with Moveit, then add a wrench overlay, so that my iiwa can execute surface processing task like polishing or sanding by planning defined paths. What do you think on it? Thanks!

mhubii, 2024-05-13:
Sorry @WZHwerk5 , been caught up. Will investigate this now. Thank you for sharing this issue

WZHwerk5, 2024-06-13:
Hi, I already turned the wrench command controller into a chained controller, now should I launch the bringup with wrench command controller in wrench command mode first,  then load the joint trajectory controller from another launch file?

mhubii, 2024-06-14:
wow sounds awesome @WZHwerk5! Would you want to open a pull request? I personally can't answer the question as I haven't investigated

jomon250, 2025-03-04:
@WZHwerk5 did you find a solution for the problem?