lbr-stack/lbr_fri_ros2_stack #120: Troubles finding a connection to Kuka iiwa 14 controller

jomonn, 2023-09-26:
Hello, I am having troubles connecting to my kuka controller, my HMI shows the following dialouge:
[image]

In addition, I get the warning that I have to use Real Time Kernel - is this the reason? I tried it with RT Kernel but than the FIFO scheduling didn't work out.

[image]

Do I have to set the IP of the kuka controller anywhere in the code? Or how does the connection work?

jomonn, 2023-09-28:
thank you for your quick response, that worked perfectly!

mhubii, 2023-09-28:
wow I would love that !