lbr-stack/lbr_fri_ros2_stack #124: Simulation unable to launch

EliasTDam, 2023-10-05:
Me and my group are using this package. They installed it a couple of weeks ago, and have no issues whatsoever. I however, have tried installing it since yesterday, on a fresh install of Ubuntu and ROS2 Humble, and can't get the simulation to launch. It is from the quick start guide. I try to run the command:

_source install/setup.bash
ros2 launch lbr_bringup bringup.launch.py \
    model:=iiwa7 # [iiwa7, iiwa14, med7, med14] \
    sim:=true # [true, false] \
    rviz:=true # [true, false] \
    moveit:=true # [true, false]_
    
But I get the error:

_[INFO] [launch]: All log files can be found below /home/elias/.ros/log/2023-10-05-10-50-39-318455-ROS2-Host-50919
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): LBRMoveGroupMixin.moveit_configs_builder() missing 1 required positional argument: 'base_frame'
sim:=true: command not found
rviz:=true: command not found
moveit:=true: command not found_

Is this an issue that has been introduced lately, or is something wrong on my side?

mhubii, 2023-10-05:
Linking #123

EliasTDam, 2023-10-05:
Thank you very much! I'll definitely try that then

mhubii, 2023-10-06:
Please re-open in case it doesn't solve the issue and thank you for the feedback !

EliasTDam, 2023-10-06:
The simulation now launches, so it seems to have resolved the issue. Thanks!

mhubii, 2023-10-12:
Re-open for visibility until merge of #123

mhubii, 2023-10-18:
done

closed 2023-10-18 by mhubii