lbr-stack/lbr_fri_ros2_stack #125: Incorrect link `lbr_link_ee`, it appears inside the mesh.

ardiya, 2023-10-16:
Dear Maintainers,

I'm currently using the [noetic](https://github.com/lbr-stack/lbr_fri_ros2_stack/tree/noetic) branch, and I am having troubles with the iiwa14.urdf as the `lbr_link_ee` appears inside the mesh, as shown in the image below
 
[image]

I'm expecting the `lbr_link_ee` to appears at the end effector tip.

I tried comparing it to iiwa_stack and noticed the https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_description/urdf/iiwa14.xacro#L332-L336 xyz="0 0 0.045" as opposed to xyz="0 0 0.035" in lbr_fri_ros_stack.

Is there any reason behind this?

mhubii, 2023-10-17:
hi @ardiya, thank you for the feedback. These values are obtained from the CAD model.

Interesting to see there is a discrepancy. When launching the `humble` branch, it seems correct:

[image]

There, the distance is also `0.035`. Not sure what causes the issue.

mhubii, 2023-10-17:
Hi @ardiya thank you for the feedback. I think you are right, all other values were obtained from the CAD model but this link.

Will fix!

mhubii, 2023-10-17:
`0.045` doesn't seem to be right though

ardiya, 2023-10-18:
Yeah, that's true. Your original `0.035` is the same with the technical drawing of the basic flange though

[image]
Based on the measurement in section 4.2.2 of https://www.oir.caltech.edu/twiki_oir/pub/Palomar/ZTF/KUKARoboticArmMaterial/Option_Media_Flange_en.pdf

mhubii, 2023-10-18:
thank you for sharing, working on fixing this

mhubii, 2023-10-19:
could you please pull the latest `noetic` changes and report if it works for you? Thank you again for reporting this issue.

ardiya, 2023-10-24:
Thanks @mhubii! It works great. Quick question: how do you get the CAD models?
I have different flange and it's difficult to get the Mesh

closed 2023-10-24 by ardiya

mhubii, 2023-10-25:
the CAD models are available on xpert.kuka.com , do you want me to add another model for you that has the correct flange?

ardiya, 2023-10-25:
It's okay, I was able to find `LBR iiwa 14 R820 with media flanges` and able to download it.

@mhubii Do you you http://wiki.ros.org/urdf_from_step to convert step to urdf or is there any package that you recommend to convert it to URDF+mesh file?

mhubii, 2023-11-11:
Hi @ardiya sorry for coming back so late. Personally, I had access to autodesk fusion 360 at the time and used the URDF exporter https://github.com/syuntoku14/fusion2urdf

commit b0b4abcf (2023-10-17): ee fix #125
commit 500a70f7 (2023-10-17): ee fix #125
commit acf6ac69 (2023-10-17): ee fix #125
commit 416559db (2023-10-17): ee fix #125