lbr-stack/lbr_fri_ros2_stack #128: 4 packages had stderr output: fri_vendor lbr_demos_fri_ros2_advanced_python lbr_demos_fri_ros2_python lbr_demos_ros2_control_python

Tianhuanyu, 2023-10-23:
Hi, when I install lbr_fri_ros2_stack, I find packages had stderr output including fri_vendor lbr_demos_fri_ros2_advanced_python lbr_demos_fri_ros2_python lbr_demos_ros2_control_python.

Here is my environment:
Python: 3.10.12
ROS version: Humble

-----------------------------------------------------------------------------------------------------------------------------------

Here is the Error Message

--- stderr: lbr_demos_fri_ros2_advanced_python
/home/thy/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************
---------------------------------------------------------------------------------------------------------------------

It seems to fail to install KDL, showing the message after vcs install:

lbr_demos_fri_ros2_advanced_cpp: Cannot locate rosdep definition for [orocos_kdl]

-----------------------------------------------------------------------------------------------------------------------

Tianhuanyu, 2023-10-23:
Try to install orocos using
sudo apt-get install ros-humble-orocos*
and
sudo apt-get install ros-humble-python-orocos*

They didn't work.

Tianhuanyu, 2023-10-23:
It works after giving: 
pip install setuptools==58.2.0

Now this message is gone:
--- stderr: lbr_demos_fri_ros2_advanced_python
/home/thy/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.

But still have stderr on fri_vendor

Tianhuanyu, 2023-10-23:
Refering to This issue:  https://github.com/ros2/ros2/issues/1310

It shows orocos_kdl is removed in favor of orocos_kdl_vendor.

Tianhuanyu, 2023-10-23:
To skip orocos-dependt part,  try:

`
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --symlink-install --packages-skip lbr_demos_fri_ros2_advanced_python
`
but still meet this error

--- stderr: fri_vendor                                            
HEAD is now at 785a8d8 updated logo
Found nanopb of version: 0.2.8
Found FRIClient of version: 1.15
---

Tianhuanyu, 2023-10-23:
It seems to work though having errors in fri_vendor.
So this is fine if we only see the stderr.
You can treat this as a warning.(Tested from my side.)

mhubii, 2023-10-26:
> Refering to This issue: [ros2/ros2#1310](https://github.com/ros2/ros2/issues/1310)
> 
> It shows orocos_kdl is removed in favor of orocos_kdl_vendor.

The orocos dependency requires an update in:

- CMake: https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/f4bed49acd33ed4a8c954638eff83d2f76296dff/lbr_demos/lbr_demos_fri_ros2_advanced_cpp/CMakeLists.txt#L21

- package.xml https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/f4bed49acd33ed4a8c954638eff83d2f76296dff/lbr_demos/lbr_demos_fri_ros2_advanced_cpp/package.xml#L20

For the advanced_cpp demo https://github.com/lbr-stack/lbr_fri_ros2_stack/tree/humble/lbr_demos/lbr_demos_fri_ros2_advanced_cpp

closed 2023-11-21 by mhubii