lbr-stack/lbr_fri_ros2_stack #131: Missing documentation on multi-robot setups

TheMipmap, 2023-10-31:
Lack of documentation regarding the implementation of multi-robot setups.

mhubii, 2023-11-02:
thanks for sharing this issue @TheMipmap .

So in order to do a multi-robot setup, it is necessary to 

- Configure a unique port id for each robot in the java application
- Configure a unique IP address for each robot

Once that's done, one can connect the robots and the controlling PC to a switch. Launch files can then e.g. be used as follows:

```shell
source install/setup.bash
ros2 launch lbr_bringup bringup.launch.py \
    model:=iiwa7 # [iiwa7, iiwa14, med7, med14] \
    sim:=false \
    robot_name:=robot1 \
    port_id:=30200 # assuming port was set to 30200
```

The same launch file can then be re-run with a different robot name / port id. The transforms will be pre-fixed, so, in rviz, to display the robot correctly, one has to set a tf prefix in the RobotModel.

Multi-robot setups with moveit are currently not supported in this stack, but I haven't found a good way to do this through moveit yet.

mhubii, 2023-11-02:
This issue requires:

- [ ] How to set port id documentation
- [ ] How to IP documentation
- [ ] How to launch documentation

EliasTDam, 2023-11-13:
> thanks for sharing this issue @TheMipmap .
> 
> So in order to do a multi-robot setup, it is necessary to
> 
> * Configure a unique `port id` for each robot in the java application
> * Configure a unique `IP address` for each robot
> 
> Once that's done, one can connect the robots and the controlling PC to a switch. When using this stack, launch files can then e.g. be used as follows:
> 
> ```shell
> ros2 launch lbr_bringup bringup.launch.py \
>     model:=iiwa7 # [iiwa7, iiwa14, med7, med14] \
>     sim:=false \
>     robot_name:=robot1 \
>     port_id:=30200 # assuming port was set to 30200
> ```
> 
> or (depending what is used)
> 
> ```shell
> ros2 launch lbr_fri_ros2 app.launch.py model:=iiwa7 robot_name:=robot1 port_id:=30200
> ```
> 
> The same launch file can then be re-run with a different `robot_name` / `port_id`. The transforms will be pre-fixed, so, in rviz, to display the robot correctly, one has to:
> 
> * Configure the `Description Topic` to `/robot_name/robot_description`
> * Set a `TF prefix` in the `RobotModel` to `robot_name`
> 
> See below:
> 
> [image]
> 
> Multi-robot setups with moveit are currently not supported in this stack, but I haven't found a good way to do this through moveit yet. One interesting way to investigate multi-robot setups with moveit is the namespace tag in xacro [ros/xacro#305](https://github.com/ros/xacro/issues/305).

Hey! How/where do we set the port number? I am looking through LBRServer.java, and cannot find anything related to the port number.

mhubii, 2023-11-13:
Hi @EliasTDam , in the java application, you can configure the FRI connection.

```java
fri_configuration_ = FRIConfiguration.createRemoteConfiguration(lbr_, client_name_);
fri_configuration_.setSendPeriodMilliSec(send_period_);
fri_configuration_.setPortOnRemote(30201);  // add this line and set desired port ID
```

EliasTDam, 2023-11-14:
Thanks @mhubii ! We'll continue to keep you posted on our progress.