lbr-stack/lbr_fri_ros2_stack #132: Use of tf2 trees

Shenwei1234, 2023-10-31:
Do you use tf2 for managing all the coordinate frames, and if so how?

mhubii, 2023-10-31:
Hi @Shenwei1234 and thank you for the response. When you run the `lbr_bringup` or `lbr_ros2_control` packages, e.g.

```shell
ros2 launch lbr_bringup bringup.launch.py # model specifications / simulation etc.
```

The `joint_state_publisher` and the `robot_state_publisher` nodes will be launched. The `robot_state_publisher` publishes using `tf2`. Hope this answers the question. Let me know if this helps.

Shenwei1234, 2023-10-31:
Thanks for the reply:). If I understand correctly, the joint_state_publisher publishes the robot's joint states using tf2. But to which topic does it publish to, and who is subscribed/listener to the published joint states? Also, what's the difference between joint_state_publisher and robot_state_publisher? What does robot_state_publisher do?

mhubii, 2023-11-01:
no problem. So the `joint_state_publisher` is part of `ros2_control`. It reads joint states from the hardware and publishes under `/joint_states`.

The `robot_state_publisher` subscribes to `/joint_states`. Given the URDF and joint states, the `robot_state_publisher` computes kinematics and publishes the transforms to `/tf`.

Shenwei1234, 2023-11-02:
I understand now. Thanks again:)

mhubii, 2023-11-02:
okay great, hopefully it is okay to close this issue for now? Please feel free to ask any other questions or re-open.

closed 2023-11-02 by mhubii