lbr-stack/lbr_fri_ros2_stack #133: MoveIt

TobiasAxelsen, 2023-11-02:
Hello

I am looking into using the stack to control my kuka iiwa robot.
I'm also thinking that I could use MoveIt to help me move the robot, and also in aiding in grasping of objects, is this something that you've experience in, or what methods have you seen work using this stack?

Best regards.

mhubii, 2023-11-02:
hi @TobiasAxelsen , once you finished setting things up (see for example https://lbr-fri-ros2-stack-doc.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_fri_ros2_stack/doc/robot_setup.html), you can launch moveit to control the robot via

```shell
ros2 launch lbr_bringup bringup.launch.py \
    model:=iiwa7 # [iiwa7, iiwa14, med7, med14] \
    sim:=true # [true, false] \
    rviz:=true # [true, false] \
    moveit:=true # [true, false]
```

personally, I have no experience with adding a gripper, so if you could share your insights that would be great.

TobiasAxelsen, 2023-11-03:
Hello @mhubii Mhubii, appreciate the answer!

Which topic is it that moveit subscribes to, to gain the information of which positions to go to.

Best regards.

mhubii, 2023-11-05:
I believe moveit implements an action server but they also have a Python api. Is that what you meant? Can try to write a demo

TobiasAxelsen, 2023-11-05:
That is exactly what I meant actually :)

TobiasAxelsen, 2023-11-07:
Hello again @mhubii .

I dug a bit more around in your documentation, and saw that you've implemented ros-control, but only for joint positions. 
Maybe I am just unable to find it, but have you guys also made ros-control by the use of cartesian coordinates, such that you can just give the robot a pose that you wish to move to?

Best regards.

mhubii, 2023-11-10:
hi @TobiasAxelsen , so `ros2_control` doesn't provide this functionality (as far as I am aware of). The joint trajectory controller in `ros2_control` just provides a smooth interpolation for going from `q0` to `q1`. It is the user's responsibility to compute the target `q1`. This can e.g. be done using path planning, that is what moveit does.

mhubii, 2023-11-10:
A custom example for interfacting the moveit action server is this

```python
#!/usr/bin/python3
from typing import List

import rclpy
from geometry_msgs.msg import Pose
from moveit_msgs.action import MoveGroup
from moveit_msgs.msg import (
    BoundingVolume,
    Constraints,
    OrientationConstraint,
    PositionConstraint,
)
from rclpy.action import ActionClient
from rclpy.node import Node
from shape_msgs.msg import SolidPrimitive
from std_msgs.msg import Header
from geometry_msgs.msg import Quaternion, Point

import csv

class MoveGroupActionClientNode(Node):
    def __init__(self, node_name: str) -> None:
        super().__init__(node_name)

        self.declare_parameters(
            namespace="",
            parameters=[
                ("move_group_action_server_name", "/move_action"),
                ("move_group_name", "med7_arm"),
                ("base", "world"),
                ("end_effector", "lbr_link_ee"),
            ],
        )
        self.move_group_action_server_name = (
            self.get_parameter("move_group_action_server_name")
            .get_parameter_value()
            .string_value
        )
        self.move_group_name = (
            self.get_parameter("move_group_name").get_parameter_value().string_value
        )
        self.base = self.get_parameter("base").get_parameter_value().string_value
        self.end_effector = (
            self.get_parameter("end_effector").get_parameter_value().string_value
        )

        self.move_group_action_client = ActionClient(
            self, MoveGroup, self.move_group_action_server_name
        )

        self.get_logger().info(
            f"Waiting for action server {self.move_group_action_server_name}..."
        )
        if not self.move_group_action_client.wait_for_server(timeout_sec=1):
            raise RuntimeError(
                f"Couldn't connect to action server {self.move_group_action_server_name}."
            )
        self.get_logger().info(f"Done.")

    def send_goal_async(self, target: Pose):
        goal = MoveGroup.Goal()
        goal.request.allowed_planning_time = 1.0
        goal.request.goal_constraints.append(
            Constraints(
                position_constraints=[
                    PositionConstraint(
                        header=Header(frame_id=self.base),
                        link_name=self.end_effector,
                        constraint_region=BoundingVolume(
                            primitives=[SolidPrimitive(type=2, dimensions=[0.0001])],
                            primitive_poses=[Pose(position=target.position)],
                        ),
                        weight=1.0,
                    )
                ],
                orientation_constraints=[
                    OrientationConstraint(
                        header=Header(frame_id=self.base),
                        link_name=self.end_effector,
                        orientation=target.orientation,
                        absolute_x_axis_tolerance=0.001,
                        absolute_y_axis_tolerance=0.001,
                        absolute_z_axis_tolerance=0.001,
                        weight=1.0,
                    )
                ],
            )
        )
        goal.request.group_name = self.move_group_name
        goal.request.max_acceleration_scaling_factor = 0.1
        goal.request.max_velocity_scaling_factor = 0.01
        goal.request.num_planning_attempts = 1

        return self.move_group_action_client.send_goal_async(goal)

def load_poses(file: str) -> List:
    data = []
    with open(file, "r") as f:
        reader = csv.reader(f)
        for row in reader:
            row = [float(xi) for xi in row]
            data.append(row)
    return data

def main(args: List = None) -> None:
    rclpy.init(args=args)
    move_group_action_client_node = MoveGroupActionClientNode(
        "move_group_action_client_node"
    )

    poses = load_poses("/tmp/trajectory_replay_ws/src/trajectory_replay/poses.csv")

    # load list
    for pose in poses:
        pose = Pose(
            position=Point(x=pose[0], y=pose[1], z=pose[2]),
            orientation=Quaternion(x=pose[3], y=pose[4], z=pose[5], w=pose[6]),
        )
        future = move_group_action_client_node.send_goal_async(pose)
        rclpy.spin_until_future_complete(
            move_group_action_client_node, future
        )  # gets stuck for invalid goals

    rclpy.shutdown()

if __name__ == "__main__":
    main()

```

TobiasAxelsen, 2023-11-13:
Hello @mhubii.

I created the copy pasted the code you gave me, and put it in a file called move_robot.py

I then put this file under ~/ros_ws/src

When I open the simulation, and then run the move_robot.py file I am told that 

"tobias@tobias-Yoga-Slim-7-14ARE05:~/ros_ws/src$ python3 move_robot.py 
[INFO] [1699862721.227189770] [move_group_action_client_node]: Waiting for action server /lbr/move_action...
[INFO] [1699862721.227765550] [move_group_action_client_node]: Done."

I'd assume that this would have moved the simulation, but when I look in the simulation, nothing has moved.
Best regards.

mhubii, 2023-11-13:
Hi @TobiasAxelsen , yes, you can execute the above code in simulation. You can launch

```shell
ros2 launch lbr_bringup bringup.launch.py \
    model:=iiwa7 \
    sim:=true \
    moveit:=true
```

then run the above script

mhubii, 2023-11-13:
> tobias@tobias-Yoga-Slim-7-14ARE05:~/lbr-stack$ python3 move_robot.py 
> [INFO] [1699866221.474760477] [move_group_action_client_node]: Waiting for action server /lbr/move_action...
> [INFO] [1699866221.475212626] [move_group_action_client_node]: Done.

This output is expected. It just informs that the action client established a connection to the action server.

mhubii, 2023-11-13:
This repository clearly misses documentation on how to use it through the python moveit API

mhubii, 2023-11-13:
> [move_group-9] [ERROR] [1699866222.485828873] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - arm/arm: Unable to sample any valid states for goal tree

This error suggests that the ompl path planner had some error, maybe that is what causes the fail?

TobiasAxelsen, 2023-11-14:
> A custom example for interfacting the moveit action server is this. Create a file `move_robot.py`, copy paste this:
> 
> ```python
> #!/usr/bin/python3
> from typing import List
> 
> import rclpy
> from geometry_msgs.msg import Point, Pose, Quaternion
> from moveit_msgs.action import MoveGroup
> from moveit_msgs.msg import (
>     BoundingVolume,
>     Constraints,
>     OrientationConstraint,
>     PositionConstraint,
> )
> from rclpy.action import ActionClient
> from rclpy.node import Node
> from shape_msgs.msg import SolidPrimitive
> from std_msgs.msg import Header
> 
> 
> class MoveGroupActionClientNode(Node):
>     def __init__(self, node_name: str) -> None:
>         super().__init__(node_name)
> 
>         self.action_server = "/lbr/move_action"
>         self.move_group_name = "arm"
>         self.base = "link_0"
>         self.end_effector = "link_ee"
> 
>         self.move_group_action_client = ActionClient(
>             self, MoveGroup, self.action_server
>         )
> 
>         self.get_logger().info(f"Waiting for action server {self.action_server}...")
>         if not self.move_group_action_client.wait_for_server(timeout_sec=1):
>             raise RuntimeError(
>                 f"Couldn't connect to action server {self.action_server}."
>             )
>         self.get_logger().info(f"Done.")
> 
>     def send_goal_async(self, target: Pose):
>         goal = MoveGroup.Goal()
>         goal.request.allowed_planning_time = 1.0
>         goal.request.goal_constraints.append(
>             Constraints(
>                 position_constraints=[
>                     PositionConstraint(
>                         header=Header(frame_id=self.base),
>                         link_name=self.end_effector,
>                         constraint_region=BoundingVolume(
>                             primitives=[SolidPrimitive(type=2, dimensions=[0.0001])],
>                             primitive_poses=[Pose(position=target.position)],
>                         ),
>                         weight=1.0,
>                     )
>                 ],
>                 orientation_constraints=[
>                     OrientationConstraint(
>                         header=Header(frame_id=self.base),
>                         link_name=self.end_effector,
>                         orientation=target.orientation,
>                         absolute_x_axis_tolerance=0.001,
>                         absolute_y_axis_tolerance=0.001,
>                         absolute_z_axis_tolerance=0.001,
>                         weight=1.0,
>                     )
>                 ],
>             )
>         )
>         goal.request.group_name = self.move_group_name
>         goal.request.max_acceleration_scaling_factor = 0.1
>         goal.request.max_velocity_scaling_factor = 0.01
>         goal.request.num_planning_attempts = 1
> 
>         return self.move_group_action_client.send_goal_async(goal)
> 
> 
> def main(args: List = None) -> None:
>     rclpy.init(args=args)
>     move_group_action_client_node = MoveGroupActionClientNode(
>         "move_group_action_client_node"
>     )
> 
>     pose = Pose(
>         position=Point(x=0.0, y=0.0, z=1.0),
>         orientation=Quaternion(x=0.0, y=0.0, z=0.0, w=1.0),
>     )
>     future = move_group_action_client_node.send_goal_async(pose)
>     rclpy.spin_until_future_complete(
>         move_group_action_client_node, future
>     )  # gets stuck for invalid goals
> 
>     rclpy.shutdown()
> 
> 
> if __name__ == "__main__":
>     main()
> ```
> 
> Launch
> 
> ```shell
> ros2 launch lbr_bringup bringup.launch.py model:=iiwa7 sim:=true moveit:=true base_frame:=world
> ```
> 
> Run
> 
> ```shell
> python3 move_robot.py
> ```
> 
> Action servers can be a little difficult to deal with in the beginning, but you can read a bit about them here https://docs.ros.org/en/humble/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html

Hello.

By the use of this code, is it possible to have the robot move in cartesian space, instead of joint space? I am looking to move in cartesian space, as we find that this often creates a bit more predictable movement, and thus allows for less space for the robot to be used.

mhubii, 2023-11-14:
Hi @TobiasAxelsen , fair question. So moveit has a variable called "Use Cartesian Path". Surely that is accessible somewhere through the action server, but I didn't check

TobiasAxelsen, 2023-11-21:
Hi @mhubii. When using the MoveIt GUI, and checking the variable "Use Cartesian Path", and then planning & executing a trajectory, Rviz will immediately crash, and I am provided with the following output in my terminal, from the launch command to Rviz dying on me.

```
tobias@tobias-Yoga-Slim-7-14ARE05:~/lbr-stack$ ros2 launch lbr_bringup bringup.launch.py moveit:=true sim:=true
[INFO] [launch]: All log files can be found below /home/tobias/.ros/log/2023-11-21-09-49-51-134306-tobias-Yoga-Slim-7-14ARE05-6199
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [6619]
[INFO] [gzclient-2]: process started with pid [6621]
[INFO] [spawn_entity.py-3]: process started with pid [6624]
[INFO] [spawner-4]: process started with pid [6626]
[INFO] [robot_state_publisher-5]: process started with pid [6628]
[INFO] [spawner-6]: process started with pid [6630]
[INFO] [static_transform_publisher-7]: process started with pid [6632]
[INFO] [static_transform_publisher-8]: process started with pid [6634]
[INFO] [move_group-9]: process started with pid [6637]
[static_transform_publisher-8] [INFO] [1700556592.178000872] [static_transform_publisher_NG8X75i22ZLKdZrS]: Spinning until stopped - publishing transform
[static_transform_publisher-8] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-8] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-8] from 'base_frame' to 'lbr/base_frame'
[static_transform_publisher-7] [INFO] [1700556592.178243568] [static_transform_publisher_W247E5SDT6XqZRHE]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'world' to 'base_frame'
[robot_state_publisher-5] [INFO] [1700556592.179903399] [lbr.robot_state_publisher]: got segment base_frame
[robot_state_publisher-5] [INFO] [1700556592.180085753] [lbr.robot_state_publisher]: got segment link_0
[robot_state_publisher-5] [INFO] [1700556592.180102864] [lbr.robot_state_publisher]: got segment link_1
[robot_state_publisher-5] [INFO] [1700556592.180114318] [lbr.robot_state_publisher]: got segment link_2
[robot_state_publisher-5] [INFO] [1700556592.180127658] [lbr.robot_state_publisher]: got segment link_3
[robot_state_publisher-5] [INFO] [1700556592.180139042] [lbr.robot_state_publisher]: got segment link_4
[robot_state_publisher-5] [INFO] [1700556592.180150146] [lbr.robot_state_publisher]: got segment link_5
[robot_state_publisher-5] [INFO] [1700556592.180162438] [lbr.robot_state_publisher]: got segment link_6
[robot_state_publisher-5] [INFO] [1700556592.180174870] [lbr.robot_state_publisher]: got segment link_7
[robot_state_publisher-5] [INFO] [1700556592.180187371] [lbr.robot_state_publisher]: got segment link_ee
[move_group-9] [INFO] [1700556592.211893320] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-9] [INFO] [1700556592.211992354] [moveit_robot_model.robot_model]: Loading robot model 'iiwa7'...
[move_group-9] [INFO] [1700556592.212004646] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[spawn_entity.py-3] [INFO] [1700556592.750761817] [lbr.spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1700556592.751120448] [lbr.spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [1700556592.753504317] [lbr.spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-3] [INFO] [1700556592.764899716] [lbr.spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1700556592.765218817] [lbr.spawn_entity]: Waiting for service /spawn_entity
[move_group-9] [INFO] [1700556592.838651441] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-9] [INFO] [1700556592.838841408] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-9] [INFO] [1700556592.839570753] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-9] [INFO] [1700556592.840029186] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/attached_collision_object' for attached collision objects
[move_group-9] [INFO] [1700556592.840055866] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-9] [INFO] [1700556592.840407233] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/planning_scene'
[move_group-9] [INFO] [1700556592.840426719] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-9] [INFO] [1700556592.840796525] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[... cut ...]
[ERROR] [rviz2-10]: process has died [pid 7879, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/tobias/lbr-stack/install/iiwa7_moveit_config/share/iiwa7_moveit_config/config/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_1xwxz3zf --params-file /tmp/launch_params_3m3fvpks -r robot_description:=lbr/robot_description -r robot_description_semantic:=lbr/robot_description_semantic -r display_planned_path:=lbr/display_planned_path -r monitored_planning_scene:=lbr/monitored_planning_scene'].

```

Do you have any idea of why this happens?

mhubii, 2023-11-21:
woah that's strange, can confirm this happens, but don't know why

TobiasAxelsen, 2023-11-21:
Will ask around and see if I can have some of my co-students check if this also happens for them... Possibly test on an ur3, will report back if I find anything

mhubii, 2023-11-21:
Just out of curiosity, how much are you dependent on moveit?

EliasTDam, 2023-11-22:
Hey Martin! Thank you for everything so far, I have no idea where we would be without your help! We appreciate it!

Any updates regarding MoveIt? We are considering different approaches to making a cartesian trajectory. I tried to get it working through the action server, but I haven't found any documentation, so it feels like sharpshooting blindfolded. Do you know where to find any documentation for it? 

Specifically, I tried writing to referenceTrajectories in the action server, and add a custom path I created by interpolating between two poses. The server denies my request though, as it detects some error with the classes if I recall correctly. I have yet to spot my error, but I don't even know if the action server works this way, so it feels demotivating to sink even more time into this method without knowing for sure. In the end, I am just guessing that giving it a cartesian path in referenceTrajectories will make it move like I want to. I didn't see any easy toggle in the interface for using a cartesian path instead.

We have considered if it is an option to just skip MoveIt, and send a Pose directly to the Kuka and ask it to move there in cartesian space. We have done this with UR robots in the past, but we don't know if this is possible with the Kuka through FRI, and if there is a "simple" way to do this through lbr stack. We would lose out on collision detection, but at this point we just need the robot to move properly.

We also tried briefly with the C++ library for MoveIt, without success. We wanted to try the newly developed Python library, but there is basically no documentation, so we have no idea how to use it.

mhubii, 2023-11-22:
hi @EliasTDam . So if you can skip Moveit, and just want to execute a Cartesian motion, you can do something like: `dq = J^# @ dx`. There are many ways to achieve that. I can provide you with a script, np

EliasTDam, 2023-11-22:
Thanks Martin! This is exactly the kind of insight we needed. We just tested it, and also gained a better understanding of the /lbr/command/position topic. Perhaps we might just implement our own trajectory planner for the sake of learning, or use the library you sent. Thank you very much!

mhubii, 2023-11-22:
Okay hope this helps you move forward. Just be aware that the robot will execute what you ask it to execute. There are some velocity checks but be mindful regardless and execute in T1 mode until you are confident.

EliasTDam, 2023-12-05:
I'd just like to get back to you here Martin, and tell you that we now have it working in cartesian space as we wanted and interfacing with our own systems as a service.  It is custom for our usecase, but I'll attach the code here for future reference for others: https://github.com/andreasHovaldt/pose_commander

mhubii, 2023-12-05:
very excited to hear about this! Maybe this could be added to the demos?

EliasTDam, 2023-12-05:
Theoretically yes, but it is setup as a service with a custom interface (it's just geometry_msgs/msg/Pose combined with a "bool success" as the response). It would probably work if anyone wants to use it, as long as they modify the URDF imported and the links for the kinematic chain in Kinpy (like mentioned earlier in this issue).

For us it is a handy tool, as it interpolates straight cartesian paths and executes them rather smoothly. I would love to modify it to make it more universal, so it can easily be used by others, but like many others we are caught up in our semester project.

mhubii, 2023-12-05:
makes a lot of sense. In case you find time, you can fork this repo, add the `pose_commander` as is somewhere to demos, and I can review / fork your fork to suggest changes etc.

mhubii, 2023-12-29:
hi @EliasTDam , the Cartesian path through moveit is now fixed, thanks to @josefinemonnet