lbr-stack/lbr_fri_ros2_stack #134: New package with our own robot setup.

Shenwei1234, 2023-11-06:
Hi

We have used your bringup system as a means to create a simulation of our own dual-arm setup (currently there's only one arm). We have created our own 'package' called 'left_arm_moveit_config' which is a copy of your 'iiwa7_moveit_config', but changed all necessary parameters in all necessary files (we think so...) to suit ours (see image NewPackage.png). 
[image]

We have also created a new URDF directory containing our URDF in the lbr_description directory(see image OurURDF.png), and changed all the files (we think so...) where URDF is called with our own URDF.

We then run 'ros2 launch lbr_bringup bringup.launch.py model:=left_arm sim:=true rviz:=true moveit:=true'. However, in the motion planning panel, there is no content in the 'Move Group Namespace', which we assume is why we are unable to plan&execute (see DualArmRviz.png).

We can also see using rqt_graph, that some nodes and connections are missing, compared to when running your iiwa7 simulation (see images OurRQTgraph2.png and YourRQTgraph.png). As you can see, we are missing the action connected from the node /lbr/position_trajectory_controller, and also our /lbr/move_group node is only connected to /lbr_display_planned_path, as compared to yours which connected to several other actions.
[image]
[image]

Can you help?:)

mhubii, 2023-11-10:
hi @Shenwei1234 , thanks for coming back. Could you please share the code repository then I can have a look

Shenwei1234, 2023-11-22:
Sorry for the very late response!

We eventually found that this behaviour could be fixed be changing the "Move Group Namespace" that was shown in Rviz directly from the moveit.rviz file in the custom moveit config package instead of the GUI. This seems to have fixed the error.  Although I still don't understand why this couldn't be done from the GUI 🤔

Shenwei1234, 2023-11-22:
Yes that is correct:)