lbr-stack/lbr_fri_ros2_stack #138: How to cite this repository?

TobiasAxelsen, 2023-11-21:
Hello

Throughout my project work I will be having to cite this repository, but I don't find any bibtex or something along those lines. Is there a preferred way that you want to be cited?

mhubii, 2023-11-21:
That is a very fair question and I very much appreciate that! Let me upload something to arxiv..

tvercaut, 2023-11-21:
@mhubii I suggest you also add a JOSS submitted badge on the main README here. You will find teh badge used in your JOSS review thread:
https://github.com/openjournals/joss-reviews/issues/5629

Additionally, once the arxiv paper is live, you should add a cff file and a "how to cite" section in the README. We can update from arxiv to JOSS as soon as the JOSS paper is (hopefully) accepted. Doc on cff for github:
https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files

mhubii, 2023-11-22:
There is now a citation section in the readme @TobiasAxelsen

closed 2023-11-22 by mhubii

TobiasAxelsen, 2023-11-22:
Thank you very much :)

commit c10d0a54 (2023-11-22): Merge pull request #139 from lbr-stack/dev-humble-citation

Closes #138