lbr-stack/lbr_fri_ros2_stack #141: FRI Session could not be established

adamleon, 2023-12-05:
Hello, I'm trying to set up our LBR Med 14 robot, but I cannot establish the FRI session.

[image]

I can ping the robot on 172.31.1.147, so I guess there is a connection. But it immediatly shuts down when I try to run the server.

My setup is Humble on 22.04. And I'm running
```
ros2 launch lbr_bringup bringup.launch.py model:=med14 sim:=false rviz:=true moveit:=false
```

I saw the error is similar to [this issue](https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/24), but I cannot resolve it in the same way.

Thanks in advance

adamleon, 2023-12-05:
172.31.1.148
255.255.255.0

Wow, that was a fast response

mhubii, 2023-12-05:
no problem, was just working on something. Can you try to update your netmask?

adamleon, 2023-12-05:
I updated it to 255.255.0.0 and it still failed if that was what you meant.

mhubii, 2023-12-05:
there is a chance that your FRI version is slightly different

adamleon, 2023-12-05:
I checked my FRI version and it's 1.15.1.SMedV121-15-0-35. If this is a version you want me to extract, then I'll do it

I tried to open the ports with the command spesified in the issue, but it didn't help

adamleon, 2023-12-05:
Hmm... I tired to save the version as outlined, however I get the error:
[image]

Could this be an issue?

mhubii, 2023-12-05:
no, this indicates that you'll have to add your robot in the `StationSetup.cat` -> `Topology` tab before you do something else

adamleon, 2023-12-05:
I fixed the issue of the Topology version. Seemed like it was set to Version 2. Setting it to Version 1, made me able to save the FRI version.

I have followed the robot setup. I ran into a few issues with Java versions, since I got errors in the LBRServer file when running on the default jdk. Changed the jdk to jdk1.7.0_80 and the compiler to 1.6, which somehow fixed the issue and I was able to upload the project

mhubii, 2023-12-05:
so in that case, could it be that the LBRServer java application does not work with your version?

adamleon, 2023-12-05:
Could be. The error I got was:
```
The type java.io.ObjectInputStream cannot be resolved. It is indirectly referenced from required .class files
```
after some Googling it seemed to be a compatibility issue with breaks at Java 1.8 and up. And I don't know what Java version my robot should be running on.

mhubii, 2023-12-05:
it might make sense for you to try and run KUKA's FRI demos first, i.e. without any ROS 1/2 dependency.

You might want to run the LBRJointSineOverlay java application and the corresponding FRI client side application

[image]

adamleon, 2023-12-06:
I did that now. Seems I get the same error. This is probably an issue with the FRI on the robot. I've contacted KUKA Service to help with this. I'll let you know when this is resolved

adamleon, 2023-12-06:
Thanks BTW. You've helped me alot!

mhubii, 2023-12-06:
happy to help, please let me know in case you find a solution!

adamleon, 2023-12-12:
Hey! I found the solution. The solution was that I had not installed any packages in the Sunrise Project. 
When I ran the SineOverlay test, it worked.
However, the adventure did not end there. The LBRServer did not compile because of a compile error "The type java.io.ObjectInputStream cannot be resolved", which was solved by setting the JRE System Library to jdk.1.7.0_80 and the Compiler to 1.6
It has something to do with enum not being supported in some versions of Java.

closed 2023-12-12 by adamleon

mhubii, 2023-12-13:
Thank you for reporting back @adamleon. Okay that is quite interesting and also inconvenient for you, glad you found a solution. 

Do you think this is something that could be changed in the code, rather than by changing the compiler?