lbr-stack/lbr_fri_ros2_stack #153: Joint Torque Control

xlzhu0317, 2024-01-30:
Thank you for your great work.
I am using this respository for KUKA IIWA LBR Med14 R820, and I want send Joint Torque to robot, but I cant found the python interface in this respository. So, I would appreciate it if you can give some advices on how to send Joint Torque to robot. 

 Thank you again!

mhubii, 2024-01-30:
hi @xlzhu0317. You can launch the system via:

```shell
ros2 launch lbr_bringup bringup.launch.py model:=med14 sim:=false ctrl:=forward_lbr_torque_command_controller
```

You'll see a topic `/lbr/command/torque`. The corresponding message definition is here https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/lbr_fri_msgs/msg/LBRTorqueCommand.msg

A demo node that sends torque commands can be found here https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/lbr_demos/lbr_demos_fri_ros2_python/lbr_demos_fri_ros2_python/torque_sine_overlay_node.py

mhubii, 2024-05-13:
this feature is now generally supported and documented here https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_cpp/doc/lbr_demos_cpp.html#torque-sine-overlay

closed 2024-05-13 by mhubii