lbr-stack/lbr_fri_ros2_stack #162: Got erros when running the simulation demo: LBR Demos ROS 2 Control Python

Kylemaycry, 2024-03-05:
Link to the demo steps: https://lbr-fri-ros2-stack-doc.readthedocs.io/en/humble/lbr_fri_ros2_stack/lbr_demos/lbr_demos_ros2_control_python/doc/lbr_demos_ros2_control_python.html#joint-trajectory-controller

Erros with Launch the LBRBringup:

[INFO] [rviz2-8]: process started with pid [36237]
[rviz2-8] qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
[rviz2-8] /opt/ros/humble/lib/rviz2/rviz2: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE
[ERROR] [rviz2-8]: process has died [pid 36237, exit code 127, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/caiwu/lbr_fri_ros2_stack_ws/install/lbr_bringup/share/lbr_bringup/config/config.rviz --ros-args -r __node:=rviz2'].
[spawner-4] [INFO] [1709654203.996651466] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [1709654204.206215242] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [1709654206.021395524] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [1709654206.235516987] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [1709654208.123110627] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [1709654208.333946508] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [1709654210.194092068] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [1709654210.409598284] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [ERROR] [1709654212.028141016] [lbr.spawner_joint_state_broadcaster]: Controller manager not available
[spawner-6] [ERROR] [1709654212.234718976] [lbr.spawner_joint_trajectory_controller]: Controller manager not available
[ERROR] [spawner-4]: process has died [pid 36073, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager controller_manager --ros-args -r __ns:=/lbr'].
[ERROR] [spawner-6]: process has died [pid 36080, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_trajectory_controller --controller-manager controller_manager --ros-args -r __ns:=/lbr'].

Info with running the lbr_joint_trajectory_executioner_node:
INFO] [1709654639.479362413] [joint_trajectory_executioner_node]: Waiting for action server to become available...
[INFO] [1709654640.493026495] [joint_trajectory_executioner_node]: Waiting for action server to become available...
[INFO] [1709654641.545950540] [joint_trajectory_executioner_node]: Waiting for action server to become available...
[INFO] [1709654642.552215065] [joint_trajectory_executioner_node]: Waiting for action server to become available...
[INFO] [1709654643.564740777] [joint_trajectory_executioner_node]: Waiting for action server to become available...

mhubii, 2024-03-06:
hi @Kylemaycry. Sometimes the very first launch launch fails because of Gazebo. A re-launch usually fixes it.

I see an error related to `wayland`: Have you configured your window manager? Could this thread be related: https://unix.stackexchange.com/questions/598099/could-not-find-the-qt-platform-plugin-wayland?

As an alternative you could try to run inside a Docker container: https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/docker/doc/docker.rst

mhubii, 2024-03-08:
closing this issue for now @Kylemaycry , hope the answer above helped. Please feel free to re-open.

closed 2024-03-08 by mhubii

BabuBabuDUT, 2025-01-02:
hello @mhubii I'm having some problems replicating this demo.
It's at [https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/lbr_demos/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py](url)
The following error occurs when no modifications are made
```
[INFO] [1735882172.589733034] [lbr.joint_trajectory_client]: Action server available.
[INFO] [1735882172.590060335] [lbr.joint_trajectory_client]: Rotating odd joints.
[ERROR] [1735882172.591373681] [lbr.joint_trajectory_client]: Goal was rejected by server.
[INFO] [1735882172.591650790] [lbr.joint_trajectory_client]: Moving to zero position.
[ERROR] [1735882172.592469636] [lbr.joint_trajectory_client]: Goal was rejected by server.
```
At the same time, the following ERROR occurs in the lbr_bringup terminal
```[ruby $(which ign) gazebo-3] [INFO] [1735882172.590558540] [lbr.joint_trajectory_controller]: Received new action goal
[ruby $(which ign) gazebo-3] [ERROR] [1735882172.590616486] [lbr.joint_trajectory_controller]: Incoming joint A1 doesn't match the controller's joints.
[ruby $(which ign) gazebo-3] [INFO] [1735882172.591971219] [lbr.joint_trajectory_controller]: Received new action goal
[ruby $(which ign) gazebo-3] [ERROR] [1735882172.592006359] [lbr.joint_trajectory_controller]: Incoming joint A1 doesn't match the controller's joints.
```
In gazebo Entity tree, the joint name is lbr_A1.. So I changed line 38 in the file joint_trajectory_client.py from ```joint_trajectory_goal.trajectory.joint_names.append(f"A{i + 1}")```
 to
 ```joint_trajectory_goal.trajectory.joint_names.append(f"lbr_A{i + 1}")```
There are no errors in the lbr_bringup  terminal, and the action seemed to be accepted
```
[ruby $(which ign) gazebo-3] [INFO] [1735882893.824892561] [lbr.joint_trajectory_controller]: Received new action goal
[ruby $(which ign) gazebo-3] [INFO] [1735882893.824969735] [lbr.joint_trajectory_controller]: Accepted new action goal
```
BUT the following error occurs about two seconds after running again:
```
[INFO] [1735882893.823955360] [lbr.joint_trajectory_client]: Action server available.
[INFO] [1735882893.824274793] [lbr.joint_trajectory_client]: Rotating odd joints.
[INFO] [1735882893.825519684] [lbr.joint_trajectory_client]: Goal was accepted by server.
Traceback (most recent call last):
  File "/home/u051/lbr-stack/install/lbr_demos_py/lib/lbr_demos_py/joint_trajectory_client", line 33, in <module>
    sys.exit(load_entry_point('lbr-demos-py', 'console_scripts', 'joint_trajectory_client')())
  File "/home/u051/lbr-stack/build/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py", line 73, in main
    joint_trajectory_client.execute(
  File "/home/u051/lbr-stack/build/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py", line 60, in execute
    result_future.result().result.error_code
AttributeError: 'NoneType' object has no attribute 'result'
[ros2run]: Process exited with failure 1
```
So far I don't know how to solve it, I am indeed very beginner in ros, please forgive my unprofessional description and tell me what to do. thanks!

mhubii, 2025-01-04:
thanks for sharing the issue @BabuBabuDUT. I have created a new issue for it under #244. As you already saw, there is an issue with the joint names in this demo. There is probably another issue there with the response handling. Sorry for the late response.