lbr-stack/lbr_fri_ros2_stack #163: Add support for controllers

mhubii, 2024-03-14:
- Cartesian controllers: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2
- Admittance controller: https://control.ros.org/master/doc/ros2_controllers/admittance_controller/doc/userdoc.html

mhubii, 2024-03-25:
you mean end-effector velocity? Yes, we should add support for MoveIt Servo etc

CarlDegio, 2024-03-25:
Actually I am referring to the end position controller, but it can also be achieved with the end speed controller.

Also, I have tried moveit servo. It's not work well. It seems that its solving frequency cannot reach 100Hz so there are noise when running. And it will move to all joints zero when error in config files, dangerous. It broke my heart :(

mhubii, 2024-03-25:
well lets get that sorted then, I am quite positive this should work now. Moving the end-effector to a target position can be achieved in a variety of ways

CarlDegio, 2024-03-25:
The Cartesian controllers mentioned here may work better. Especially I notice it has force controller and impedance controller, which only realized in kuka workbench (SmartServo) now.

mhubii, 2024-03-25:
jap, would be great to use an open-source version for Cartesian impedance control etc

mhubii, 2024-03-25:
by the way, if you would want to create a pull request for the moveit servo, I am more than happy to review it

matthias-mayr, 2024-03-25:
FYI, we ran the `cartesian_controllers` package on an iiwa7 with `iiwa_ros` in ROS 1 some years ago, but the results were a bit mixed. It was fairly difficult to find a stable configuration and it came of the cost of slow and sluggish behavior. There is [a master thesis report on that](https://lup.lub.lu.se/student-papers/search/publication/9064756). But I know that Stefan did some updates on the package that could have changed that.

For us, we proceeded to implement real Cartesian impedance control and made it [available on Github](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) and also [wrote a short paper](https://joss.theoj.org/papers/10.21105/joss.05194). It's fairly easy to integrate. However unfortunately I did not have time to port it to ROS 2 yet and I will also not have time in the near future.  
If anyone wants to do that, I am happy to answer questions and review the code.

mhubii, 2024-03-25:
hi @matthias-mayr , thank you for reaching out, been following your implementation, congrats on the PhD ! Would love to see what can be done re ROS 2 on your implementation. Let's see!

matthias-mayr, 2024-03-25:
Thanks. There have already been requests for ROS 2 and [there's an open issue](https://github.com/matthias-mayr/Cartesian-Impedance-Controller/issues/10). If one goes down the "easy" road like `cartesian_controllers` and has a separate branch it should be fairly straight forward. The base library that implements the algorithms is separated and just has a ROS wrapper.