lbr-stack/lbr_fri_ros2_stack #164: Controller spawner fails to find controller manager

liver121888, 2024-03-24:
Hi, I am using ROS 2 Humble and trying to launch the command:

```bash
ros2 launch lbr_bringup bringup.launch.py model:=med7 sim:=true rviz:=true moveit:=true
```

The terminal output is as follows:

```bash
[INFO] [launch]: All log files can be found below /home/liver/.ros/log/2024-03-24-12-33-19-252489-liver-Vivobook15-12674
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [12676]
[INFO] [gzclient-2]: process started with pid [12678]
[INFO] [spawn_entity.py-3]: process started with pid [12680]
[INFO] [spawner-4]: process started with pid [12682]
[INFO] [robot_state_publisher-5]: process started with pid [12684]
[INFO] [spawner-6]: process started with pid [12686]
[INFO] [static_transform_publisher-7]: process started with pid [12688]
[INFO] [move_group-8]: process started with pid [12690]
[static_transform_publisher-7] [INFO] [static_transform_publisher_456ND5uA8AE2MhvX]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'world' to 'lbr/world'
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_0
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_1
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_2
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_3
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_4
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_5
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_6
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_7
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_ee
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment world
[move_group-8] [INFO] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-8] [INFO] [moveit_robot_model.robot_model]: Loading robot model 'iiwa7'...
[move_group-8] [INFO] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-8] [INFO] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-8] [INFO] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/attached_collision_object' for attached collision objects
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/planning_scene'
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-8] [WARN] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-8] [ERROR] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[spawn_entity.py-3] [INFO] [lbr.spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [lbr.spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [lbr.spawn_entity]: Waiting for entity xml on robot_description
[move_group-8] [INFO] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-8] [INFO] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-8] [INFO] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-8] [INFO] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-8] [INFO] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-8] [INFO] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[move_group-8] [INFO] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-8] [INFO] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-8] [INFO] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was set to 0.050000
[move_group-8] [INFO] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[move_group-8] [INFO] [moveit.plugins.moveit_simple_controller_manager]: Added FollowJointTrajectory controller for joint_trajectory_controller
[... 28 lines cut ...]
[move_group-8] Loading 'move_group/MoveGroupMoveAction'...
[move_group-8] Loading 'move_group/MoveGroupPlanService'...
[move_group-8] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-8] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-8] 
[move_group-8] You can start planning now!
[move_group-8] 
[spawner-6] [INFO] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [INFO] [lbr.spawner_joint_trajectory_controller]: Waiting for '/lbr/controller_manager' node to exist
[spawner-4] [INFO] [lbr.spawner_joint_state_broadcaster]: Waiting for '/lbr/controller_manager' node to exist
[spawner-6] [ERROR] [lbr.spawner_joint_trajectory_controller]: Controller manager not available
[spawner-4] [ERROR] [lbr.spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner-6]: process has died [pid 12686, exit code 1, cmd '/home/liver/paradocs_ws/install/controller_manager/lib/controller_manager/spawner joint_trajectory_controller --controller-manager controller_manager --ros-args -r __ns:=/lbr'].
[ERROR] [spawner-4]: process has died [pid 12682, exit code 1, cmd '/home/liver/paradocs_ws/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --controller-manager controller_manager --ros-args -r __ns:=/lbr'].
```

A potential cause of the error may be this breaking change:
https://github.com/ros-controls/ros2_control/pull/1410
As on my laptop, I am using ros-humble-controller-manager (2.40.0-1jammy.20240304.152357), the error occured.
But on another laptop, we are using ros-humble-controller-manager 2.37.0 and the controllers can be successfully spawned.

mhubii, 2024-03-24:
thank you very much for sharing @liver121888 ! Will get this sorted ASAP

mhubii, 2024-03-24:
One thing you can try is to checkout the 'dev-humble-launch' branch until it gets merged. This should solve the issues

liver121888, 2024-03-26:
Unfortunately, that didn't work for me. Is there a way for me to check each node separately to see where is the root cause?
I checkout the dev-humble-launch branch, now the robot_state_publisher can be spawned, but 

ros2 topic echo /lbr/robot_description gives following output:

```bash
data: <?xml version="1.0" ?> <!-- =================================================================================== --> <!-- |    Th...
---
```

ros2 topic echo /lbr/joint_states gives nothing.

full log:
```bash
[INFO] [launch]: All log files can be found below /home/liver/.ros/log/2024-03-25-23-03-19-784397-liver-Vivobook15-25064
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [25066]
[INFO] [gzclient-2]: process started with pid [25068]
[INFO] [spawn_entity.py-3]: process started with pid [25070]
[INFO] [spawner-4]: process started with pid [25072]
[INFO] [robot_state_publisher-5]: process started with pid [25074]
[INFO] [spawner-6]: process started with pid [25076]
[INFO] [static_transform_publisher-7]: process started with pid [25078]
[INFO] [move_group-8]: process started with pid [25080]
[static_transform_publisher-7] [INFO] [static_transform_publisher_VYyIn1DvmNSUCnye]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'world' to 'lbr/world'
[move_group-8] [INFO] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-8] [INFO] [moveit_robot_model.robot_model]: Loading robot model 'med7'...
[move_group-8] [INFO] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_0
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_1
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_2
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_3
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_4
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_5
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_6
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_7
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment link_ee
[robot_state_publisher-5] [INFO] [lbr.robot_state_publisher]: got segment world
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-8] [INFO] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-8] [INFO] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/attached_collision_object' for attached collision objects
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/planning_scene'
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-8] [INFO] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-8] [WARN] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-8] [ERROR] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[move_group-8] [INFO] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-8] [INFO] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[spawn_entity.py-3] [INFO] [lbr.spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [lbr.spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[move_group-8] [INFO] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-8] [INFO] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-8] [INFO] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-8] [INFO] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-8] [INFO] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[... cut ...]
[spawner-6] [ERROR] [lbr.spawner_joint_trajectory_controller]: Controller manager not available
[ERROR] [spawner-6]: process has died [pid 25076, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_trajectory_controller --controller-manager controller_manager --ros-args -r __ns:=/lbr'].
[ERROR] [spawner-4]: process has died [pid 25072, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager controller_manager --ros-args -r __ns:=/lbr'].
```

mhubii, 2024-03-26:
hm quite strange indeed. Did you upgrade the `controller_manager` to the latest version? I.e. `ros-humble-controller-manager`

liver121888, 2024-03-31:
I tried to run the cmd in the docker environment, it works perfectly fine.
When comparing the apt pkgs, I found out I installed ros2_control, but I didn't install ros-humble-gazebo-ros2-control, somehow this package didn't get installed when I installed the gazebo.
Maybe it is due to my recent upgrade from ubuntu 20.04 to 22.04.
Now it works perfectly fine, thank you so much for your help, I am closing this issue.

closed 2024-03-31 by liver121888

mhubii, 2024-04-01:
glad this worked for you