lbr-stack/lbr_fri_ros2_stack #166: Setup iiwa14 with MoveItConfigsBuilder

antonio1matos, 2024-04-09:
Hello,
I was trying to launch the robot iiwa14 in Rviz with MoveItConfigsBuilder. I made a code similar to the launch file demo_launch that is present in moveit2 humble tutorials (ros2 launch moveit2_tutorials demo.launch.py)  that is used to launch the robot panda in the rviz  -> https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html

I tried to adapt that code to the package lbr_fri_ros2_stack using the robot iiwa14: 
this is my code:

import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration, PathJoinSubstitution
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from launch.actions import ExecuteProcess
from ament_index_python.packages import get_package_share_directory
from moveit_configs_utils import MoveItConfigsBuilder
from lbr_description import LBRDescriptionMixin, RVizMixin
from lbr_bringup import LBRMoveGroupMixin

def generate_launch_description():

    declared_arguments = []
    declared_arguments.append(
        DeclareLaunchArgument(
            "rviz_config",
            default_value="miariiwa1.rviz",
            description="RViz configuration file",
        )
    )

    return LaunchDescription(
        declared_arguments + [OpaqueFunction(function=launch_setup)]
    )

def launch_setup(context, *args, **kwargs):

    moveit_config = (
        MoveItConfigsBuilder("iiwa14")
        .robot_description(os.path.join(
                get_package_share_directory("lbr_description"),
                "urdf/iiwa14/iiwa14.urdf.xacro",
            )
        )
        .trajectory_execution(file_path="config/moveit_controllers.yaml")
        .planning_scene_monitor(
            publish_robot_description=True, publish_robot_description_semantic=True
        )
        .planning_pipelines(
            pipelines=["ompl", "pilz_industrial_motion_planner"]
        )
        .robot_description_kinematics(file_path="config/kinematics.yaml")
        .to_moveit_configs()   
    )

    # Start the actual move_group node/action server
    run_move_group_node = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        output="screen",
        parameters=[moveit_config.to_dict()],
    )

    rviz_base = LaunchConfiguration("rviz_config")
    rviz_config = PathJoinSubstitution(
        #[FindPackageShare("lbr_description"), "config", rviz_base]
        [FindPackageShare("miar_robot"), "launch", rviz_base]
        
    )

    # RViz
    rviz_node = Node(
        package="rviz2",
        executable="rviz2",
        name="rviz2",
        output="log",
        arguments=["-d", rviz_config],
        parameters=[
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
            moveit_config.robot_description_kinematics,
            moveit_config.planning_pipelines,
            moveit_config.joint_limits,
        ],
    )

    # Static TF
    static_tf = Node(
        package="tf2_ros",
        executable="static_transform_publisher",
        name="static_transform_publisher",
        output="log",
        arguments=["0.0", "0.0", "0.0", "0.0", "0.0", "0.0", "world", "link_0"],
    )

    # Publish TF
    robot_state_publisher = Node(
        package="robot_state_publisher",
        executable="robot_state_publisher",
        name="robot_state_publisher",
        output="both",
        parameters=[moveit_config.robot_description],
    )

    # ros2_control using FakeSystem as hardware
    ros2_controllers_path = os.path.join(
        get_package_share_directory("lbr_ros2_control"),
        "config",
        "lbr_controllers.yaml",
    )
    ros2_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[moveit_config.robot_description, ros2_controllers_path],
        output="both",
    )

    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "joint_state_broadcaster",
            "--controller-manager-timeout",
            "300",
            "--controller-manager",
            "/controller_manager",
        ],
    )

    arm_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["joint_trajectory_controller", "-c", "/controller_manager"],
    )

    nodes_to_start = [
        rviz_node,
        static_tf,
        robot_state_publisher,
        run_move_group_node,
        ros2_control_node,
        joint_state_broadcaster_spawner,
        arm_controller_spawner,
    ]

    return nodes_to_start

-----------
In this code in the line:  default_value="miariiwa1.rviz",        I call the rviz file miariiwa1.rviz that itś not present in the package br_fri_ros2_stack. That code was created in a other package that i create called "miar_robot" however the miariiwa1.rviz code is attached below.
[miariiwa1.zip](https://github.com/lbr-stack/lbr_fri_ros2_stack/files/14920854/miariiwa1.zip)

The code works. The robot appears in the rviz and i can use the command Plan for the robot plan  a trajectory (using the planner OMPL for example) however the command Execute or Plan & Execute doesnt work. I think itś because moveit itś not reading the controller manager cus when i run the launch file appears in the terminal:  [spawner-6] [INFO] [1712674899.497416360] [spawner_joint_state_broadcaster]: Waiting for ‘/controller_manager’ node to exist . Somehow i think moveit itś not reading the file: # ros2_control using FakeSystem as hardware
    ros2_controllers_path = os.path.join(
        get_package_share_directory("lbr_ros2_control"),
        "config",
        "lbr_controllers.yaml",
    )

mhubii, 2024-04-09:
okay, so I'll try to understand you code in more detail in a bit. Just a brief question, the `lbr_bringup` uses the config builder, see e.g. https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/036b23b2f91af70e1ca490e0c81660b6b17ae114/lbr_bringup/launch/move_group.launch.py#L21 

Is there anything that the `lbr_brinup` doesn't provide that you need? Just so I understand where you are coming from better.

antonio1matos, 2024-04-09:
In that case i will have to do a code similar to the one in lbr_fri_ros2_stack/lbr_bringup/launch/real.launch.py ?

antonio1matos, 2024-04-09:
Iḿ running in simulation

antonio1matos, 2024-04-09:
no

mhubii, 2024-04-09:
Okay so you could do:

- Launch using `ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true`
- Configure RViz to your needs
- Save configs: In RViz `File` -> `Save Config As` (save to your package)
- Copy paste the launch file and update the origin RViz config file to your saved file, here:
https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/036b23b2f91af70e1ca490e0c81660b6b17ae114/lbr_bringup/launch/sim.launch.py#L89

antonio1matos, 2024-04-09:
Do you have any code using the iiwa1a from the package br_fri_ros2_stack
where the robot from to a desired position (goal pose)? I need to do a pick
and place projec where i was planning to use the move_group to control the
robot. My idea was to use a launch file to appear the robot in the rviz and
then i would try to use a node to execute the movements of the robot. If
you could provide me with something to make my project easier, I would
greatly appreciate it.

antonio1matos, 2024-04-09:
Yes would prefer to avoid gazebbo, amd only use rviz

mhubii, 2024-04-10:
yes, you can e.g. checkout 

https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/133#issuecomment-1805480962
and 
https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/152#issuecomment-1924424184

Closing this issue for now, as it seems somewhat unrelated to this repo, but more general. You might need to simulated physics if you are planning to do grasping, RViz might not suffice.

closed 2024-04-10 by mhubii

mhubii, 2024-04-11:
hey, you can just do

```shell
ros2 launch lbr_bringup bringup.launch.py model:=iiwa14
```

This will launch the simulation with the `iiwa14` model

antonio1matos, 2024-04-11:
Yes i know, but i wanted to run it in the code. I just wanted to put in the terminal: ros2 launch lbr_bringup bringup.launch.py (i don't want to write everytime model:=iiwa14 moveit:=true)
However somehow the output is still iiwa7.
Still thank you for your attention

antonio1matos, 2024-04-11:
Ok, i already solve the problem. I had to change the default value from iiwa7 to iiwa14 in the file launch_mixin.py in lbr_description

antonio1matos, 2024-04-11:
Hello. I noticed that if you wanted to use ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true  you cant´t use the ompl planner and you cant even plan adn execute the robot in the rviz. However i changed the  robot files that are in  iiwa14_moveit_config to be equal to the files that were on the other packages (iiwa7_moveit_config , med7_moveit_config, med14_moveit_config). After making the changes the code    ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true  worked well. Now i can use the ompl planner with iiwa using that code. You might check that package (iiwa14_moveit_config) with the others becuase that are some differencaces and that's why it wasnt wrkung the planner. Thank you for providign the lbr_fri_ros2_stack

mhubii, 2024-04-11:
hm that is quite strange indeed, glad it worked for you in the end

antonio1matos, 2024-04-12:
Yes, thatś a way. Thank you

antonio1matos, 2024-04-12:
Hi @mhubii 
Im sorry to bother.
I will make a strange question and unrelated to the subsject of this issue. Right now i have a robot arm iiwa doing path planning and i would like to have associated to the end-effector a gripper. I don´t have a custom gripper for the robot arm. I wanted to know if you know if is possible to integrate a gripper of other robot for example the gripper of robot panda that is in the moveit2 tutorials. Obviously this is not an adequated gripper for the robot, but in simulation terms do you know if something like that is possible? Use a gripper of other robot in the iiwa?
Regards

mhubii, 2024-04-12:
no worries at all @antonio1matos ! Great question!

Maybe we could try integrating one of these grippers: https://github.com/PickNikRobotics/ros2_robotiq_gripper

Feel free to open a new issue for that!

antonio1matos, 2024-04-14:
ok, thank you. I will check that gripper