lbr-stack/lbr_fri_ros2_stack #169: LBR iiwa with gripper robotiq_2f_85

antonio1matos, 2024-04-15:
Recently, I started using the lbr iiwa14 arm. I used the launch file code: sim.launch.py from the lbr_bringup package. I wanted to add a gripper to the arm to start performing pick and place operations in the future, and the gripper I used was the robotiq_2f_85 found in   https://github.com/PickNikRobotics/ros2_robotiq_gripper.
To add the gripper to the end-effector of the iiwa14 arm, I went to the iiwa14 urdf and added the following line:
<xacro:include filename="$(find robotiq_description)/urdf/robotiq_2f_85_gripper.urdf.xacro" />
So the iiwa14 urdf became:
-----------------------------------------
<?xml version="1.0"?>
<robot name="iiwa14" xmlns:xacro="http://www.ros.org/wiki/xacro">

    
    <xacro:include filename="$(find lbr_description)/urdf/iiwa14/iiwa14_description.urdf.xacro" />

    <xacro:arg name="robot_name" default="lbr" />
    <xacro:arg name="port_id" default="30200" />
    <xacro:arg name="sim" default="true" />

    
    <link name="world" />

    
    <joint name="world_joint" type="fixed">
        <parent link="world" />
        <child link="link_0" />
    </joint>

    
    <xacro:iiwa14 robot_name="$(arg robot_name)" port_id="$(arg port_id)" sim="$(arg sim)" />
    
    
    <xacro:include filename="$(find robotiq_description)/urdf/robotiq_2f_85_gripper.urdf.xacro" />

</robot>
-----------------------------------------
In the gripper urdf, I set the parent as link_ee so that the gripper would be fixed to the end-effector of the arm, making the gripper urdf:
----------------------------------------------
<?xml version="1.0"?>
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="robotiq_gripper">
    
    <xacro:arg name="use_fake_hardware" default="true" />

    
    <xacro:include filename="$(find robotiq_description)/urdf/robotiq_2f_85_macro.urdf.xacro" />

    <xacro:robotiq_gripper name="RobotiqGripperHardwareInterface" prefix="" parent="link_ee" use_fake_hardware="$(arg use_fake_hardware)">
        <origin xyz="0 0 0" rpy="0 0 0" />
    </xacro:robotiq_gripper>
</robot>
---------------------------------------------------------
Then, from the iiwa14 urdf, which calls both the iiwa14 description and the gripper, I generate a new iiwa14_moveit_config using the setup assistant, which already includes the gripper. For example, in this new package (iiwa14_moveit_config), the moveit_controllers.yaml file already includes the gripper as follows:
-----------------------------------------------
# MoveIt uses this configuration for controller management

moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - joint_trajectory_controller
    - arm_controller
    - hand_controller

  joint_trajectory_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - A1
      - A2
      - A3
      - A4
      - A5
      - A6
      - A7
    action_ns: follow_joint_trajectory
    default: true
  arm_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - A1
      - A2
      - A3
      - A4
      - A5
      - A6
      - A7
  hand_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - robotiq_85_left_knuckle_joint
----------------------------------------------------------
Finally, when I try to run the sim.launch.py launch file again, the arm already appears with the gripper, and it is possible to plan a movement, as shown in the attached image
[image]
However, when I use the execute command, it fails. I cannot execute the movement. I believe this is due to the fact that the terminal shows this:

[move_group-8] [WARN] [1713220742.262760755] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing robotiq_85_left_knuckle_joint

It is not able to detect the robotiq_85_left_knuckle_joint joint. I believe this may be due to the fact that in the sim.launch.py launch file, functions like LBRDescriptionMixin, LBRROS2ControlMixin are called, where in the launch files where I use these classes, I do not define the gripper, such as in LBRROS2ControlMixin I only define the controllers related to the arm and not the gripper. Therefore, I would like to know if anyone could tell me if my addition process to the arm is correct, and if so, what I should edit in the launch files present in the lbr-stack so that I can already execute the trajectory of the arm+gripper.

antonio1matos, 2024-04-16:
Ok, I'll try to do that. I'll let you know once it's done.

antonio1matos, 2024-04-16:
@mhubii do i have the rights to push the package i have on my pc?
antonio@antonio-HP-Pavilion-Laptop-15-cs1xxx:~/lbr-stack/src/lbr_fri_ros2_stack$  git push origin iiwa14_gripper
ERROR: Permission to lbr-stack/lbr_fri_ros2_stack.git denied to antonio1matos.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

mhubii, 2024-04-16:
so you'd have to create a fork, see below

[image]

you can then make changes to your fork and create a pull request

antonio1matos, 2024-04-16:
Hi @mhubii
Sorry to bother. I'm having difficulty adding the lbr-stack package with my changes to include the gripper on my GitHub. Do you think it's possible for me to send you the package in a zip file and for you to add it to GitHub so that both you and I have access?

mhubii, 2024-04-16:
that would be quite cumbersome, and I would prefer not to do it that way

antonio1matos, 2024-04-16:
OK, one question. After forking your repository, do I need to clone the forked repository again? Or since I already have the package on my PC with the changes made, do I just need to push it to GitHub?

antonio1matos, 2024-04-16:
Hi @mhubii 
I think i already created the fork with my changes. You can see in this link: https://github.com/antonio1matos/lbr_fri_ros2_stack

mhubii, 2024-04-16:
Oh yes, true. That is something we can work on

antonio1matos, 2024-04-16:
Ok nice. If you then run: ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true      you will see that the robot will appear with the gripper and you will be able to plan a trajetory. To do that you need to put in MOtion Planning -> Interactive Marker Size 0.2 to be able to move the arm the way you want in the rviz as you can see the in the following picture.
[image]
Btw you will also have to have the package from the github as you mentioned before: https://github.com/PickNikRobotics/ros2_robotiq_gripper

Btw in the robotiq_gripper package in the package robotiq_description -> folder urdf -> file 2f_85.ros2_control.xacro i added the link_ee in: 
                <param name="link_ee">false</param> 

Instead of being:                 
                <param name="use_dummy">false</param> 
I put: 
------------
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:macro name="robotiq_gripper_ros2_control" params="
        name
        prefix
        sim_ignition:=false
        sim_isaac:=false
        isaac_joint_commands:=/isaac_joint_commands
        isaac_joint_states:=/isaac_joint_states
        use_fake_hardware:=false
        fake_sensor_commands:=false
        com_port:=/dev/ttyUSB0">

        <ros2_control name="${name}" type="system">
            
            <hardware>

                
                <param name="link_ee">false</param>
-----------------------------
After that if you run: ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true    you will that problem that was showing in the terminal that was:
 [move_group-8] [WARN] [1713220742.262760755] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known. Missing robotiq_85_left_knuckle_joint 
I think this is a simple problem, but as i said earlier i dont know what i need to change to be able to also execute the robot movement in the rviz.
Thank you for trying to help me

mhubii, 2024-04-16:
do you think it would be possible to put these changes into a new package under `lbr_demos`

antonio1matos, 2024-04-16:
Yes, i think so

antonio1matos, 2024-04-16:
@mhubii 
I dont know very well how to do that.
Can i put the roboiq package inside lbr_demos?
In that case you have all the packages needed to run the ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true correctly.

antonio1matos, 2024-04-17:
@mhubii do you want me to add you as collaborator in the repository: antonio1matos/lbr_fri_ros2_stack?

mhubii, 2024-04-17:
I can attempt to give you some high level suggestions, but you might need to work on some details yourself

antonio1matos, 2024-04-17:
Ok, sure. Any suggestion is always welcome.
I already was able to put the robot to execute the trajectory. The robot with the gripper is able to plan aand execute the trajectpry to the point that i want. However in the terminal it still always appears the following warninng: 
------
[move_group-8] [WARN] [1713220742.262760755] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known. Missing robotiq_85_left_knuckle_joint  
---------

antonio1matos, 2024-04-18:
Hi @mhubii 
AS you can see in this picture the robot is able to execute a certain trajectory: 
[image]
However a problem that i have that im not understanding is the fact that the links of the gripper robotiq_85_left_inner_knuckle_link, robotiq_85_right_inner_knuckle_link, robotiq_85_left_finger_tip_link and robotiq_85_right_finger_tip_link are not fixed in the rest of the gripper. This gripper are the links are loose and shaking in the gazebo. Do you know what could be the reason. In the rviz they are alright.

antonio1matos, 2024-04-18:
antonio@antonio-HP-Pavilion-Laptop-15-cs1xxx:~$ ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true 
[INFO] [launch]: All log files can be found below /home/antonio/.ros/log/2024-04-18-12-15-45-866222-antonio-HP-Pavilion-Laptop-15-cs1xxx-3996
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [3998]
[INFO] [gzclient-2]: process started with pid [4000]
[INFO] [spawn_entity.py-3]: process started with pid [4002]
[INFO] [spawner-4]: process started with pid [4004]
[INFO] [robot_state_publisher-5]: process started with pid [4006]
[INFO] [spawner-6]: process started with pid [4008]
[INFO] [static_transform_publisher-7]: process started with pid [4010]
[INFO] [move_group-8]: process started with pid [4012]
[static_transform_publisher-7] [INFO] [1713438947.622829782] [static_transform_publisher_1Dbz6kiJ4JkcRTD7]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'world' to 'lbr/world'
[robot_state_publisher-5] [INFO] [1713438947.638512117] [lbr.robot_state_publisher]: got segment link_0
[robot_state_publisher-5] [INFO] [1713438947.638683118] [lbr.robot_state_publisher]: got segment link_1
[robot_state_publisher-5] [INFO] [1713438947.638716110] [lbr.robot_state_publisher]: got segment link_2
[robot_state_publisher-5] [INFO] [1713438947.638741566] [lbr.robot_state_publisher]: got segment link_3
[robot_state_publisher-5] [INFO] [1713438947.638765657] [lbr.robot_state_publisher]: got segment link_4
[robot_state_publisher-5] [INFO] [1713438947.638789088] [lbr.robot_state_publisher]: got segment link_5
[robot_state_publisher-5] [INFO] [1713438947.638813234] [lbr.robot_state_publisher]: got segment link_6
[robot_state_publisher-5] [INFO] [1713438947.638837248] [lbr.robot_state_publisher]: got segment link_7
[robot_state_publisher-5] [INFO] [1713438947.638860677] [lbr.robot_state_publisher]: got segment link_ee
[robot_state_publisher-5] [INFO] [1713438947.638881733] [lbr.robot_state_publisher]: got segment robotiq_85_base_link
[robot_state_publisher-5] [INFO] [1713438947.638906532] [lbr.robot_state_publisher]: got segment robotiq_85_left_finger_link
[robot_state_publisher-5] [INFO] [1713438947.638930628] [lbr.robot_state_publisher]: got segment robotiq_85_left_finger_tip_link
[robot_state_publisher-5] [INFO] [1713438947.638954146] [lbr.robot_state_publisher]: got segment robotiq_85_left_inner_knuckle_link
[robot_state_publisher-5] [INFO] [1713438947.638967238] [lbr.robot_state_publisher]: got segment robotiq_85_left_knuckle_link
[robot_state_publisher-5] [INFO] [1713438947.638980072] [lbr.robot_state_publisher]: got segment robotiq_85_right_finger_link
[robot_state_publisher-5] [INFO] [1713438947.638999874] [lbr.robot_state_publisher]: got segment robotiq_85_right_finger_tip_link
[robot_state_publisher-5] [INFO] [1713438947.639024434] [lbr.robot_state_publisher]: got segment robotiq_85_right_inner_knuckle_link
[robot_state_publisher-5] [INFO] [1713438947.639047695] [lbr.robot_state_publisher]: got segment robotiq_85_right_knuckle_link
[robot_state_publisher-5] [INFO] [1713438947.639071715] [lbr.robot_state_publisher]: got segment world
[move_group-8] [INFO] [1713438947.714738343] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-8] [INFO] [1713438947.715144539] [moveit_robot_model.robot_model]: Loading robot model 'iiwa14'...
[spawn_entity.py-3] [INFO] [1713438948.221822485] [lbr.spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1713438948.222283691] [lbr.spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [1713438948.224545853] [lbr.spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-3] [INFO] [1713438948.226879103] [lbr.spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1713438948.227275256] [lbr.spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-3] [INFO] [1713438948.732115218] [lbr.spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-3] [INFO] [1713438949.129035349] [lbr.spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [lbr]
[gzserver-1] [INFO] [1713438949.270125702] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1713438949.274652822] [lbr.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /lbr
[gzserver-1] [INFO] [1713438949.274880558] [lbr.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1713438949.278217544] [lbr.gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1713438949.280370042] [lbr.gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-1] [INFO] [1713438949.280457858] [lbr.gazebo_ros2_control]: Loading parameter files /home/antonio/lbr-stack/install/lbr_ros2_control/share/lbr_ros2_control/config/lbr_controllers.yaml
[gzserver-1] [INFO] [1713438949.297255995] [lbr.gazebo_ros2_control]: Loading joint: A1
[... cut ...]
[move_group-8] [WARN] [1713438971.951567070] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing robotiq_85_left_knuckle_joint
[move_group-8] [WARN] [1713438972.962452921] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing robotiq_85_left_knuckle_joint
[move_group-8] [WARN] [1713438973.967562591] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing robotiq_85_left_knuckle_joint

mhubii, 2024-04-18:
this could be a relevant error

```shell
[gzserver-1] [ERROR] [1713438949.299166307] [lbr.gazebo_ros2_control]: The plugin failed to load for some reason. Error: According to the loaded plugin descriptions the class mock_components/GenericSystem with base class type gazebo_ros2_control::GazeboSystemInterface does not exist. Declared types are gazebo_ros2_control/GazeboSystem
```

Miftahur92, 2024-08-01:
@antonio1matos hi, were you able to solve this problem? 
I am trying to use robotiq hand-e gripper with kuka. I can see the kuka+gripper in the custom moveit config package and execute motion plan. However, similar to yours, I get an error for missing gripper joint.

antonio1matos, 2024-08-01:
Yes. At the time the problem was solved. Right now i'm using another gripper with the robot iiwa14. Now im a little busy but later i can send the package that i had to use the robot iiwa14 with that gripper

Miftahur92, 2024-08-01:
> Yes. At the time the problem was solved. Right now i'm using another gripper with the robot iiwa14. Now im a little busy but i had to change some parts in the urdf files to solve that problem.

Will it be possible to share the solution in your free time? Thanks in advance.

antonio1matos, 2024-08-01:
Sure. Do you have already the package of the arm with the gripper as presented in the pictures above?

Miftahur92, 2024-08-02:
Hi @antonio1matos I am using the real.launch.py as I am using real robot. I have the URDF file with arm+gripper (picture attached).
[image]

antonio1matos, 2024-08-05:
Hi @Miftahur92 . Until now I have never used communication with the real robot so I don't know if I will be of much help, because the package I made was intended to only be used with a physical simulator, in this case the gazebo. But I can tell you some things I did:

antonio1matos, 2024-08-05:
In my case, as I was using the gazebo, I added the ld.action of the arm controller to the sim.lunch.py file. In your case, since you will use real.launch.py i think you will have to do the same.
[image]
The same was also added in the file lbr_ros2_control/lbr_ros2_control/launch_mixin.py in LBRROS2ControlMixin:
[image]
In the iiwa14_moveit_config/config/moveit_controllers.yaml file I put the hand_controller like this:
[image]
And in the ros2_controllers.yaml file I have this:
[image]

However, in this case this file is not very important because since you are using the sim.launch.py or real.launch.py launch files you are not using the ros2_controllers.yaml file that is in the iiwa14_moveit_config/config package, but the lbr_controllers.yaml file in lbr_ros2_control/config . It should be noted that I have in hand_controller /**/.
[image]

antonio1matos, 2024-08-05:
If you are using the file sim.launch.py or real.launch.py you are reading the urdf: iiwa14.urdf.xacro, which is reading the file iiwa14_description.urdf.xacro and which this one is reading the file lbr_system_interface.xacro in lbr_ros2_control/config. From what I understand, this last file is responsible for sending the controllers to the plugin hardware you want. For example, if you wanted the gazebo you would define in the launch file that sim= True, which is the case for the sim.launch.py file. yes=False uses the real robot's plugin hardware (<plugin>lbr_ros2_control::SystemInterface</plugin>) which is the case with the real robot.
[image]
The same logic applies to the gripper. In the gripper package, this file corresponds to 2f_85.ros2_control.xacro, where depending on the value, I define whether I want the fake_hardware plugin, real robot…

In your case, if you want to control the opening of the gripper through this package, you may have to add a condition to the launch file real.launch.py that indicates that the variables sim_isaac, sim_ignition or use_fake_hardware are not active to control the gripper. 
[image]
As in my case I wanted to use the gazebo software and didn't know how to define the conditions for this, I ended up going to the robotiq_2f_85_macro.urdf.xacro file and commented on the line that includes the 2f_85.ros2_control.xacro file. This way, this file that is in the gripper package is no longer used.
[image]
Instead, I went to the lbr_system_interface.xacro file and included a new file that I created equivalent to gripper's ros2_control.xacro that I called robotiq_ros2_control_joints.xacro.
[image]
The robotiq_ros2_control_joints.xacro file (created ros2_control file) was placed also in lbr_ros2_control/config, that has this: 
<?xml version="1.0"?>

<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:property name="prefix" value="" />
    
    
    <joint name="${prefix}robotiq_85_left_knuckle_joint">
        <command_interface name="position" />
        <state_interface name="position">
            <param name="initial_value">0.7929</param>
        </state_interface>
        <state_interface name="velocity"/>
    </joint>

    <joint name="${prefix}robotiq_85_right_knuckle_joint">
        <param name="mimic">${prefix}robotiq_85_left_knuckle_joint</param>
        <param name="multiplier">-1</param>

        <command_interface name="position"/>
        
        

    </joint>
    <joint name="${prefix}robotiq_85_left_inner_knuckle_joint">
        <param name="mimic">${prefix}robotiq_85_left_knuckle_joint</param>
        <param name="multiplier">1</param>

        <command_interface name="position"/>
        
        

    </joint>
    <joint name="${prefix}robotiq_85_right_inner_knuckle_joint">
        <param name="mimic">${prefix}robotiq_85_left_knuckle_joint</param>
        <param name="multiplier">-1</param>

        <command_interface name="position"/>
         
        

    </joint>
    <joint name="${prefix}robotiq_85_left_finger_tip_joint">
        <param name="mimic">${prefix}robotiq_85_left_knuckle_joint</param>
        <param name="multiplier">-1</param>

        <command_interface name="position"/>
        
        

    </joint>
    <joint name="${prefix}robotiq_85_right_finger_tip_joint">
        <param name="mimic">${prefix}robotiq_85_left_knuckle_joint</param>
        <param name="multiplier">1</param>

        <command_interface name="position"/>
        
        

    </joint>
</robot>

I did it this way, which means, I didn't use the original file 2f_85.ros2_control.xacro because I think I would have to define in the launch file  the value to choose which hardware plugin for the gripper I should use and I didn't know very well how to define this in the code so I created the robotiq_ros2_control_joints.xacro file. But in your case if you want to use the hardware plugin related to the real gripper: <plugin>robotiq_driver/RobotiqGripperHardwareInterface</plugin> this way I don't think it will work. So as I said it might not be much help in your case. Anyway, you can always try to do what I did and see if you get better results. I advise you to back up the package you have (ex:  save the one you have now in a zip file in case you have any problems later). In my case, when I wanted to open the gripper or close it in moveit, i did this:

ros2 action list -t  

and the following action should appear:

/lbr/hand_controller/gripper_cmd control_msgs/action/GripperCommand

If you want to control the gripper through the terminal, I would do, for example:

ros2 action send_goal /lbr/hand_controller/gripper_cmd control_msgs/action/GripperCommand "command:
  position: 0.0 
  max_effort: 0.0"  

where in position I defined how much I wanted to open/close.

I hope I have helped in something.

Miftahur92, 2024-08-05:
Hi @antonio1matos thank you for your detail instruction. Certainly this will be helpful.

mhubii, 2024-09-10:
okay so #193 is now supported, hence a gripper demo is now needed