lbr-stack/lbr_fri_ros2_stack #171: Question regarding internal PID Controller Values

StephanSchwarz96, 2024-04-29:
Hello,

first of all, thanks for the great work! I am currently using the stack to control the robot using the "/lbr/command/joint_position" command, similar to the joint_sine_overlay example. I implemented my own limit check and publish small steps along my trajectory (within the limits).

Unfortunately, the robot moves very slow. After some digging in the lbr_fri_ros2 package I found, that an additional pid control is used with extra checks. 

What I am curious about is, why the pid parameter are set multiplied by the step_size:
pid.initPid(pid_parameters.p * dt, pid_parameters.i * dt, pid_parameters.d * dt, pid_parameters.i_max * dt,
                pid_parameters.i_min * dt, pid_parameters.antiwindup); 

After looking into the ros control_toolbox, this does not make much sense to me since it results in a very slow motion. Is there anything I am missing?

Thanks for any advice!

mhubii, 2024-05-01:
hi @StephanSchwarz96 sorry for the late reply. True, there is a PID to ensure smooth motion. The idea here was to ensure PID work across all robot communication rates. Alternative suggestions are very welcome!

To you question, have you tried increasing the P gain? You can do this e.g. here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/036b23b2f91af70e1ca490e0c81660b6b17ae114/lbr_ros2_control/config/lbr_system_interface.xacro#L24

StephanSchwarz96, 2024-05-02:
Hi @mhubii  thanks for the reply! Right now I just bridged the internal PID control since my program ensures smooth motions, but it is good to know about having this additional option.

closed 2024-05-02 by StephanSchwarz96

StephanSchwarz96, 2024-05-02:
Tbh I simply skipped the computation of the pid controller in:
```cpp
void JointPIDArray::compute(const value_array_t& command_target, const double* state,
                            const std::chrono::nanoseconds& dt, value_array_t& command) {

  std::for_each(command.begin(), command.end(), [&, i = 0](double& command_i) mutable {
      command_i = command_target[i];
      // command_i += pid_controllers_[i].computeCommand(command_target[i] - state[i], dt.count());
      ++i;
  });
}
```

I tried to remove more of the existing pid controller structure for performance but had some issues (still working on it) and it worked by just passing my own command.

I launch app.launch.py.