lbr-stack/lbr_fri_ros2_stack #172: Pose Demo as ROS 2 Controller

mhubii, 2024-05-05:
okay @BoWangFromMars , formatting mostly finished. Unfortunately the intended use of this repository is less documented atm, but you can find an explanation for your demo here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/dev-humble-cart-vel-demo/lbr_demos/lbr_demos_fri_ros2_advanced_cpp/doc/lbr_demos_fri_ros2_advanced_cpp.rst#pose-controller

this leaves me with 2 questions:

- [ ] Check pose limits?
- [ ] The `pose_control` should ideally be a `ros2_controller` a la https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/dev-humble-cart-vel-demo/lbr_ros2_control/include/lbr_ros2_control/forward_lbr_position_command_controller.hpp

would you be interested in turning the `pose_control` into a `ros2_controller`? I could assist you, otherwise I could do that myself. Let me know. We should open a new PR for that though.

_Originally posted by @mhubii in https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/170#issuecomment-2094873630_

BoWangFromMars, 2024-05-06:
Thank you@mhubii. I would like to try to do it on my own at first. If any problems in this process, I will ask for your help. I want to clarify the requirenments in this task, which I need to rewrite pose_control as ros2_controller(e.g https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/dev-humble-cart-vel-demo/lbr_ros2_control/include/lbr_ros2_control/forward_lbr_position_command_controller.hpp) and consider pose limits(such as position limits and velocity limits of the robot) simultaneously. Is it right?

mhubii, 2024-05-06:
that is correct. The goal would be for the user to run your `pose_control` via

```shell
ros2 launch lbr_bringup bringup.launch.py ctrl:=pose_control
```

or similar. Writing a `ros2_controller` can be a bit of a headache at first.

BoWangFromMars, 2024-05-20:
Hello, @mhubii. I feel sorry for that I did not have enough time to do this task two weeks ago,  because I was busy for my course and lab works. When I got ready to process it last week, I found you had done it. Great jobs, thank you.

mhubii, 2024-05-20:
hey @BoWangFromMars , thanks for returning to this. No worries at all!

So your demo is still a node, but ideally would be a ros2_controller. This hasn't been done yet.