lbr-stack/lbr_fri_ros2_stack #173: CK_COMPOUND_RETURN_ERROR

SUUHOLEE, 2024-05-13:
Hi mhubii,

Until yesterday, I was able to operate the robot well. (POSITION_CONTROL - POSITION & terminal: ros2 launch lbr_bringup bringup.launch.py model:=med14 sim:=false moveit:=true)

However, today when I try to communicate with the robot, I'm experiencing CK_COMPOUND_RETURN_ERROR as shown in the picture of the SMART PAD below.

[image]

Do you happen to know under what circumstances this ERROR occurs?

Thank you

SUUHOLEE, 2024-05-13:
It seems like you didn't quite understand it accurately. Could you please explain a bit more in detail?

mhubii, 2024-05-13:
did you execute the brake test?

SUUHOLEE, 2024-05-13:
No, I haven't tested the brakes yet. I don't know exactly how to do it.

mhubii, 2024-05-13:
okay no problem. Select:

Applications -> BrakeTestMonitorS...

Run the test as you run the LBRServer. Careful, when the test is finished, the robot will execute a sample motion.

mhubii, 2024-05-13:
This is unrelated to your problem, but could you re-install the `lbr-stack` following these instructions:

https://github.com/lbr-stack/lbr_fri_ros2_stack#quick-start

SUUHOLEE, 2024-05-13:
We are currently undergoing a brake test. Is it normal for the A1 joint to continuously move left and right?

Do I have to quit it midway? Or do I have to wait until it ends?

mhubii, 2024-05-13:
this test will move each joint left right, from A1 to A7. You have to press and hold play if you are in test mode

SUUHOLEE, 2024-05-13:
I reinstalled according to quick-start.

Can I end the break test and try connecting to the robot again?

SUUHOLEE, 2024-05-13:
LBRServer
Select the desired FRI send period [ms] : 5
Select your remote IP address : 192.170.10.5
Select the desired FRI control mode : POSITION_CONTROL
Select the desired FRI client command mode : POSITION
-> terminal : ros2 launch lbr_bringup bringup.launch.py  model:=med14 sim:=false rviz:=true moveit:=true

but same error occuired the first picture.

mhubii, 2024-05-13:
when you go to Station -> Safety -> Status

what can you see?

SUUHOLEE, 2024-05-13:
State : Warning
Time : ~

Safety Control
          Position sensor not referenced (K1:A1, A2, A3, A4, A5, A6, A7).
          These are deactivated rows in the PSM configurations.

mhubii, 2024-05-13:
what version of the FRI are you running and what did you install on the robot controller?

SUUHOLEE, 2024-05-13:
[image]

this is robot controller version??

yes, rotot was worked well yesterday :(
I also very confused why robot don't connect labtop

mhubii, 2024-05-13:
looks good to me. I would suggest a restart see if the error persists

SUUHOLEE, 2024-05-13:
Does restart mean to turn the robot off and on again?

If so, I think I did it at least 5 times.

mhubii, 2024-05-13:
hm okay

mhubii, 2024-05-13:
maybe you are somehow using the wrong version of the FRI

SUUHOLEE, 2024-05-13:
https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/152#issuecomment-1924424184

Probably so. I am using the code above with some modifications.

mhubii, 2024-05-13:
but the driver doesn't even connect right?

mhubii, 2024-05-13:
have you tried running the raw FRI?

SUUHOLEE, 2024-05-13:
how to running raw FRI?

SUUHOLEE, 2024-05-13:
suho@suho:~/lbr-stack3$ ros2 launch lbr_bringup bringup.launch.py model:=med14 sim:=false rviz:=true moveit:=true
[INFO] [launch]: All log files can be found below /home/suho/.ros/log/2024-05-14-02-30-16-275628-suho-18406
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [18408]
[INFO] [static_transform_publisher-2]: process started with pid [18410]
[INFO] [move_group-3]: process started with pid [18412]
[INFO] [spawner-4]: process started with pid [18414]
[INFO] [spawner-5]: process started with pid [18416]
[INFO] [spawner-6]: process started with pid [18418]
[INFO] [spawner-7]: process started with pid [18420]
[static_transform_publisher-2] [INFO] [1715621416.827391994] [static_transform_publisher_qQc4a9bmBimOE1qE]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'lbr/world'
[ros2_control_node-1] [WARN] [1715621416.885057154] [lbr.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1715621416.885884393] [resource_manager]: Loading hardware 'lbr_system_interface' 
[ros2_control_node-1] [INFO] [1715621416.892627664] [resource_manager]: Initialize hardware 'lbr_system_interface' 
[ros2_control_node-1] [INFO] [1715621416.892824800] [lbr_fri_ros2::AsyncClient]: Configuring client.
[ros2_control_node-1] [INFO] [1715621416.892839274] [lbr_fri_ros2::AsyncClient]: Command guard variant: 'default'.
[ros2_control_node-1] [INFO] [1715621416.892852058] [lbr_fri_ros2::CommandGuard]: *** Parameters:
[ros2_control_node-1] [INFO] [1715621416.892861489] [lbr_fri_ros2::CommandGuard]: *   Joint A1 limits: Position: [-170.0, 170.0] deg, velocity: 85.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892878676] [lbr_fri_ros2::CommandGuard]: *   Joint A2 limits: Position: [-120.0, 120.0] deg, velocity: 85.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892891875] [lbr_fri_ros2::CommandGuard]: *   Joint A3 limits: Position: [-170.0, 170.0] deg, velocity: 100.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892904667] [lbr_fri_ros2::CommandGuard]: *   Joint A4 limits: Position: [-120.0, 120.0] deg, velocity: 75.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892917404] [lbr_fri_ros2::CommandGuard]: *   Joint A5 limits: Position: [-170.0, 170.0] deg, velocity: 130.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892929877] [lbr_fri_ros2::CommandGuard]: *   Joint A6 limits: Position: [-120.0, 120.0] deg, velocity: 135.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892943138] [lbr_fri_ros2::CommandGuard]: *   Joint A7 limits: Position: [-175.0, 175.0] deg, velocity: 135.0 deg/s, torque: 200.0 Nm
[ros2_control_node-1] [INFO] [1715621416.892956714] [lbr_fri_ros2::CommandInterface]: *** Parameters:
[ros2_control_node-1] [INFO] [1715621416.892964801] [lbr_fri_ros2::CommandInterface]: *   pid.p: 1.0
[ros2_control_node-1] [INFO] [1715621416.892973524] [lbr_fri_ros2::CommandInterface]: *   pid.i: 0.0
[ros2_control_node-1] [INFO] [1715621416.892981818] [lbr_fri_ros2::CommandInterface]: *   pid.d: 0.0
[ros2_control_node-1] [INFO] [1715621416.892989932] [lbr_fri_ros2::CommandInterface]: *   pid.i_max: 0.0
[ros2_control_node-1] [INFO] [1715621416.892998237] [lbr_fri_ros2::CommandInterface]: *   pid.i_min: 0.0
[ros2_control_node-1] [INFO] [1715621416.893006182] [lbr_fri_ros2::CommandInterface]: *   pid.antiwindup: false
[ros2_control_node-1] [INFO] [1715621416.893017069] [lbr_fri_ros2::StateInterface]: *** Parameters:
[ros2_control_node-1] [INFO] [1715621416.893024606] [lbr_fri_ros2::StateInterface]: *   external_torque_cutoff_frequency: 10.0 Hz
[ros2_control_node-1] [INFO] [1715621416.893033723] [lbr_fri_ros2::StateInterface]: *   measured_torque_cutoff_frequency: 10.0 Hz
[ros2_control_node-1] [INFO] [1715621416.893041935] [lbr_fri_ros2::AsyncClient]: Open loop: 'true'.
[ros2_control_node-1] [INFO] [1715621416.893050106] [lbr_fri_ros2::AsyncClient]: Client configured.
[ros2_control_node-1] [INFO] [1715621416.899799450] [resource_manager]: Successful initialization of hardware 'lbr_system_interface'
[ros2_control_node-1] [WARN] [1715621416.900048072] [lbr.controller_manager]: [Deprecated]: Automatic activation of all hardware components will not be supported in the future anymore. Use hardware_spawner instead.
[ros2_control_node-1] [INFO] [1715621416.900109661] [resource_manager]: 'configure' hardware 'lbr_system_interface' 
[ros2_control_node-1] [INFO] [1715621416.900126510] [resource_manager]: Successful 'configure' of hardware 'lbr_system_interface'
[ros2_control_node-1] [INFO] [1715621416.900163230] [resource_manager]: 'activate' hardware 'lbr_system_interface' 
[ros2_control_node-1] [INFO] [1715621416.900180582] [lbr_fri_ros2::App]: Opening UDP socket with port_id 30200.
[ros2_control_node-1] [INFO] [1715621416.900217494] [lbr_fri_ros2::App]: Socket opened successfully.
[ros2_control_node-1] [INFO] [1715621416.900327469] [lbr_ros2_control::SystemInterface]: Waiting for robot heartbeat [1/10]. Port ID: 30200.
[... cut ...]
[move_group-3] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "lbr/link_7" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[move_group-3]          at line 235 in ./src/buffer_core.cpp
[move_group-3] Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "lbr/link_7" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)

above code is error code in terminal

mhubii, 2024-05-13:
I mean you can essentially see here

```shell
[ros2_control_node-1] [INFO] [1715621420.902732391] [lbr_ros2_control::SystemInterface]: Waiting for robot heartbeat [5/10]. Port ID: 30200.
```

that no connection is established ever

mhubii, 2024-05-13:
could you please share how you built this project?

SUUHOLEE, 2024-05-13:
There is nothing called project.

Using ROS is all about moving the robot using the path planning code mentioned above.

This is a project that reads magnetic field sensor values into another laptop after movement.

mhubii, 2024-05-13:
I am wondering how you compile this project

SUUHOLEE, 2024-05-13:
I am conducting experiments to optimize the arrangement of eight electromagnets in an electromagnet system to maximize force and torque, and validating this through experiments.

Is this information do you want information?

mhubii, 2024-05-13:
no.

What did you follow to build this projects?

SUUHOLEE, 2024-05-14:
Thank you for your kind reply.

I came back today and tried to communicate and confirmed that the error was gone.
That error code seems to occur when the Ethernet cable is not connected to the robot. (I connected it.)
Therefore, the Ethernet cable may be the problem or the cable connecting the Ethernet cable to the laptop.

mhubii, 2024-05-14:
awsome!

Make sure you re-install the `lbr-stack`:

- 1.
```shell
source /opt/ros/humble/setup.bash
export FRI_CLIENT_VERSION=1.15
mkdir -p lbr-stack/src && cd lbr-stack
vcs import src --input https://raw.githubusercontent.com/lbr-stack/lbr_fri_ros2_stack/humble/lbr_fri_ros2_stack/repos-fri-${FRI_CLIENT_VERSION}.yaml
rosdep install --from-paths src -i -r -y
```

2.
```shell
colcon build --symlink-install
```

3.
```shell
source install/setup.bash
ros2 launch lbr_bringup bringup.launch.py model:=med14 sim:=false moveit:=true
```

closed 2024-05-14 by mhubii