lbr-stack/lbr_fri_ros2_stack #175: Potential Error in CommandGuard

StephanSchwarz96, 2024-05-14:
Hello everyone,

I recently run into some issues with the received robot state data which lead me to digging into the lbr_fri_ros2 package. I am not sure, but i might found an error within the code in command_guard.cpp in line 53: 
```cpp
for (std::size_t i = 0; i < lbr_command.joint_position[i]; ++i) {
```

Unlike the other checks (for position or torque limits), this does not iterate over the size() but the value, which in my case always stops after the first iteration since my joint_1 was lower than 1. Am I missing something here?

Thanks for feedback :)

mhubii, 2024-05-14:
oh this looks fishy :O, where in the code can this be found?

mhubii, 2024-05-14:
thank you so much for sharing this @StephanSchwarz96 . Do you wish to create a PR for this or should I fix this quickly?

mhubii, 2024-05-15:
closing this for now as resolved. Thank you again very much! Please feel free to reach out if you have any questions / re-open the issue etc

closed 2024-05-15 by mhubii

mhubii, 2024-05-17:
hi @StephanSchwarz96 , this should now be fully resolved. Let me know if you have any other questions