lbr-stack/lbr_fri_ros2_stack #177: end_effector current position

SUUHOLEE, 2024-05-21:
hello mhubii

I want to return the current position of the end effector while the robot is moving, and I found that I should use the tf_2ros package. I installed it using the command:

`sudo apt-get install ros-humble-tf2-ros`

After that, I wrote the following code to return the current position of the end effector:

import rclpy
from rclpy.node import Node
import tf2_ros
import geometry_msgs.msg

class EndEffectorPositionSubscriber(Node):
    def __init__(self):
        super().__init__('end_effector_position_subscriber')
        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self)

        self.timer = self.create_timer(1.0, self.get_current_pose)

        # Getting the names of base_link and end_effector_link as defined in the URDF file
        self.base_link = "link_0"
        self.end_effector_link = "link_ee"

    def get_current_pose(self):
        try:
            transform = self.tf_buffer.lookup_transform(
                self.base_link,  # Robot's base link
                self.end_effector_link,  # Robot's end effector link
                rclpy.time.Time())
            
            current_pose = transform.transform.translation
            self.get_logger().info("Current End Effector Position: x={}, y={}, z={}".format(
                current_pose.x,
                current_pose.y,
                current_pose.z))
        except (tf2_ros.LookupException, tf2_ros.ConnectivityException, tf2_ros.ExtrapolationException) as e:
            self.get_logger().error("Failed to lookup transform: %s" % str(e))

def main(args=None):
    rclpy.init(args=args)
    end_effector_position_subscriber = EndEffectorPositionSubscriber()
    rclpy.spin(end_effector_position_subscriber)
    end_effector_position_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()

This is the code that I am running. I don't know why the code isn't being written all at once.

However, I am getting the following error codes:
`[ERROR] [1716293460.470951577] [end_effector_position_subscriber]: Failed to lookup transform: "link_0" passed to lookupTransform argument target_frame does not exist.`
`[ERROR] [1716293535.857385495] [end_effector_position_subscriber]: Failed to lookup transform: "link_ee" passed to lookupTransform argument source_frame does not exist.`

I used this method, but an error occurred. Do you know where the error might be occurring? :(
Alternatively, is there any other simple way to determine the position of the end effector?

mhubii, 2024-05-21:
thanks for sharing the issue. The frames in this package use a prefix.

Can you try `lbr/link_0` and `lbr/link_ee`

mhubii, 2024-05-21:
awesome! The prefix changes with the `robot_name` from the launch file

closed 2024-05-21 by mhubii