lbr-stack/lbr_fri_ros2_stack #178: Cartesian control

vopsi99, 2024-05-23:
Hello, I tried using this repository but my knowledge in ros is rather mediocre. I am curious if this stack has cartesian control implementation or not.

It may be a stupid question, but I am stuck for quite some time and I could really use some help..

mhubii, 2024-05-23:
no worries @vopsi99 and thank you for raising this issue. In principle yes, in practise users have to put in some effort to do this right now. I can pinpoint you to some examples and try to improve the demos.

What are you trying to achieve:

- Move to an absolute pose?
- Specify an end-effector velocity?

vopsi99, 2024-05-23:
I managed to move the end-effector incrementally but it is really unstable, so I need to move to an absolute pose with a certain velocity. I supposed this kind of motion would be the best to further integrate a stereo camera to position the robot in correlation to some markers in the workspace.

mhubii, 2024-05-23:
so there is a demo here: https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_advanced_cpp/doc/lbr_demos_advanced_cpp.html#pose-controller

You'd have to add a node that sends your desired pose. This can be done in Python, similar to this e.g. https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/lbr_demos/lbr_demos_advanced_cpp/src/pose_planning_node.cpp

vopsi99, 2024-05-23:
So I should only send from another node my desired pose to this demo? This is the demo that I modified to send incremental values for x,y,z on position and orientation, but on Y orientation, for example, it always went crazy even if I was doing really small movement increments.

mhubii, 2024-05-23:
okay I'll have to double check this demo, as it was community contributed. But this demo sends an absolute pose. 

You will have to make sure you:
- Get the initial pose
- Send a small increment on top of the initial pose

mhubii, 2024-05-23:
I think this demo needs an update and should really allow users to control a velocity rather than absolute poses maybe

vopsi99, 2024-05-23:
Yes, I realised that a really small increment (0.001) would make the movement smoother.For positioning on x,y and z everything seems fine, but when I am trying to orient it, it just breaks everything.

After sending the increment on Y for longer it tends to jump into a position and i get the inverse kinematics failed error.

I think velocity control would be really useful for everyone.

mhubii, 2024-05-23:
thank you for the feedback. Let's add support for this then.

BoWangFromMars, 2024-10-29:
Hello, I am back. I am the contributor for "pose control". The initial pose and position in this demo is singular to some extent, so that it can only support the movement in z axis. However, if we want to orient it, the inverse kinematics would be calculated in a wrong way.
The solution is setting a non-singular pose and position as the initial one. For example, I set the initial joint position as   
**ros2 topic pub --once /forward_position_controller/commands std_msgs/msg/Float64MultiArray "{data: [2.175727, 0.928842, -0.975961, -1.3158, 0.3883, 1.2622, 0.01274]}"**
And I use the following code to orient it, please take it as reference code (interpolation of quaternion, rotating the end effector around its x-axis, y-axis, z-axis)

```
#include "rclcpp/rclcpp.hpp"
#include "cmath"
#include <iostream>
#include <string>
#include "geometry_msgs/msg/pose.hpp"

#include "tf2/transform_datatypes.h"
#include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"
#include "tf2/convert.h"
#include "tf2/LinearMath/Quaternion.h"
#include "tf2/LinearMath/Matrix3x3.h"

using std::placeholders::_1;

class CartesianPosePublisherNode:public rclcpp::Node
{
  private:
    rclcpp::Publisher<geometry_msgs::msg::Pose>::SharedPtr cartesian_pose_publisher_;
    rclcpp::Subscription<geometry_msgs::msg::Pose>::SharedPtr cartesian_pose_subscriber_;

    geometry_msgs::msg::Pose initial_cartesian_pose_; // robot starts from this pose
    bool is_init_; 

    double amplitude_; // rad
    double frequency_; // Hz
    double sampling_time_; // sampling time for sending position command
    double phase_; // initial phase
    geometry_msgs::msg::Pose previous_cmd_pose_;

    geometry_msgs::msg::Pose target_pose; // target quanternion (target pose)
    geometry_msgs::msg::Pose cartesian_pose_command; // cartesian pose command which is published to other ROS topics
    double t = 0.0;
        
  private:
    /**
     * @function: callback function for Cartesian Pose Subscriber
     * @param msg Cartesian Pose of the robot
    */
    void topic_callback(const geometry_msgs::msg::Pose& msg)
    {            
      if(!is_init_) 
      {
        initial_cartesian_pose_ = msg;
        cartesian_pose_command = initial_cartesian_pose_;
        target_pose = initial_cartesian_pose_;

        rotating(target_pose, initial_cartesian_pose_, 'y', 10.0);

        std::cout << "target pose calculated by function rotating: " << std::endl;
        std::cout << "target_pose.orientation.x " << target_pose.orientation.x << std::endl;
        std::cout << "target_pose.orientation.y " << target_pose.orientation.y << std::endl;
        std::cout << "target_pose.orientation.z " << target_pose.orientation.z << std::endl;
        std::cout << "target_pose.orientation.w " << target_pose.orientation.w << std::endl;

        /* set "initial joints" of KUKA iiwa robot as "2.175727, 0.928842, -0.975961, -1.3158, 0.3883, 1.2622, 0.01274"(the unit is rad),
        and the following quanternion corresponds to the target pose of the end effector, expressed in the "Base" Frame of the robot.
        In addition, the target position of the end effector is as same as the initial position, which can be calculated from "initial joints".
        */

        // make the initial pose matrix of end effector rotate its x-axis for 5 degrees
        /*
        target_quanternion.orientation.w = 0.1350; 
        target_quanternion.orientation.x = -0.5842;
        target_quanternion.orientation.y = 0.7808;
[... 120 lines cut ...]
      frequency_ = 0.5;
      sampling_time_ = 0.01;
      phase_ = 0.0;

      // the following ros topics for gazebo simulation 
	    cartesian_pose_publisher_ = this->create_publisher<geometry_msgs::msg::Pose>(
          "/lbr/command/cartesian_pose", 10);
      cartesian_pose_subscriber_ = this->create_subscription<geometry_msgs::msg::Pose>(
          "/lbr/state/cartesian_pose", 10, 
          std::bind(&CartesianPosePublisherNode::topic_callback, this, _1));
	  }
};

int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);
	
  rclcpp::spin(std::make_shared<CartesianPosePublisherNode>()); 

  rclcpp::shutdown(); 
  return 0;
}
```

Another suggestion is, when you are not sure whether the position and pose is a singular one, please run it in Gazebo or other simulation environment first instead of real robot and print the inverse kinematics solution in "pose control node".  Comparing it with the current joint position, it is wrong when these two values differ a lot. Hope it is helpful and please contact me if any other problems. Thank you.

TalentLeshen-DLUT, 2025-01-07:
I checked all answers, but I didn't find any information related to velocity control (maybe I missed something). Could you please give me a demo of how to achieve the velocity control of the end joint of the robotic arm?

TalentLeshen-DLUT, 2025-01-09:
@mhubii Maybe I was too stupid. I ran the twist controller and found that the arm did move at absolute velocity, but how did I get it to move in a given trajectory? I want the robotic arm to travel a spiral at an absolute velocity.

TalentLeshen-DLUT, 2025-01-10:
@mhubii Maybe I'm too stupid! I tried running twist_controller and sent linear_msgs and angular_msgs to /lbr/command/twist_controller, which you'll see in the code below. I found that the movements of the robotic arms were not the same as I expected, they were about 10 times different (10 times different in the y direction, with a small error in the xz plane).

```
#include "rclcpp/rclcpp.hpp"
#include "geometry_msgs/msg/twist.hpp"
#include <cmath>

class SpiralMotionPublisher : public rclcpp::Node
{
public:
    SpiralMotionPublisher()
        : Node("spiral_motion_publisher"), 
        r_(0.012),           // Spiral radius (m)
        v_xz_(0.004),        // Linear velocity in the plane (m/s)
        h_(-0.02),          // Descending distance per revolution (m)
        time_step_(0.01),    // Time step (s)
        t_(0.0),             // Time counter (s)
        H_(0.1)             // Total descending distance (m)
    {
        // Calculate parameters
        double circumference = 2 * M_PI * r_;         // Circumference of the circle (m)
        T_ = circumference / v_xz_;                  // Time per revolution (s)
        omega_ = 2 * M_PI / T_;                       // Angular velocity (rad/s)
        vy_ = h_ / T_;                                // Vertical velocity in y direction (m/s)
        max_time_ = H_ * T_ / std::abs(h_);           // Maximum runtime (s)

        // Create publisher
        publisher_ = this->create_publisher<geometry_msgs::msg::Twist>("/lbr/command/twist", 10);

        // Create timer, callback is executed every 10 ms
        timer_ = this->create_wall_timer(
            std::chrono::milliseconds(static_cast<int>(time_step_ * 1000)),
            std::bind(&SpiralMotionPublisher::publishTwist, this));
    }

private:
    void publishTwist()
    {
        // Check if the maximum runtime is exceeded
        if (t_ >= max_time_)
        {
            RCLCPP_INFO(this->get_logger(), "Reached maximum runtime. Shutting down.");
            rclcpp::shutdown(); // Terminate program
            return;
        }

        // Create Twist message
        auto twist_msg = geometry_msgs::msg::Twist();

        // Calculate spiral linear velocity components
        twist_msg.linear.x = -r_ * omega_ * std::sin(omega_ * t_);  // Linear velocity in x direction
        twist_msg.linear.z = r_ * omega_ * std::cos(omega_ * t_);   // Linear velocity in z direction
        twist_msg.linear.y = vy_;                                  // Linear velocity in y direction

        twist_msg.angular.x = 0.0;  // Acceleration in x direction
        twist_msg.angular.z = 0.0;  // Acceleration in z direction
        twist_msg.angular.y = 0.0;  // Acceleration in y direction

        // Publish message
        publisher_->publish(twist_msg);

        // Update time
        t_ += time_step_;
    }

    rclcpp::Publisher<geometry_msgs::msg::Twist>::SharedPtr publisher_;  // Publisher
    rclcpp::TimerBase::SharedPtr timer_;  // Timer

    // Spiral parameters
    double r_;         // Spiral radius
[... 83 lines cut ...]
    rclcpp::TimerBase::SharedPtr timer_;  // Timer

    double time_step_; // Time step
    double t_;         // Current time
    double max_time_;  // Maximum runtime
};

int main(int argc, char *argv[])
{
    // Initialize ROS2
    rclcpp::init(argc, argv);

    // Create node
    auto node = std::make_shared<ConstantMotionPublisher>();

    // Run node
    rclcpp::spin(node);

    // Shutdown ROS2
    rclcpp::shutdown();
    return 0;
}

```
I guess this error has something to do with the fact that I didn't set the acceleration, because when I set the acceleration of y to 1, the arm started to move sideways at a fairly fast speed, but this movement was obviously incorrect. : (

mhubii, 2025-01-10:
everything is fine on your end @TalentLeshen-DLUT. This is because there is an exponential smooth on the command:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_fri_ros2/src/interfaces/position_command.cpp#L47

Configurable here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_description/ros2_control/lbr_system_config.yaml#L13

This is a known issue of this stack.

TalentLeshen-DLUT, 2025-01-13:
@mhubii Based on my tests, joint position tau does not seem to be a control parameter for a smooth command. We guessed he was a robot receiving parameters at a frequency. Because when we increase it, the robotic arm can't receive all the msg, and when we decrease it will vibrate. I tried to adjust the cartesian gains in the twist controller.

      joint_gains: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0] # joint gains

I changed its value to (1.6, 1.35, 3.5, 1.0, 1.0, 1.0), and the accuracy of the robot arm was improved to some extent. At present, in the y and z directions, its accuracy has reached about 0.02mm, but in the x direction, its accuracy is about 1.1mm. I want to improve the accuracy of the x direction, but I am at my wits' end. Can you give me some suggestions?

TalentLeshen-DLUT, 2025-01-13:
@mhubii Based on my tests, joint position tau does not appear to be a control parameter for the smoothing command. We guess this is a receiving parameter on a frequency. Because when we increase it, the robotic arm can't receive all the information, and when we decrease it will vibrate. So I adjusted this parameter to 0.01, which is consistent with the message frequency. I'm trying to adjust the Cartesian gain in the twist controller.

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_description/ros2_control/lbr_controllers.yaml#L123

I changed its value to (1.6, 1.35, 3.5, 1.0, 1.0, 1.0), and the accuracy of the robot arm was improved to some extent. At present, in the y and z directions, its accuracy has reached about 0.02mm, but in the x direction, its accuracy is about 1.1mm. I want to improve the accuracy of the x direction, but I am at my wits' end. Can you give me some suggestions?

TalentLeshen-DLUT, 2025-01-13:
Also, I ran the ros2 topic echo /lbr/joint states and it shows the following:
```
- lbr_A2 
- lbr_A3 
- lbr_A4 
- lbr_A6 
- lbr_A1 
- lbr_A5 
- lbr_A7
```
I'm not sure if the order of the joints will make a difference to the result.

mhubii, 2025-01-13:
> Can you give me some suggestions?

you could add some form of PID. I.e., specify a target trajectory, then compute a `twist` using `pose_target - pose`.

> I'm not sure if the order of the joints will make a difference to the result.

Yes, so this is the default `joint_state_broadcaster` of `ros2_control`. This might be changed by adding joint names to the configuration file

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_description/ros2_control/lbr_controllers.yaml#L8

TalentLeshen-DLUT, 2025-02-17:
@mhubii As shown in the figure, I changed joint_state_broadcaster in lbr_controllers.yaml.

[image]

But when I run ros2 topic echo /lbr/joint_states, I find that the order of the joints has not changed and I don't know how to correct it.

[image]

In addition, I want to obtain the actual Cartesian velocity of the end joint of the robotic arm, and I would like to ask if there is a more convenient method, such as I can directly obtain the Jacobian pseudo-inverse.

BoWangFromMars, 2025-02-20:
> I find that the above does not seem to be a problem, when I output robot_description, it is correct.
> At present, I am still confused about making the end-effector spiral, and I am not sure whether it is easier to use twist or cubic or quintic polynomials. Could you please provide me with your ideas, I would appreciate it.
>
>
>
> I find that the above does not seem to be a problem, when I output robot_description, it is correct.
> At present, I am still confused about making the end-effector spiral, and I am not sure whether it is easier to use twist or cubic or quintic polynomials. Could you please provide me with your ideas, I would appreciate it.

Hello, if you want to make the end-effector rotate. It is usual to use quaternion and interpolate using quaternion like q_{t} = t*q_{start} + (1-t)*q_{end}, t \belongs [0,1]. In my blog, I talked about what is quaternion, its property with rigorous mathematical proof, and its geometrical meanings and how to interpolate (https://blog.csdn.net/modest_laowang/article/details/142891820 and  https://blog.csdn.net/modest_laowang/article/details/143196862). But I write the blog in Chinese, maybe you can translate it or find some other English version. For me, I use quaternion interpolation to make the robot rotate and control the robot in real time. It always works well. If you want to calcalate the velocity of end effector (I do not know why you need, for ensuring the velocity not to exceed the limit?), you can use dv/dt = J dq/dt, and there are some related C++ library we can use to calculate J. Instead of this, we can calculate the angular velocity only based on the rotation matrix (it is easy to find how to do this in robotics books or books on theoretical mechanics) and calculate the linear velocity using the origin point of the end-effector. Hope this is helpful.

------------------------------------------------------------------------------------------------------------

Bo Wang, Ph.D. candidate

Neuro Engineering and Medical Robotics Lab, Medical Robotics Section (NearLab, MRS)
Department of Electronics, Information, and Bioengineering (DEIB)
Politecnico di Milano, Piazza Leonardo da Vinci, 32, 20133 Milan, Italy

Phone: +39 3444714398

TalentLeshen-DLUT, 2025-02-22:
[image]
As shown in the figure, this is the situation where I control the end of the manipulator with the Cartesian velocity by using the twist_controller, where /ee_velocity represents the actual speed and/ lbr/command/twist represents the target speed. I want to know how I can get my actual speed to the target speed. I've put in a lot of effort, but I still can't get a good result.

mhubii, 2025-04-15:
this should now be resolved with https://github.com/ros-controls/ros2_controllers/pull/1572