lbr-stack/lbr_fri_ros2_stack #181: UDP Connection not starting

OmidRezayof, 2024-05-28:
Hi,
I'm using an lbr med 7 R800 with FRI med 3.0. Unfortunately I was not allowed by my manager to contribute the package to the community due to legal issues. 
I cloned everything as explained in the [start up guide](https://lbr-stack.readthedocs.io/en/latest/) and followed the step by step guide to first run everything with FRI 2.5 using the simulators and things looked fine. 
After, I replaced the FRI-Client-SDK_Cpp.zip with my new zip file and had to make some changes to the LBRServer.java to meet the new syntax requirements (things have been changed slightly between FRI 2.5 and 3.0). I rebuilt the package as well after replacing the zip file. 
Finally LBRServer.java looks to run without errors on the smart pad. However, when I try to run the demos with the real hardware, the LBRServer app on the smart pad gives the error: "Timeout before FRI connection quality reached GOOD or EXCELLENT" after 10 seconds. I suspect that the UDP connection responsible for the FRI connection is not getting started thus getting this timeout error.
(I have checked the connection between the robot and PC by pinging the robot.) 
I tried to look into the example codes but I wasn't able to find the place that the UDP connection is initiated (i.e. I'm not sure where the App and AsyncClient are being called when trying to tun "Joint Trajectory Controller" python demo.)
I have also attached a picture from my terminal which looks like the UDP connection is not getting started for some reason.
 
I would greatly appreciate any help.
[image]
[image]

mhubii, 2024-05-29:
hi @OmidRezayof, thanks for sharing. Let me try to answer the questions that can be answered first:

- Where is the connection established? The system interface runs the connection here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/a752eb21271b42be7f6e45cbb333def87731b175/lbr_ros2_control/src/system_interface.cpp#L206

The `run_async` method of `lbr_fri_ros2::App` creates an asynchronous thread

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/a752eb21271b42be7f6e45cbb333def87731b175/lbr_fri_ros2/src/app.cpp#L105

- Why is the connection to established?

Maybe try running the plain FRI (without the ROS 2 wrapper), as described here: https://github.com/lbr-stack/fri#build

- License / legal? You can check KUKA's license note in the FRI files. From FRI `1.11` onward, these allow re-distribution. I don't know of `3.0`

mhubii, 2024-05-29:
hi @fredRocs, could you please do me a massive favor and try connecting your robot? See if you get a similar error?

OmidRezayof, 2024-05-29:
Thanks @mhubii for your reply!
I tried removing that line from the file you sent, rebuilt the whole lbr-stack workspace and now, I've got new errors.
Any idea about this?
[image]

mhubii, 2024-05-29:
Sorry about that. Will fix this proper soon.

See the error: expected 7, got 6?

There is a line in the code that requires changing to 6. Second.

OmidRezayof, 2024-05-29:
No need to apologize I really appreciate your support.
It looks like I was finally able to open the UDP channel, but now it's complaining about not being able to find the robot I guess?
What I do is I first start the LBRServer with the recommended specs and then quickly run the launch file as in [demo](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_py/doc/lbr_demos_py.html#joint-trajectory-controller).  

PS I have pinged the robot and it seems to be fine.
[image]

fredRocs, 2024-05-29:
> hi @fredRocs, could you please do me a massive favor and try connecting your robot? See if you get a similar error?

It seems to work fine on my side when I run ros2 launch lbr_bringup bringup.launch.py sim:=false ctrl:=joint_trajectory_controller model:=med14. The signal stays EXCELLENT

OmidRezayof, 2024-05-29:
Dear @mhubii 
I tried to fork the repo and send a PR so you can see my codes. Please accept my apologies since I'm not a pro in Github and may have made mistakes in this process (Github can be very confusing!!)

The final comments that I made on this issue thread were: 
1. Apparently after changing the required numbers for state interface to 6, the UDP socket is opened, however now it cannot find the robot on the UDP channel.
2. I also reported the fact that I'm having warnings when building the packages in my lbr-stack workspace which I'm not sure if they're contributing to the issue we are discussing.

mhubii, 2024-05-29:
okay great. Thanks for the PR. 

In your version of the `LBRServer` Java application, what IP do you set for the FRI connection? When you configure your FRI connection, there should be something similar to:

```java
fri_configuration_ = FRIConfiguration.createRemoteConfiguration(lbr_, "172.31.1.148");
```

This IP should match your laptop's IP address.

OmidRezayof, 2024-05-30:
Yes, I'm indeed using the same IP (.148) for my PC and am choosing the same on the SmartPad. Still not sure why the robot cannot find the communication line. 
It's hard for me to tell if you have made any changes to the branch I made and did a PR. Have you made any changes that can possibly solve this?

mhubii, 2024-05-30:
no problem. Your changes are fine. One thing I would suggest now, is to try and run KUKA's demo applications with the raw FRI (no `LBRServer`, not ROS 2 integration). You can find some simple instructions here:

- Build: https://github.com/lbr-stack/fri#build
- Example applications: https://github.com/lbr-stack/fri#run-the-apps

OmidRezayof, 2024-05-30:
I suspect it can be from the netmask issue. 
I have set my PC ip: 172.31.1.148 and the netmask of my pc as: 255.240.0.0
On the StationSetup > Configuration I see:
Subnet mask > FFFF0000
and KONI > Subnet mask > FFFFFF00

On my PC I have also tried using net masks of 255.255.0.0 and 255.255.255.0

Still connection not established. 
[image]
Please let me know if you see anything wrong. 
I will be trying their own example applications soon.

OmidRezayof, 2024-05-30:
@mhubii I finally got the permission from my PI to share the fri 3.0.
I'm not sure why I couldn't find my updated fri-3.0 branch on lbr-stack/fri but I uploaded the sdk under my forked repo which you should be able to see.
Let me know if you see anything that can fix the build issue.

PS the license wasn't the issue I just had to get the permission from my boss.

mhubii, 2024-05-30:
okay 2.7 is now live. One thing, I think I re-based your PR https://github.com/lbr-stack/lbr_fri_ros2_stack/pull/182

This might remove your contribution from the git history.

Could you please re-open a new PR with the same changes against this branch:

https://github.com/lbr-stack/lbr_fri_ros2_stack/tree/dev-humble-state-interface-fix

Want to make sure your contribution is rightfully attributed.

OmidRezayof, 2024-05-30:
Hmm, not sure why 2.7 because it has been mentioned as 3.0 on all their documentations and everything.
I will try it with the new codes and see if it fixes the build problems.
And for the new PR, do I have to do it just for the lbr_fri_ros2_stack or also for fri-2.7?

mhubii, 2024-05-30:
For `lbr_fri_ros2_stack`. It is strange because in the source code it says version `2.7.0`

OmidRezayof, 2024-05-30:
@mhubii I tried running the [example application](https://lbr-stack.readthedocs.io/en/latest/fri.html#run-the-apps) and I was getting all kinds of different errors on my PC every time I run the LBRJointSineOverlay.java on the smartpad:
!!decoding error on Monitor message: invalid wire_type!!
!!decoding error on Monitor message: missing required field!!
!!decoding error on Monitor message: end-of-stream!!
!!decoding error on Monitor message: parent stream too short!!
!!decoding error on Monitor message: end-of-stream!!
 
I suspect this is an issue with the new released fri package cuz when I downgraded to Sunrise 2.6 and fri 2.5, I didn't have this problem. 
I was able to succesfully establish the ROS/fri connection and move the joints using [demo](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_py/doc/lbr_demos_py.html#joint-trajectory-controller).
As of now, I think I will continue using fri 2.5 and if any updates after 3.0 was made by KUKA, I can maybe try to see if I can get it to work.   
I just wanted to thank you a lot for the troubleshooting @mhubii .

mhubii, 2024-05-30:
Ah okay great, well this issue here still needs fixing, will merge your contribution asap.

I would contact KUKA because there appears to be some version mismatch between the FRI client SDK (2.7.0) and what is on the controller. This might be a bug on their end?

OmidRezayof, 2024-05-31:
Sorry for my late reply. I'm not super sure what is exactly the version mismatch that you are talking about. 
I do see the version 2.7.0 on the header files but that could be potentially the Sunrise.workbench version, since we were actually using Sunrise.workbench 2.7.0, but I'm pretty sure all the documents that we have, even the file name of the fri option package we have received have been labeled fri-3.0. 

It could be potentially that the CMake support codes that you have written for fri<=2.x is not suitable for compiling fri-3.0? I have no idea but I wasn't able to run KUKA's example fri codes using the fri-3.0 SDK so I'm assuming it's either a KUKA issue or an issue from the CMake compatibility codes.

mhubii, 2024-05-31:
So this is the documentation that's included in the `.zip` file

[image]

It says version `2.7.0`, which is a little confusing. It could of course be, that something with the CMake compilation is odd. Would have to check that

mhubii, 2024-05-31:
thank you again @OmidRezayof  for you contribution! You should now be listed for you efforts.

I am closing this as the initial issue was resolved. As for the FRI, please feel free to open a new issue, either here or here https://github.com/lbr-stack/fri.

Please make sure to pull the latest changes. When running, make sure to specify your FRI version here

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/77f6979b3bf19596286d723598de6c4419d2ea48/lbr_ros2_control/config/lbr_system_parameters.yaml#L4

Happy to dig further into FRI 3 / 2.7. Have a good weekend

commit 77f6979b (2024-05-31): Merge pull request #186 from lbr-stack/dev-humble-state-interface-fix

closes #181
closed 2024-05-31 by mhubii