lbr-stack/lbr_fri_ros2_stack #188: no response from LBRServer

Miftahur92, 2024-06-07:
Hi all, seeking help to connect a KMR omnimove 200 LBR iiwa 14 with ROS2 using FRI.

I tried to replicate the example in the readme. The IP address is different for me, instead of 172.31.1.147 it is 172.31.1.10. Whenever I select the LBRServer application on KUKA controller, nothing happens. Also I get failure message in the terminal as well. I have no idea what is going on. 
[image]
[image]
[image]
[image]
[image]
[image]

mhubii, 2024-06-08:
hi @Miftahur92 , thanks for the feedback.

Your IP configuration should be fine and seems consistent.

Simple question first: Did you run the application?

Miftahur92, 2024-06-10:
Thank you @mhubii for answering.

for 1. I have tried to run the application by pressing the enabling switch and pressing the play button. However, nothing happens.
for 2. I have changed the IP address as indicated.

There is a safety controller warning. Still, I can move the robot manually with the KUKA controller.
[image]

mhubii, 2024-06-10:
I can suggest you to try and

1. Ping the robot (`ping 172.31.1.10`)
2. Check the FRI version (yours should be 1.16)
3. Run any of KUKA's plain demos: https://github.com/lbr-stack/fri?tab=readme-ov-file#build  
4. Check firewall: https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/67#issuecomment-1479794623

Miftahur92, 2024-06-11:
Hi @mhubii thank you again for your quick response. I was able to run the application. The problem was with the calibration and mastering. However, I am still unable to connect with ros2 properly because of time mismatch between the laptop and KUKA controller. I tried to find a solution but the kuka controller interface is different in all the tutorials and guidelines. Do you have any suggestion for changing the time in kuka controller?
[image]

mhubii, 2024-06-11:
okay, great, progress!

don't have experience with changing the time, however, if you connect a screen, a keyboard, and a mouse to the KUKA controller, you can use it like any other Windows PC. Maybe that's how you can set the time?

Miftahur92, 2024-06-13:
Hi, I couldn,t fix the time. But, I was able to run one of the plain demos you mentioned (https://github.com/lbr-stack/fri?tab=readme-ov-file#build)
Next i tried to run lbr_demos_py. In the KUKA smartpad I can see the connection was established. But, there was no motion from the robot and in the Rviz, the robot was not properly displayed. 
[image]
[image]

Miftahur92, 2024-06-13:
Hi @mhubii. I have found a solution to make it work. I changed the value of "use_sim_time" from false to true in real.launch.py. Now, I can move the kuka iiwa14 with moveit.

mhubii, 2024-06-13:
hi @Miftahur92, thank you for returning to this. Could you please share where you changed that value? 

`use_sim_time` should be false for the hardware, hence a little surprised