lbr-stack/lbr_fri_ros2_stack #189: issue: Dual-IIWA7 Arm Setup using ROS2 FRI Stack

rysabh, 2024-06-09:
# issue: dual_kuka_fri

I want to create a dual-iiwa7 arm setup in ROS2 using the FRI stack. I have already created a custom Xacro file that uses the robot namespace to spawn two robots. However, for simplifying the issue here I will only spawn one robot in my moveit package named "green_moveit_config". Our public Github repo is [here](https://github.com/rysabh/dual_kuka_fri).

In the Xacro file, I have also renamed link and joint names with a prefix to manage both arms in the same environment.

After creating the custom URDF, I used the MoveIt Setup Assistant to generate the MoveIt configurations.

I followed the instructions [here](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_moveit_config/doc/lbr_moveit_config.html#generate-moveit-configuration) in the documentation to modify the controller and other related files.

Additionally, I have remapped certain topics as follows to connect topics from our MoveIt package to topics from the robot driver (bringup.launch.py):

- `{robot_name}_display_planned_path` -> `display_planned_path`
- `{robot_name}_joint_states` -> `joint_states`
- `{robot_name}_monitored_planning_scene` -> `monitored_planning_scene`
- `{robot_name}_planning_scene` -> `planning_scene`
- `{robot_name}_planning_scene_world` -> `planning_scene_world`
- `{robot_name}_robot_description` -> `robot_description`
- `{robot_name}_robot_description_semantic` -> `robot_description_semantic`

However, I am encountering several issues:

**Steps to Reproduce the Error:**

1. Launch the default `bringup.launch.py` with `model:=iiwa7` and `rviz:=false` and `moveit:=false` and `sim:=false`.
2. Launch the custom MoveIt `ros2 launch green_moveit_config demo.launch`.

**Error:**
There is a mismatch in joint names. Planning works, but execution on the real robot fails. 

We observe this when doing a ROS2 service call `/kuka_green/controller_manager/list_controllers controller_manager_msgs/srv/ListControllers` and get the following response:

```
response:
controller_manager_msgs.srv.ListControllers_Response(controller=[
    controller_manager_msgs.msg.ControllerState(name='force_torque_broadcaster', state='active', type='force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster', claimed_interfaces=[], required_command_interfaces=[], required_state_interfaces=['estimated_ft_sensor/force.x', 'estimated_ft_sensor/force.y', 'estimated_ft_sensor/force.z', 'estimated_ft_sensor/torque.x', 'estimated_ft_sensor/torque.y', 'estimated_ft_sensor/torque.z'], is_chainable=False, is_chained=False, reference_interfaces=[], chain_connections=[]), 
    controller_manager_msgs.msg.ControllerState(name='joint_trajectory_controller', state='active', type='joint_trajectory_controller/JointTrajectoryController', claimed_interfaces=['A1/position', 'A2/position', 'A3/position', 'A4/position', 'A5/position', 'A6/position', 'A7/position'], required_command_interfaces=['A1/position', 'A2/position', 'A3/position', 'A4/position', 'A5/position', 'A6/position', 'A7/position'], required_state_interfaces=['A1/position', 'A1/velocity', 'A2/position', 'A2/velocity', 'A3/position', 'A3/velocity', 'A4/position', 'A4/velocity', 'A5/position', 'A5/velocity', 'A6/position', 'A6/velocity', 'A7/position', 'A7/velocity'], is_chainable=False, is_chained=False, reference_interfaces=[], chain_connections=[]), 
    controller_manager_msgs.msg.ControllerState(name='joint_state_broadcaster', state='active', type='joint_state_broadcaster/JointStateBroadcaster', claimed_interfaces=[], required_command_interfaces=[], required_state_interfaces=['A1/commanded_joint_position', 'A1/commanded_torque', 'A1/effort', 'A1/external_torque', 'A1/ipo_joint_position', 'A1/position', 'A1/velocity', 'A2/commanded_joint_position', 'A2/commanded_torque', 'A2/effort', 'A2/external_torque', 'A2/ipo_joint_position', 'A2/position', 'A2/velocity', 'A3/commanded_joint_position', 'A3/commanded_torque', 'A3/effort', 'A3/external_torque', 'A3/ipo_joint_position', 'A3/position', 'A3/velocity', 'A4/commanded_joint_position', 'A4/commanded_torque', 'A4/effort', 'A4/external_torque', 'A4/ipo_joint_position', 'A4/position', 'A4/velocity', 'A5/commanded_joint_position', 'A5/commanded_torque', 'A5/effort', 'A5/external_torque', 'A5/ipo_joint_position', 'A5/position', 'A5/velocity', 'A6/commanded_joint_position', 'A6/commanded_torque', 'A6/effort', 'A6/external_torque', 'A6/ipo_joint_position', 'A6/position', 'A6/velocity', 'A7/commanded_joint_position', 'A7/commanded_torque', 'A7/effort', 'A7/external_torque', 'A7/ipo_joint_position', 'A7/position', 'A7/velocity', 'auxiliary_sensor/client_command_mode', 'auxiliary_sensor/connection_quality', 'auxiliary_sensor/control_mode', 'auxiliary_sensor/drive_state', 'auxiliary_sensor/operation_mode', 'auxiliary_sensor/overlay_type', 'auxiliary_sensor/safety_state', 'auxiliary_sensor/sample_time', 'auxiliary_sensor/session_state', 'auxiliary_sensor/time_stamp_nano_sec', 'auxiliary_sensor/time_stamp_sec', 'auxiliary_sensor/tracking_performance', 'estimated_ft_sensor/force.x', 'estimated_ft_sensor/force.y', 'estimated_ft_sensor/force.z', 'estimated_ft_sensor/torque.x', 'estimated_ft_sensor/torque.y', 'estimated_ft_sensor/torque.z'], is_chainable=False, is_chained=False, reference_interfaces=[], chain_connections=[]), 
    controller_manager_msgs.msg.ControllerState(name='lbr_state_broadcaster', state='active', type='lbr_ros2_control/LBRStateBroadcaster', claimed_interfaces=[], required_command_interfaces=[], required_state_interfaces=['A1/commanded_joint_position', 'A1/commanded_torque', 'A1/effort', 'A1/external_torque', 'A1/ipo_joint_position', 'A1/position', 'A1/velocity', 'A2/commanded_joint_position', 'A2/commanded_torque', 'A2/effort', 'A2/external_torque', 'A2/ipo_joint_position', 'A2/position', 'A2/velocity', 'A3/commanded_joint_position', 'A3/commanded_torque', 'A3/effort', 'A3/external_torque', 'A3/ipo_joint_position', 'A3/position', 'A3/velocity', 'A4/commanded_joint_position', 'A4/commanded_torque', 'A4/effort', 'A4/external_torque', 'A4/ipo_joint_position', 'A4/position', 'A4/velocity', 'A5/commanded_joint_position', 'A5/commanded_torque', 'A5/effort', 'A5/external_torque', 'A5/ipo_joint_position', 'A5/position', 'A5/velocity', 'A6/commanded_joint_position', 'A6/commanded_torque', 'A6/effort', 'A6/external_torque', 'A6/ipo_joint_position', 'A6/position', 'A6/velocity', 'A7/commanded_joint_position', 'A7/commanded_torque', 'A7/effort', 'A7/external_torque', 'A7/ipo_joint_position', 'A7/position', 'A7/velocity', 'auxiliary_sensor/client_command_mode', 'auxiliary_sensor/connection_quality', 'auxiliary_sensor/control_mode', 'auxiliary_sensor/drive_state', 'auxiliary_sensor/operation_mode', 'auxiliary_sensor/overlay_type', 'auxiliary_sensor/safety_state', 'auxiliary_sensor/sample_time', 'auxiliary_sensor/session_state', 'auxiliary_sensor/time_stamp_nano_sec', 'auxiliary_sensor/time_stamp_sec', 'auxiliary_sensor/tracking_performance', 'estimated_ft_sensor/force.x', 'estimated_ft_sensor/force.y', 'estimated_ft_sensor/force.z', 'estimated_ft_sensor/torque.x', 'estimated_ft_sensor/torque.y', 'estimated_ft_sensor/torque.z'], is chainable=False, is chained=False, reference_interfaces=[], chain_connections=[])
])
```

We observe that joints are not renamed here.

**Expected Behavior:**
The system should correctly recognize and manage the custom link and joint names without any mismatches, and the controller manager should operate as expected based on the modified `lbr_controller.yaml` file.

**Actual Behavior:**
Mismatch in link and joint names leading to errors in the system's operation. The controller manager does not reflect the changes expected from the modified configuration.

Additionally we noticed that, when the link and joint names are not changed, and the launch file is launched with default `robot_name:=lbr`, it works.

mhubii, 2024-06-09:
thanks for raising this issue @rysabh.

Including two `macros` into a custom `xacro` currently does not yield the expected behavior. As you already correctly mention, this causes ambiguity in the joint / link names and all sorts of down-stream issues when trying to fix this.

We can definitely fix this!

rysabh, 2024-06-10:
Hi @mhubii, thank you for your reply. 

As you could see in the following Xacro file which we have created and referenced at all the required places. Our Xacro file takes {robot_name} as an argument to rename joints and links. However, we have already tried this to no avail.

```xml
    <xacro:macro name="xyz_lbr_system_interface"
        params="robot_name sim joint_limits system_parameters_path">
.
.
.
.

             <xacro:joint_interface name="${robot_name}_A1"
                min_position="${joint_limits[robot_name + '_A1']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A1']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A1']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A1']['effort']}"
                sim="${sim}" />
            <xacro:joint_interface name="${robot_name}_A2"
                min_position="${joint_limits[robot_name + '_A2']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A2']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A2']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A2']['effort']}"
                sim="${sim}" />
            <xacro:joint_interface name="${robot_name}_A3"
                min_position="${joint_limits[robot_name + '_A3']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A3']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A3']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A3']['effort']}"
                sim="${sim}" />
            <xacro:joint_interface name="${robot_name}_A4"
                min_position="${joint_limits[robot_name + '_A4']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A4']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A4']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A4']['effort']}"
                sim="${sim}" />
            <xacro:joint_interface name="${robot_name}_A5"
                min_position="${joint_limits[robot_name + '_A5']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A5']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A5']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A5']['effort']}"
                sim="${sim}" />
            <xacro:joint_interface name="${robot_name}_A6"
                min_position="${joint_limits[robot_name + '_A6']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A6']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A6']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A6']['effort']}"
                sim="${sim}" />
            <xacro:joint_interface name="${robot_name}_A7"
                min_position="${joint_limits[robot_name + '_A7']['lower'] * PI / 180}"
                max_position="${joint_limits[robot_name + '_A7']['upper'] * PI / 180}"
                max_velocity="${joint_limits[robot_name + '_A7']['velocity'] * PI / 180}"
                max_torque="${joint_limits[robot_name + '_A7']['effort']}"
                sim="${sim}" />
```

mhubii, 2024-06-10:
okay I see, will have to have a look

mhubii, 2024-06-11:
hi @rysabh, some more sources for joint name inconsistency:

- controller configurations (joint trajectory controller):
https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/18db6ffefe34cb52db5c680e112c40c4adfe0f17/lbr_ros2_control/config/lbr_controllers.yaml#L43

- custom controllers, (the custom controllers in this stack contain hard-coded joint names, you can e.g. simply not spawn them). See:
https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/18db6ffefe34cb52db5c680e112c40c4adfe0f17/lbr_ros2_control/include/lbr_ros2_control/lbr_state_broadcaster.hpp#L51

I am currently unsure what the best approach here would be.

samrudhmoode, 2024-06-12:
Hi @mhubii,

Thank you for your response. We implemented the suggested changes, but we are now encountering the following issue:
```
Waiting for Controller manager service to be available
```
which eventually leads to
```
Controller Manager not available
```

Despite the ROS 2 service list showing the presence of the controller manager, the bringup file fails to access it. As a result, the socket to the robot doesn't open, and we cannot establish a connection. Could there be another file we should check to resolve this issue?
[image]

mhubii, 2024-06-13:
this might be some namespace issue. Which nodes are shown when you do

```shell
ros2 node list
```

samrudhmoode, 2024-06-14:
Hi @mhubii! To clarify, I'm working with @rysabh on this project.
After adapting the changes you have mentioned,  we've updated our launch command as follows:

```ros2 launch lbr_bringup bringup.launch.py model:=kuka_green_iiwa7 sim:=false rviz:=true moveit:=true robot_name:=kuka_green```

Here's the debug information from the command:

```
[INFO] [launch]: All log files can be found below /home/battery/.ros/log/2024-06-13-22-49-10-696329-battery-XPS-8930-19723
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [19725]
[INFO] [ros2_control_node-2]: process started with pid [19727]
[INFO] [static_transform_publisher-3]: process started with pid [19729]
[INFO] [move_group-4]: process started with pid [19731]
[INFO] [spawner-5]: process started with pid [19733]
[INFO] [spawner-6]: process started with pid [19735]
[INFO] [spawner-7]: process started with pid [19745]
[INFO] [spawner-8]: process started with pid [19766]
[static_transform_publisher-3] [INFO] [1718344151.259514073] [static_transform_publisher_TgjBc7OG6KoTWhmN]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'world' to 'kuka_green/world'
[INFO] [rviz2-9]: process started with pid [19783]
[robot_state_publisher-1] [INFO] [1718344151.282705506] [kuka_green.robot_state_publisher]: got segment kuka_green_link_0
[robot_state_publisher-1] [INFO] [1718344151.282812471] [kuka_green.robot_state_publisher]: got segment kuka_green_link_1
[robot_state_publisher-1] [INFO] [1718344151.282823255] [kuka_green.robot_state_publisher]: got segment kuka_green_link_2
[robot_state_publisher-1] [INFO] [1718344151.282828784] [kuka_green.robot_state_publisher]: got segment kuka_green_link_3
[robot_state_publisher-1] [INFO] [1718344151.282833672] [kuka_green.robot_state_publisher]: got segment kuka_green_link_4
[robot_state_publisher-1] [INFO] [1718344151.282838786] [kuka_green.robot_state_publisher]: got segment kuka_green_link_5
[robot_state_publisher-1] [INFO] [1718344151.282843570] [kuka_green.robot_state_publisher]: got segment kuka_green_link_6
[robot_state_publisher-1] [INFO] [1718344151.282853811] [kuka_green.robot_state_publisher]: got segment kuka_green_link_7
[robot_state_publisher-1] [INFO] [1718344151.282858830] [kuka_green.robot_state_publisher]: got segment kuka_green_link_ee
[robot_state_publisher-1] [INFO] [1718344151.282863548] [kuka_green.robot_state_publisher]: got segment world
[move_group-4] [INFO] [1718344151.295211108] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00375446 seconds
[move_group-4] [INFO] [1718344151.295260580] [moveit_robot_model.robot_model]: Loading robot model 'iiwa7'...
[move_group-4] [INFO] [1718344151.295268806] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ros2_control_node-2] [INFO] [1718344151.301946752] [kuka_green.controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-2] [INFO] [1718344151.303076050] [kuka_green.controller_manager]: update rate is 100 Hz
[ros2_control_node-2] [INFO] [1718344151.303497910] [kuka_green.controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-2] [INFO] [1718344151.313360993] [kuka_green.controller_manager]: Received robot description file.
[ros2_control_node-2] [INFO] [1718344151.313689006] [resource_manager]: Loading hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1718344151.317268364] [resource_manager]: Initialize hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1718344151.317353543] [lbr_fri_ros2::AsyncClient]: Configuring client
[ros2_control_node-2] [INFO] [1718344151.317363823] [lbr_fri_ros2::AsyncClient]: Client command mode: 'POSITION'
[ros2_control_node-2] [INFO] [1718344151.317368332] [lbr_fri_ros2::AsyncClient]: Command guard variant 'default'
[ros2_control_node-2] [INFO] [1718344151.317395447] [lbr_fri_ros2::CommandGuard]: *** Parameters:
[ros2_control_node-2] [INFO] [1718344151.317400065] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A1 limits: Position: [-170.0, 170.0] deg, velocity: 98.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1718344151.317407931] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A2 limits: Position: [-120.0, 120.0] deg, velocity: 98.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1718344151.317413464] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A3 limits: Position: [-170.0, 170.0] deg, velocity: 100.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1718344151.317418544] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A4 limits: Position: [-120.0, 120.0] deg, velocity: 130.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1718344151.317423834] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A5 limits: Position: [-170.0, 170.0] deg, velocity: 140.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1718344151.317428895] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A6 limits: Position: [-120.0, 120.0] deg, velocity: 180.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1718344151.317434149] [lbr_fri_ros2::CommandGuard]: *   Joint kuka_green_A7 limits: Position: [-175.0, 175.0] deg, velocity: 180.0 deg/s, torque: 200.0 Nm
[... cut ...]
            <parent link="kuka_green_link_7" />
            <child link="kuka_green_link_ee" />
            <origin xyz="0 0 0.035" rpy="0 0 0" />
        </joint>

        <link name="kuka_green_link_ee">
        </link>
    </xacro:macro>
</robot>
```
Note: We have hard coded the link and joint names just to ensure there aren't any mistakes on our end!

Thanks for your support and prompt responses, @mhubii!. I hope this helps clarify the setup. Let me know if there are specific issues or additional information needed!

mhubii, 2024-06-18:
sorry bit busy with other things at the moment. Maybe the easiest approach here would be to try and run `ros2_control` standalone first, i.e. without MoveIt.

samrudhmoode, 2024-06-25:
Hello @mhubii,

Now **we can perform dual arm manipulation** using our **custom MoveIt package** by launching three programs as follows: (1) the driver for robot_1 with moveit:=false, (2) the driver for robot_2 with moveit:=false, and (3) the custom MoveIt package to handle motion planning. We achieve this by providing goals to the follow_joint_trajectory action for both robots.

**However, we're encountering a specific issue.**

Our custom MoveIt package, using the **default xacro** files, **successfully plans trajectories**. **However, execution fails** when pressing the "execute" button in MoveIt. Interestingly, if I **modify the xacro files to remove Gazebo and ros2_control elements** before running them through the MoveIt Setup Assistant, **both planning and execution works.** These control elements are shown below:

```xml
<!-- macros for simulation and real hardware -->
<!-- <xacro:dual_iiwa7_gazebo robot_name="${robot_name}" />
<xacro:dual_iiwa7_system_interface
    robot_name="${robot_name}"
    sim="${sim}"
    joint_limits="${joint_limits}"
    system_parameters_path="${system_parameters_path}" />
-->
```

However, if we want to visualize the live robot position in our custom MoveIt package, we need these lines enabled in our xacro file. Otherwise, if we disable these control elements and use follow_joint_trajectory action,  we can plan and execute trajectories. However, no live view of the robots in MoveIt.

**Therefore, could you please help us in visualizing the live robot position during real robot execution** (which we believe is possible using the control elements enabled in xacro?)

mhubii, 2024-06-25:
thank you for coming back to this interesting topic! Can I execute your modifications here: https://github.com/rysabh/dual_kuka_fri ?

ahnwoohyeon, 2024-08-19:
I understand that when using 2 robots, we have to name the joint differently. If so, do I just need to change the name of the joint_state macro of joint_limit.yaml, lbr_system_interface?

[lbr_fri_ros2_stack/lbr_ros2_control/include/lbr_ros2_control/lbr_state_broadcaster.hpp](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/18db6ffefe34cb52db5c680e112c40c4adfe0f17/lbr_ros2_control/include/lbr_ros2_control/lbr_state_broadcaster.hpp#L51) 
Should I change the joint name of this code? If I have to use this method, I will have to apply it separately, so I would appreciate it if you could tell me how to use this method. It is blocked here.

mhubii, 2024-09-10:
yes so the issue is exactly that joints / links need to be named uniquely